riscvarchive/riscv-gcc #5: Add option to disable divide but keep multiply

vanjoe, 2016-10-07:
I'd like to add a command line option to disable divide, but enable divide, along the lines of -mmuldiv.

Could someone point me in the right direction to go about doing this?

aswaterman, 2016-10-07:
You could look at -m[no-]fdiv and TARGET_FDIV, which is analogous (it suppresses fdiv/fsqrt on systems that otherwise have floating point).

palmer-dabbelt, 2016-10-07:
This commit https://github.com/riscv/riscv-gnu-toolchain/commit/f7a71cd82aced98c12ae4badf2e318e8d4387fef turned on fdiv by default.  It's in the old repo, so the paths will be a bit off, but that's the file that specifies -m arguments.

vanjoe, 2016-10-07:
Would splitting TARGET_MULDIV into TARGET_MUL and TARGET_DIV be a good thing from your perspective? I'm thinking if I end up merging changes back.

aswaterman, 2016-10-07:
That sounds good for the internals, but we shouldn't abandon the old command-line option.

vanjoe, 2016-10-07:
Yeah I understand that concern. but having -mmuldiv and -mno-idiv and perhaps -mno-imul looks redundant and or confusing

closed 2016-10-12 by vanjoe