riscvarchive/riscv-gcc #14: error: unrecognizable insn

rwmjones, 2016-10-26:
When compiling the Fedora package xorg-x11-server-utils-7.7-20.fc25, we hit the following compiler error:

```
  CC       iceauth.o
iceauth.c: In function 'main':
iceauth.c:166:1: error: unrecognizable insn:
 }
 ^
(insn 434 179 460 21 (set (mem/c:SI (symbol_ref:DI ("*.LANCHOR2") [flags 0x182]) [3 verbose+0 S4 A32])
        (reg:SI 224 [ i ])) iceauth.c:116 -1
     (nil))
iceauth.c:166:1: internal compiler error: in extract_insn, at recog.c:2287
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://bugzilla.redhat.com/bugzilla> for instructions.
Preprocessed source stored into /tmp/ccejULR8.out file, please attach this to your bugreport.
Makefile:498: recipe for target 'iceauth.o' failed
make[2]: *** [iceauth.o] Error 1
```

I am collecting additional information now and will update this bug.

rwmjones, 2016-10-26:
The precise command which provokes the error is:
```
gcc -c iceauth.c -O2 -g -fPIE -DHAVE_CONFIG_H
```
The -fPIE parameter seems to be crucial to causing the bug here.  Changing to -fPIC or removing it entirely makes the problem go away.

rwmjones, 2016-10-26:
Preprocessed file:
http://oirase.annexia.org/tmp/ccEDqVmw.out

rwmjones, 2016-10-26:
Source files:
http://oirase.annexia.org/tmp/config.h
http://oirase.annexia.org/tmp/iceauth.c

davidlt, 2016-10-26:
CC myself

aswaterman, 2016-10-26:
Solved with a deeply unsatisfying workaround: 2dac7c94d1d226c61526c2ddc2f596b3d87fa7f3

closed 2016-10-26 by aswaterman