riscvarchive/riscv-gcc #16: Missing TLS bits for -fstack-protector, -fstack-check

sorear, 2016-10-29:
`-fstack-protector` uses TP-relative access on other arches, which saves a couple instructions versus a variable accessed through the GOT.

`-fstack-check` does not appear to have a fallback, and is not operational.

aswaterman, 2016-10-29:
Re: `-fstack-protector`: for PIC, TP-relative accesses would shave off the GOT access, but would come at a per-thread memory cost and would require an additional ABI constraint.  It's probably a win overall, or the x86-64 ABI designers wouldn't have opted for the extra complexity.  OTOH, the AArch64 folks are doing the simple thing, like us.  Do you have any insight into their thought process?

For `-fno-pic`, this doesn't matter much, as there's no GOT access, and it might even relax to GP-relative.  For `-fpie`, I would think we could change GCC to do something similar.  So this should just affect shared libraries.

Re: `-fstack-check`: GCC provides a default implementation, which at cursory glance seems to work sanely.  Do you have a counter-example?  This one doesn't seem to impact the ABI, so could always be improved later.

sorear, 2016-10-29:
Sorry, I meant `-fsplit-stack`.  I have no idea how that typo happened and this is actually the first I knew `-fstack-check` was a valid flag.

sorear, 2016-10-29:
re. AArch64: No idea.  Probably a good idea to look into rationale.  I only looked at the x86-64 output before deciding "this optimization doesn't work on RISC-V, file a bug".

aswaterman, 2016-10-30:
Looks like only x86, s/390, and POWER support `-fsplit-stack`.  Seems like a good thing to have, but would require a lot more work than I have time to undertake.  I'll leave the issue open.

davidlt, 2016-11-01:
AArch64 split-stack support is under review (Oct 17th):  https://gcc.gnu.org/ml/gcc-patches/2016-10/msg01347.html

Might be a good reference for future.

aswaterman, 2016-11-01:
Thanks for the pointer @davidlt; at a glance, it looks like we could take a very similar approach.