riscvarchive/riscv-gcc #21: ICE: crash on graphite

kito-cheng, 2016-11-04:
I think it's graphite's bug, so don't waste time to investigate until other bug is clean.

Arch: RV32I, RV32G, RV64I, RV64G
Option: -O2 -fgraphite-identity

How to reproduce:
```
$ riscv64-unknown-linux-gnu-gcc id-pr46845.c -O2 -fgraphite-identity
id-pr46845.c: In function 'foo':
id-pr46845.c:17:1: internal compiler error: in commit_one_edge_insertion, at cfgrtl.c:2070
 }
 ^
0x633c58 commit_one_edge_insertion(edge_def*)
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/cfgrtl.c:2070
0x621953 execute
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/cfgexpand.c:6408
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://gcc.gnu.org/bugs.html> for instructions.
```

Code for id-pr46845.c:
```
typedef float V2SF __attribute__ ((vector_size (128)));

V2SF
foo (int x, V2SF a)
{
  V2SF b;
  if (x & 42)
    b = a;
  else
    b = a + (V2SF) {1.0f/0.0f - 1.0f/0.0f, 1.0f/0.0f - 1.0f/0.0f};
  while (x--)
    a += b;

  return a;
}
```

aswaterman, 2016-11-04:
Ack.

kito-cheng, 2017-05-03:
This issue seem disappear on GCC 7.

closed 2017-05-03 by kito-cheng