riscvarchive/riscv-gcc #23: Atomic test fail

kito-cheng, 2016-11-04:
Maybe it just same as @sorear report in https://github.com/riscv/riscv-gcc/issues/15?

Option: -std=c11 -latomic
Arch: RV64I and RV64G

How to reproduce:
```
$ riscv64-unknown-linux-gnu-gcc c11-atomic-exec-3.c -std=c11 -latomic
$ qemu-riscv64 -L `riscv64-unknown-linux-gnu-gcc --print-sysroot` a.out 
Aborted (core dumped)
```

Code for c11-atomic-exec-3.c:
```
/* { dg-do run } */
/* { dg-options "-std=c11 -pedantic-errors" } */

extern void abort (void);
extern void exit (int);

#define TEST_INCDEC(TYPE, VALUE, PREOP, POSTOP, PRE_P, CHANGE)		\
  do									\
    {									\
      static volatile _Atomic (TYPE) a = (TYPE) (VALUE);		\
      if (PREOP a POSTOP != (PRE_P					\
			     ? (TYPE) ((TYPE) (VALUE) + (CHANGE))	\
			     : (TYPE) (VALUE)))				\
	abort ();							\
      if (a != (TYPE) ((TYPE) (VALUE) + (CHANGE)))			\
	abort ();							\
    }									\
  while (0)

#define TEST_INCDEC_ARITH(VALUE, PREOP, POSTOP, PRE_P, CHANGE)		\
  do									\
    {									\
      TEST_INCDEC (_Bool, (VALUE), PREOP, POSTOP, (PRE_P), (CHANGE));	\
      TEST_INCDEC (char, (VALUE), PREOP, POSTOP, (PRE_P), (CHANGE));	\
      TEST_INCDEC (signed char, (VALUE), PREOP, POSTOP, (PRE_P),	\
		   (CHANGE));						\
      TEST_INCDEC (unsigned char, (VALUE), PREOP, POSTOP, (PRE_P),	\
		   (CHANGE));						\
      TEST_INCDEC (signed short, (VALUE), PREOP, POSTOP, (PRE_P),	\
		   (CHANGE));						\
      TEST_INCDEC (unsigned short, (VALUE), PREOP, POSTOP, (PRE_P),	\
		   (CHANGE));						\
      TEST_INCDEC (signed int, (VALUE), PREOP, POSTOP, (PRE_P),		\
		   (CHANGE));						\
      TEST_INCDEC (unsigned int, (VALUE), PREOP, POSTOP, (PRE_P),	\
		   (CHANGE));						\
      TEST_INCDEC (signed long, (VALUE), PREOP, POSTOP, (PRE_P),	\
		   (CHANGE));						\
      TEST_INCDEC (unsigned long, (VALUE), PREOP, POSTOP, (PRE_P),	\
		   (CHANGE));						\
      TEST_INCDEC (signed long long, (VALUE), PREOP, POSTOP, (PRE_P),	\
		   (CHANGE));						\
      TEST_INCDEC (unsigned long long, (VALUE), PREOP, POSTOP, (PRE_P), \
		   (CHANGE));						\
      TEST_INCDEC (float, (VALUE), PREOP, POSTOP, (PRE_P), (CHANGE));	\
      TEST_INCDEC (double, (VALUE), PREOP, POSTOP, (PRE_P), (CHANGE));	\
      TEST_INCDEC (long double, (VALUE), PREOP, POSTOP, (PRE_P),	\
		   (CHANGE));						\
    }									\
  while (0)

#define TEST_ALL_INCDEC_ARITH(VALUE)		\
  do						\
    {						\
      TEST_INCDEC_ARITH ((VALUE), ++, , 1, 1);	\
      TEST_INCDEC_ARITH ((VALUE), --, , 1, -1);	\
      TEST_INCDEC_ARITH ((VALUE), , ++, 0, 1);	\
      TEST_INCDEC_ARITH ((VALUE), , --, 0, -1);	\
    }						\
  while (0)

static void
test_incdec (void)
{
  TEST_ALL_INCDEC_ARITH (0);
  TEST_ALL_INCDEC_ARITH (1);
  TEST_ALL_INCDEC_ARITH (2);
  TEST_ALL_INCDEC_ARITH (-1);
  TEST_ALL_INCDEC_ARITH (1ULL << 60);
  TEST_ALL_INCDEC_ARITH (1.5);
  static int ia[2];
  TEST_INCDEC (int *, &ia[1], ++, , 1, 1);
  TEST_INCDEC (int *, &ia[1], --, , 1, -1);
  TEST_INCDEC (int *, &ia[1], , ++, 0, 1);
  TEST_INCDEC (int *, &ia[1], , --, 0, -1);
}

int
main (void)
{
  test_incdec ();
  exit (0);
}

```

All fail case for atomic test:
```
gcc.dg/atomic/c11-atomic-exec-2.c
gcc.dg/atomic/c11-atomic-exec-3.c
gcc.dg/atomic/c11-atomic-exec-4.c
gcc.dg/atomic/stdatomic-compare-exchange-3.c
```

aswaterman, 2016-11-04:
Can you verify whether it's the same as #15?  If so, this might not be our bug (how unusual).

sorear, 2016-11-04:
I very much doubt this is caused by #15.  How reducable is the test case?

Once it's minimized we can inspect the assembly and see if it's a gcc bug or a qemu bug; the latter is a real possibility IME.

kito-cheng, 2016-11-04:
I will reduce the case and put reduced case later, it's time to get home for me :P

aswaterman, 2016-11-05:
FYI, this runs correctly on Spike (with -static), so it could be a QEMU bug... I will leave this issue open for tracking purposes, but I will not pursue it further for now.

kito-cheng, 2016-11-07:
Reduced case:
```
#include <stdio.h>
#include <stdlib.h>
int main()
{
  do { static volatile _Atomic (float) a = (float) (((-1)));
  if (++ a != ((1) ? (float) ((float) (((-1))) + ((1))) : (float) (((-1))))) abort ();
  if (a != (float) ((float) (((-1))) + ((1)))) abort ();
  } while (0);

}
```

atomic_cas_value_strongsi seem is the key point.
```
$ ../rv64ima/bin/riscv64-unknown-linux-gnu-gcc atomic-test.c -O  -latomic -S -dp -o -
	.file	"atomic-test.c"
	.option nopic
	.globl	__addsf3
	.globl	__nesf2
	.text
	.align	2
	.globl	main
	.type	main, @function
main:
	add	sp,sp,-48	# 86	*adddi3/2	[length = 4]
	sd	ra,40(sp)	# 87	*movdi_64bit/4	[length = 4]
	sd	s0,32(sp)	# 88	*movdi_64bit/4	[length = 4]
	sd	s1,24(sp)	# 89	*movdi_64bit/4	[length = 4]
	fence rw,rw	# 6	mem_thread_fence_1	[length = 4]
	lui	a5,%hi(a.2394)	# 83	*movdi_64bit/2	[length = 4]
	lw	a5,%lo(a.2394)(a5)	# 7	extendsidi2/2	[length = 12]
	fence rw,rw	# 8	mem_thread_fence_1	[length = 4]
	sw	a5,12(sp)	# 9	*movsf_softfloat/3	[length = 8]
	lui	a5,%hi(.LC0)	# 11	*movdi_64bit/2	[length = 4]
	lw	s1,%lo(.LC0)(a5)	# 12	*movsf_softfloat/2	[length = 12]
	lui	s0,%hi(a.2394)	# 17	*movdi_64bit/2	[length = 4]
	j	.L3	# 100	jump	[length = 4]
.L2:
	bnez	a5,.L14	# 29	*branch_ordersi	[length = 4]
.L3:
	mv	a1,s1	# 79	*movsf_softfloat/1	[length = 4]
	lw	a0,12(sp)	# 13	*movsf_softfloat/2	[length = 8]
	call	__addsf3	# 14	call_value_internal/2	[length = 8]
	lw	a5,12(sp)	# 19	extendsidi2/2	[length = 8]
	add	a3,s0,%lo(a.2394)	# 85	*lowdi	[length = 4]
	1: lr.w.sc a4,0(a3); bne a4,a5,1f; sc.w.sc a2,a0,0(a3); bnez a2,1b; 1:	# 20	atomic_cas_value_strongsi	[length = 16]
	subw	a5,a4,a5	# 21	*subsi3	[length = 4]
	seqz	a5,a5	# 22	*seq_zero_sisi	[length = 4]
	bnez	a5,.L2	# 24	*branch_ordersi	[length = 4]
	sw	a4,12(sp)	# 25	*movsi_internal/4	[length = 8]
	j	.L2	# 103	jump	[length = 4]
.L14:
	mv	a1,zero	# 31	*movsf_softfloat/1	[length = 4]
	call	__nesf2	# 33	call_value_internal/2	[length = 8]
	bnez	a0,.L15	# 35	*branch_orderdi	[length = 4]
	fence rw,rw	# 45	mem_thread_fence_1	[length = 4]
	lui	a5,%hi(a.2394)	# 82	*movdi_64bit/2	[length = 4]
	lw	a0,%lo(a.2394)(a5)	# 46	extendsidi2/2	[length = 12]
	fence rw,rw	# 47	mem_thread_fence_1	[length = 4]
	mv	a1,zero	# 48	*movsf_softfloat/1	[length = 4]
	call	__nesf2	# 50	call_value_internal/2	[length = 8]
	bnez	a0,.L16	# 52	*branch_orderdi	[length = 4]
	li	a0,0	# 65	*movsi_internal/2	[length = 4]
	ld	ra,40(sp)	# 94	*movdi_64bit/3	[length = 4]
	ld	s0,32(sp)	# 95	*movdi_64bit/3	[length = 4]
	ld	s1,24(sp)	# 96	*movdi_64bit/3	[length = 4]
	add	sp,sp,48	# 97	*adddi3/2	[length = 4]
	jr	ra	# 98	simple_return_internal	[length = 4]
.L15:
	call	abort	# 40	call_internal/2	[length = 8]
.L16:
	call	abort	# 57	call_internal/2	[length = 8]
	.size	main, .-main
	.section	.sdata,"aw",@progbits
	.align	2
	.type	a.2394, @object
	.size	a.2394, 4
a.2394:
	.word	3212836864
	.section	.srodata.cst4,"aM",@progbits,4
	.align	2
.LC0:
	.word	1065353216
	.ident	"GCC: (GNU) 6.1.0"
```

aswaterman, 2016-11-08:
This one also runs fine with spike (RV64G, linked with -static).  So, probably a QEMU bug.  @sorear @sagark can I punt to one of you to figure it out?

sorear, 2016-11-09:
failures occur on a variety of types as long as they're 32-bits and have the high bit set.  Even shorter test case:

```
#include <unistd.h>
#include <sys/syscall.h>
void _start()
{
  _Atomic int a = -1;
  a *= 3;
  syscall(SYS_exit, a != -3);
}
```

sorear, 2016-11-09:
@aswaterman riscv-qemu#36

sorear, 2016-11-09:
@aswaterman riscv/riscv-qemu#36

kito-cheng, 2016-11-09:
@sorear All atomic fail is clear with riscv/riscv-qemu#36 for RV64*
```
gcc.dg/atomic/c11-atomic-exec-2.c
gcc.dg/atomic/c11-atomic-exec-3.c
gcc.dg/atomic/c11-atomic-exec-4.c
gcc.dg/atomic/stdatomic-compare-exchange-3.c
```

sagark, 2016-11-09:
Thanks @sorear , I've merged the fix in qemu.

closed 2016-11-11 by kito-cheng