riscvarchive/riscv-gcc #24: Poor codegen for argument passing structs of doubles

sorear, 2016-11-06:
Noticed while working on libffi.

```
[root@sorear6 tmp]# cat zdemo.c
struct a { double b; double c; };
struct a d(struct a e) { return e; }
[root@sorear6 tmp]# cc -o- -S -O3 zdemo.c
        .file   "zdemo.c"
        .option nopic
        .text
        .align  2
        .globl  d
        .type   d, @function
d:
        add     sp,sp,-48
        fsd     fa0,16(sp)
        ld      a5,16(sp)
        fsd     fa1,24(sp)
        sd      a5,32(sp)
        ld      a5,24(sp)
        fld     fa0,32(sp)
        sd      a5,40(sp)
        fld     fa1,40(sp)
        add     sp,sp,48
        jr      ra
        .size   d, .-d
        .ident  "GCC: (GNU) 6.1.0 20160427 (Red Hat 6.1.0-1)"
```

`d` could be reduced to just the return

aswaterman, 2016-11-07:
Thanks for reporting.  I do not plan to handle this anytime soon, but a fix would be great.

kito-cheng, 2016-11-07:
RV64IMA get poor code gen too:
```
$ riscv64-unknown-linux-gnu-gcc zdemo.c -S -O -dp -o - -march=rv64ima
	.file	"zdemo.c"
	.option nopic
	.text
	.align	2
	.globl	d
	.type	d, @function
d:
	add	sp,sp,-48	# 24	*adddi3/2	[length = 4]
	sd	a0,32(sp)	# 8	*movdi_64bit/4	[length = 4]
	sd	a1,40(sp)	# 10	*movdi_64bit/4	[length = 4]
	ld	a0,32(sp)	# 18	*movdf_softfloat/2	[length = 4]
	ld	a1,40(sp)	# 19	*movdf_softfloat/2	[length = 4]
	add	sp,sp,48	# 29	*adddi3/2	[length = 4]
	jr	ra	# 30	simple_return_internal	[length = 4]
	.size	d, .-d
	.ident	"GCC: (GNU) 6.1.0"
```

And can be fix by remove movdf_softfloat pattern:
```
$ riscv64-unknown-linux-gnu-gcc zdemo.c -S -O -dap -o -  -march=rv64ima
	.file	"zdemo.c"
	.option nopic
	.text
	.align	2
	.globl	d
	.type	d, @function
d:
	add	sp,sp,-48	# 28	*adddi3/2	[length = 4]
	add	sp,sp,48	# 33	*adddi3/2	[length = 4]
	jr	ra	# 34	simple_return_internal	[length = 4]
	.size	d, .-d
	.ident	"GCC: (GNU) 6.1.0"
```

However I don't have benchmarking it for RISV-V yet, @aswaterman do you have any benchmarking mechanism for RISC-V in UCB or SiFive? Coremark, Dhrystone or MiBench?

sorear, 2016-11-07:
`struct a { char b[15]; };` also makes quite a few unnecessary instructions

aswaterman, 2016-11-08:
This shouldn't affect coremark or dhrystone due to the datatypes involved.

I'm fine with removing the movdf_softfloat pattern if it does not cause any regressions.

aswaterman, 2016-11-08:
@kito-cheng can you run regression tests without the movdf_softfloat pattern and submit a PR removing it if there are no new regressions?

kito-cheng, 2016-11-09:
@aswaterman it's ok for run tests without the movdf_softfloat pattern, but we eventually need some benchmark mechanism when tuning code gen.

aswaterman, 2017-01-05:
I think I've figured this one out.  Re-running test suite to confirm...

sorear, 2017-01-05:
Does it also address the char array example?

aswaterman, 2017-01-05:
Nope, but that one's bad on AArch64 and MIPS, too, so may not be a backend issue.

aswaterman, 2017-01-05:
https://github.com/riscv/riscv-gcc/commit/cd5c51b0e8cabe5cb723dee35c020122d7920eb0

I'll let you decide whether to close the issue, reopen one for the char[15] issue, or whatever.

sorear, 2017-01-06:
If the char[15] affects multiple backends then I'm fine with closing this.

closed 2017-01-06 by sorear