riscvarchive/riscv-gcc #26: Wrong code gen for overflow builtin function

kito-cheng, 2016-11-07:
This bug is come from wrong gcc code gen, not wrong back-end porting.
Here is the fix https://github.com/kito-cheng/riscv-gcc/commit/ea72a129b813802985c23b09e500d8069de3f3b1, we will send patch to gcc upstream in next days.

Option: -O0
Arch: RV64I and RV64G

How to reproduce:
```
$ riscv64-unknown-linux-gnu-gcc test.c
$ qemu-riscv64 --L `riscv64-unknown-linux-gnu-gcc --print-sysroot` a.out
```

Code for test.c
```
int v;

__attribute__((noinline, noclone)) void
bar (void)
{
  v++;
}

__attribute__((noinline, noclone)) signed int t115_1mul (unsigned int x, signed int y) {
  signed int r;
  if (__builtin_mul_overflow (x, y, &r))
    bar ();
  return r;
}
__attribute__((noinline, noclone)) signed int t115_2mul (signed int y) {
  unsigned int x = ((unsigned int) 0x7fffffff + 1); signed int r;
  if (__builtin_mul_overflow (x, y, &r))
    bar ();
  return r;
}
__attribute__((noinline, noclone)) signed int t115_3mul (signed int y) {
  signed int r;
  if (__builtin_mul_overflow ((unsigned int) ((unsigned int) 0x7fffffff + 1), y, &r))
    bar ();
  return r;
}
__attribute__((noinline, noclone)) signed int t115_4mul (unsigned int x) {
  signed int y = (-1);
  signed int r;
  if (__builtin_mul_overflow (x, y, &r))
    bar ();
  return r;
}
__attribute__((noinline, noclone)) signed int t115_5mul (unsigned int x) {
  signed int r;
  if (__builtin_mul_overflow (x, (signed int) (-1), &r))
    bar ();
  return r;
}

void t115mul (void) {
  unsigned int x = ((unsigned int) 0x7fffffff + 1);
  signed int y = (-1);
  signed int r1, r2; v = 0;
  if (t115_1mul (x, y) != (signed int) ((-0x7fffffff - 1))
      || t115_2mul (y) != (signed int) ((-0x7fffffff - 1))
      || t115_3mul (y) != (signed int) ((-0x7fffffff - 1))
      || t115_4mul (x) != (signed int) ((-0x7fffffff - 1))
      || t115_5mul (x) != (signed int) ((-0x7fffffff - 1)))
     __builtin_abort ();
  if (__builtin_mul_overflow (x, y, &r1))
    bar ();
  if (r1 != (signed int) ((-0x7fffffff - 1)))
     __builtin_abort ();
  if (__builtin_mul_overflow ((unsigned int) ((unsigned int) 0x7fffffff + 1),
                              (signed int) (-1),
                              &r2))
     bar ();
  if (r2 != (signed int) ((-0x7fffffff - 1)) || v != 7 * 0)
    __builtin_abort ();
}
int main()
{
  t115mul();
}
```

Original test case:
c-c++-common/torture/builtin-arith-overflow-12.c

aswaterman, 2016-11-08:
Nice job finding a mainline bug.  I will take no further action on this one.

aswaterman, 2017-01-05:
Did you ever send the patch upstream?

kito-cheng, 2017-01-06:
I have no time to setup a mips64 environment for running gcc testsuite past weeks,
I will get high priority for this in next week.

aswaterman, 2017-01-06:
cool

kito-cheng, 2017-01-13:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=79079

kito-cheng, 2017-01-18:
Fixed on trunk https://gcc.gnu.org/viewcvs/gcc?view=revision&revision=244539 :)

aswaterman, 2017-01-18:
Nice :)

closed 2017-01-18 by aswaterman