riscvarchive/riscv-gcc #27: Compile fail

kito-cheng, 2016-11-07:
This fail is come from C front-end due to RV64's TRULY_NOOP_TRUNCATION, and mips64el-elf toolchain will also fail for this case, so I think we should report to upstream.

Arch: RV64I and RV64G

How to reproduce:
```
$ riscv64-unknown-linux-gnu-gcc pr66178.c 
pr66178.c: In function 'test':
pr66178.c:3:21: error: initializer element is not computable at load time
     static int a =  ((char *)&&l1-(char *)&&l2)-1;

```

Code for pr66178.c:
```
int test(void)
{
    static int a =  ((char *)&&l1-(char *)&&l2)-1;
l1:
l2:
    return a;
}

int test2(void)
{
    static int a =  ((char *)&&l2-(char *)&&l3)+((char *)&&l1-(char *)&&l2);
l1:
l2:
l3:
    return a;
}
```

Note:

How to build a simple mips64el-elf gcc:
```
$ <path-to>/gcc/configure --prefix=<PREFIX> --with-abi=64 --target=mips64el-elf
$ make all-gcc
$ make install-gcc
$ mips64el-elf-gcc  pr66178.c 
pr66178.c: In function ‘test’:
pr66178.c:3:21: error: initializer element is not computable at load time
     static int a =  ((char *)&&l1-(char *)&&l2)-1;
                     ^
```

kito-cheng, 2016-11-07:
Bugzilla for gcc:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=78230

aswaterman, 2016-11-08:
Yeah, this should fail for anything with !TRULY_NOOP_TRUNCATION(32, 64).  Do you think the test should simply be changed to use uintptr_t instead of int?

kito-cheng, 2016-11-09:
I see the original bugzilla for the testcase that is about middle-end and back-end so I think send a patch to fix is ok.
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=66178

kito-cheng, 2016-11-09:
Patch:
https://gcc.gnu.org/ml/gcc-patches/2016-11/msg00782.html

kito-cheng, 2016-11-24:
Fixed at trunk, https://gcc.gnu.org/viewcvs/gcc?view=revision&revision=242792

However I think this issue do not close yet until we upstream done.

aswaterman, 2016-11-24:
Agreed.

aswaterman, 2017-01-05:
Closing, since it's fixed on riscv-next branch.

closed 2017-01-05 by aswaterman