riscvarchive/riscv-gcc #30: ICE: Compile __builtin_unwind_init with RV64IMAF

kito-cheng, 2016-11-13:
Arch: RV64IMAF
Option: -fno-dwarf2-cfi-asm x.c -g -march=rv64imaf

How to reproduce:
```
$ riscv64-unknown-linux-gnu-gcc  -fno-dwarf2-cfi-asm x.c  -g  -march=rv64imaf
x.c:6:1: internal compiler error: in div_data_align, at dwarf2cfi.c:387
 }
 ^
0x52d709 div_data_align
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/dwarf2cfi.c:387
0x71df8c div_data_align
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/dwarf2cfi.c:3021
0x71df8c output_cfi(dw_cfi_node*, dw_fde_node*, int)
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/dwarf2cfi.c:3080
0x7284eb output_fde
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/dwarf2out.c:687
0x7284eb output_call_frame_info
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/dwarf2out.c:946
0x72ec56 dwarf2out_frame_finish()
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/dwarf2out.c:1179
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://gcc.gnu.org/bugs.html> for instructions.
```

Code for x.c:
```
void foo()
{
  __builtin_unwind_init ();
}
```

kito-cheng, 2016-11-13:
This bug block building RV64IMAF toolchain since it's occur when building libgcc

aswaterman, 2016-11-13:
5386bea69fb7ae14ad1234266f9f8aba96641755 and https://github.com/riscv/riscv-binutils-gdb/commit/b899e84bcbddde93aa76c3d582e02b203ecf3fd0 should fix this.  (MIPS has the same issue.)  Hopefully it doesn't break RV64G.

closed 2016-11-13 by aswaterman