riscvarchive/riscv-gcc #31: Atomic/fenv exceptions fail

kito-cheng, 2016-11-14:
I guess this is relate to sfp-machine.h since we add the FP_INIT_ROUNDMODE and FP_HANDLE_EXCEPTIONS stub.

Arch: RV64IMAF and RV64G
Option: -latomic -lpthread -lm -o c11-atomic-exec-5.rv64i -std=c11 -pedantic-errors -pthread -U_POSIX_C_SOURCE -D_POSIX_C_SOURCE=200809L -O

How to reproduce:
```
$ riscv64-unknown-linux-gnu-gcc c11-atomic-exec-5.c -latomic -lpthread -lm -o c11-atomic-exec-5.rv64g -std=c11 -pedantic-errors -pthread -U_POSIX_C_SOURCE -D_POSIX_C_SOURCE=200809L -O
$ qemu-riscv64 -L `./rv64imafd/bin/riscv64-unknown-linux-gnu-gcc --print-sysroot` c11-atomic-exec-5.rv64g
float_add_invalid (a) 4323 pass, 694 fail; (b) 4983 pass, 0 fail
float_add_invalid_prev (a) 4208 pass, 765 fail; (b) 5027 pass, 0 fail
float_add_overflow (a) 0 pass, 5128 fail; (b) 4872 pass, 0 fail
float_add_overflow_prev (a) 0 pass, 4994 fail; (b) 5006 pass, 0 fail
float_add_overflow_double (a) 0 pass, 5039 fail; (b) 4961 pass, 0 fail
float_add_overflow_long_double (a) 4973 pass, 0 fail; (b) 4449 pass, 578 fail
float_add_inexact (a) 5009 pass, 0 fail; (b) 3967 pass, 1024 fail
float_add_inexact_int (a) 5068 pass, 0 fail; (b) 4242 pass, 690 fail
float_preinc_inexact (a) 5064 pass, 0 fail; (b) 4269 pass, 667 fail
float_postinc_inexact (a) 5156 pass, 0 fail; (b) 4275 pass, 569 fail
long_add_float_inexact (a) 4914 pass, 0 fail; (b) 3745 pass, 1341 fail
complex_float_add_overflow (a) 0 pass, 4983 fail; (b) 5017 pass, 0 fail
float_sub_invalid (a) 3968 pass, 1006 fail; (b) 5026 pass, 0 fail
float_sub_overflow (a) 0 pass, 5157 fail; (b) 4843 pass, 0 fail
float_sub_inexact (a) 5019 pass, 0 fail; (b) 4301 pass, 680 fail
float_sub_inexact_int (a) 4999 pass, 0 fail; (b) 3936 pass, 1065 fail
float_predec_inexact (a) 5006 pass, 0 fail; (b) 4252 pass, 742 fail
float_postdec_inexact (a) 4990 pass, 0 fail; (b) 3994 pass, 1016 fail
long_sub_float_inexact (a) 5022 pass, 0 fail; (b) 3971 pass, 1007 fail
complex_float_sub_overflow (a) 0 pass, 5035 fail; (b) 4965 pass, 0 fail
float_mul_invalid (a) 4352 pass, 654 fail; (b) 4994 pass, 0 fail
float_mul_overflow (a) 0 pass, 5029 fail; (b) 4971 pass, 0 fail
float_mul_underflow (a) 0 pass, 4999 fail; (b) 5001 pass, 0 fail
float_mul_inexact (a) 4967 pass, 0 fail; (b) 4195 pass, 838 fail
float_mul_inexact_int (a) 4961 pass, 0 fail; (b) 4213 pass, 826 fail
long_mul_float_inexact (a) 4835 pass, 0 fail; (b) 4088 pass, 1077 fail
complex_float_mul_overflow (a) 0 pass, 4927 fail; (b) 5073 pass, 0 fail
float_div_invalid_divbyzero (a) 4248 pass, 771 fail; (b) 4265 pass, 716 fail
float_div_overflow (a) 0 pass, 5060 fail; (b) 4940 pass, 0 fail
float_div_underflow (a) 0 pass, 5008 fail; (b) 4992 pass, 0 fail
float_div_inexact (a) 4904 pass, 0 fail; (b) 4217 pass, 879 fail
float_div_inexact_int (a) 5036 pass, 0 fail; (b) 4225 pass, 739 fail
int_div_float_inexact (a) 5002 pass, 0 fail; (b) 4248 pass, 750 fail
complex_float_div_overflow (a) 0 pass, 4966 fail; (b) 5034 pass, 0 fail
double_add_invalid (a) 4295 pass, 817 fail; (b) 4888 pass, 0 fail
double_add_overflow (a) 0 pass, 5063 fail; (b) 4937 pass, 0 fail
double_add_overflow_long_double (a) 4946 pass, 0 fail; (b) 3949 pass, 1105 fail
double_add_inexact (a) 5024 pass, 0 fail; (b) 4188 pass, 788 fail
double_add_inexact_int (a) 5025 pass, 0 fail; (b) 4213 pass, 762 fail
double_preinc_inexact (a) 4973 pass, 0 fail; (b) 4409 pass, 618 fail
double_postinc_inexact (a) 4989 pass, 0 fail; (b) 4259 pass, 752 fail
long_long_add_double_inexact (a) 5013 pass, 0 fail; (b) 3942 pass, 1045 fail
complex_double_add_overflow (a) 0 pass, 4874 fail; (b) 5126 pass, 0 fail
double_sub_invalid (a) 4082 pass, 960 fail; (b) 4958 pass, 0 fail
double_sub_overflow (a) 0 pass, 5000 fail; (b) 5000 pass, 0 fail
double_sub_inexact (a) 5090 pass, 0 fail; (b) 3943 pass, 967 fail
double_sub_inexact_int (a) 4969 pass, 0 fail; (b) 4054 pass, 977 fail
double_predec_inexact (a) 4955 pass, 0 fail; (b) 3790 pass, 1255 fail
double_postdec_inexact (a) 5009 pass, 0 fail; (b) 3955 pass, 1036 fail
long_long_sub_double_inexact (a) 5045 pass, 0 fail; (b) 3904 pass, 1051 fail
complex_double_sub_overflow (a) 0 pass, 5026 fail; (b) 4974 pass, 0 fail
double_mul_invalid (a) 4329 pass, 713 fail; (b) 4958 pass, 0 fail
double_mul_overflow (a) 0 pass, 5064 fail; (b) 4936 pass, 0 fail
double_mul_overflow_float (a) 0 pass, 4965 fail; (b) 5035 pass, 0 fail
double_mul_underflow (a) 0 pass, 4968 fail; (b) 5032 pass, 0 fail
double_mul_inexact (a) 5075 pass, 0 fail; (b) 4219 pass, 706 fail
double_mul_inexact_int (a) 4962 pass, 0 fail; (b) 4297 pass, 741 fail
long_long_mul_double_inexact (a) 4975 pass, 0 fail; (b) 4339 pass, 686 fail
complex_double_mul_overflow (a) 0 pass, 4954 fail; (b) 5046 pass, 0 fail
double_div_invalid_divbyzero (a) 4405 pass, 759 fail; (b) 4260 pass, 576 fail
double_div_overflow (a) 0 pass, 5013 fail; (b) 4987 pass, 0 fail
[... 564 lines cut ...]
  ret |= test_main_complex_long_double_sub_overflow ();
#endif
  ret |= test_main_long_double_mul_invalid ();
  ret |= test_main_long_double_mul_overflow ();
  ret |= test_main_long_double_mul_overflow_float ();
  ret |= test_main_long_double_mul_overflow_double ();
  ret |= test_main_long_double_mul_underflow ();
#if LDBL_MANT_DIG != 106
  ret |= test_main_long_double_mul_inexact ();
  ret |= test_main_long_double_mul_inexact_int ();
#endif
  ret |= test_main_complex_long_double_mul_overflow ();
  ret |= test_main_long_double_div_invalid_divbyzero ();
  ret |= test_main_long_double_div_overflow ();
  ret |= test_main_long_double_div_underflow ();
  ret |= test_main_long_double_div_inexact ();
  ret |= test_main_long_double_div_inexact_int ();
  ret |= test_main_int_div_long_double_inexact ();
  ret |= test_main_complex_long_double_div_overflow ();
  if (ret != 0)
    abort ();
  else
    exit (0);
}
```

kito-cheng, 2016-11-14:
btw, today is my meeting day, so the reduced case I will put later.

aswaterman, 2016-11-14:
I'll leave this one to you, unless you find something that's my fault : P

kito-cheng, 2016-11-17:
I don't know what's happen yet, it's only fail when multi-thread, and https://github.com/riscv/riscv-gcc/pull/32 doesn't make this case pass :(

Option:  -O -latomic -lpthread -lm -std=c11

Reduced case:
```
enum
  {
    FE_INEXACT = 0x01,
    FE_UNDERFLOW = 0x02,
    FE_OVERFLOW = 0x04,
    FE_DIVBYZERO = 0x08,
    FE_INVALID = 0x10,
  };
enum
  {
    FE_TONEAREST = 0x0,
    FE_TOWARDZERO = 0x1,
    FE_DOWNWARD = 0x2,
    FE_UPWARD = 0x3
  };
extern int fetestexcept (int __excepts) __attribute__ ((__nothrow__ , __leaf__));

union pthread_attr_t
{
  char __size[56];
  long int __align;
};
typedef union pthread_attr_t pthread_attr_t;
typedef unsigned long int pthread_t;
extern int pthread_create (pthread_t *__restrict __newthread,
      const pthread_attr_t *__restrict __attr,
      void *(*__start_routine) (void *),
      void *__restrict __arg) __attribute__ ((__nothrow__)) __attribute__ ((__nonnull__ (1, 3)));
extern int pthread_join (pthread_t __th, void **__thread_return);
extern void abort (void) __attribute__ ((__nothrow__ , __leaf__)) __attribute__ ((__noreturn__));
extern void exit (int __status) __attribute__ ((__nothrow__ , __leaf__)) __attribute__ ((__noreturn__));
extern int printf (const char *__restrict __format, ...);
extern int feclearexcept (int __excepts) __attribute__ ((__nothrow__ , __leaf__));
extern int feraiseexcept (int __excepts) __attribute__ ((__nothrow__ , __leaf__));

static volatile _Atomic _Bool thread_ready, thread_stop;
static volatile _Atomic float var_float_add_invalid;

static void * test_thread_float_add_invalid (void *arg) {
  thread_ready = 1;
  while (!thread_stop) {
    var_float_add_invalid = (0);
    var_float_add_invalid = (-__builtin_inff ());
  }
  return ((void *)0);
}

#define MULTI_THREAD

static int test_main_float_add_invalid (void) {
  thread_stop = 0;
  thread_ready = 0;
  var_float_add_invalid = (0);
  pthread_t thread_id;
#ifdef MULTI_THREAD
  int pret = pthread_create (&thread_id, ((void *)0),
                             test_thread_float_add_invalid, ((void *)0));
  if (pret != 0) {
    printf ("pthread_create failed: %d\n", pret);
    return 1;
  }
  while (!thread_ready) ;
#endif
  int num_1_pass = 0, num_1_fail = 0, num_2_pass = 0, num_2_fail = 0;
  for (int i = 0; i < 10000; i++) {
#ifndef MULTI_THREAD
    if (i & 0x1)
      var_float_add_invalid = (0);
    else
      var_float_add_invalid = (-__builtin_inff ());
#endif
    feclearexcept ((FE_INEXACT | FE_DIVBYZERO
                    | FE_UNDERFLOW | FE_OVERFLOW | FE_INVALID));
    feraiseexcept (0);
    float r = ( var_float_add_invalid += __builtin_inff ());
    int rexc = fetestexcept ((FE_DIVBYZERO | FE_INEXACT
                             | FE_INVALID | FE_OVERFLOW | FE_UNDERFLOW));
    if (__builtin_isinf (r)) {
      if (rexc == ((0) | (0)))
        num_1_pass++;
      else
        num_1_fail++;
      var_float_add_invalid = (-__builtin_inff ());
    } else {
      if (rexc == ((0) | (FE_INVALID)))
        num_2_pass++;
      else
        num_2_fail++;
     var_float_add_invalid = (0);
    }
  }
  thread_stop = 1;
#ifdef MULTI_THREAD
  pthread_join (thread_id, ((void *)0));
#endif
  printf ("float_add_invalid" " (a) %d pass, %d fail; (b) %d pass, %d fail\n",
          num_1_pass, num_1_fail,
          num_2_pass, num_2_fail);
  return num_1_fail || num_2_fail;
}

int
main (void)
{
  int ret = 0;
  ret |= test_main_float_add_invalid ();
  if (ret != 0)
    abort ();
  else
    exit (0);
}

```

aswaterman, 2016-11-17:
Ugh. Leave this alone until someone actually cares about FP exceptions? :-)

jim-wilson, 2018-01-12:
This was a qemu FP bug that I helped fix.
    https://github.com/riscv/riscv-qemu/issues/64

closed 2018-01-12 by jim-wilson