riscvarchive/riscv-gcc #33: ICE: Crash when compile function have V4SI argument for rv32

kito-cheng, 2016-11-24:
GCC: https://github.com/riscv/riscv-gcc/commit/de6438d70df93120ab631ac76106d6ed05fc6c1c
Arch: All RV32 family
Option: O2

How to reproduce:
```
$ riscv32-unknown-linux-gnu-gcc vector-compare-3.c  -O2
vector-compare-3.c: In function 'g':
vector-compare-3.c:20:1: internal compiler error: in max_value, at wide-int.cc:317
 g (v4i *x, v4i const *y, v4i *z, v4i *t)
 ^
0xd7b7c6 wi::max_value(unsigned int, signop)
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/wide-int.cc:317
0xd19640 integer_all_onesp(tree_node const*)
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/tree.c:2410
0xd195f9 integer_all_onesp(tree_node const*)
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/tree.c:2402
0xe3c2e7 generic_simplify_VEC_COND_EXPR
	/home/kito/riscv-workspace/rv32ima/build/build-gcc-linux-stage2/gcc/generic-match.c:26211
0xe3c2e7 generic_simplify(unsigned int, tree_code, tree_node*, tree_node*, tree_node*, tree_node*)
	/home/kito/riscv-workspace/rv32ima/build/build-gcc-linux-stage2/gcc/generic-match.c:26399
0x7c773c fold_ternary_loc(unsigned int, tree_code, tree_node*, tree_node*, tree_node*, tree_node*)
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/fold-const.c:11482
0xd8d42a gimple_resimplify3(gimple**, code_helper*, tree_node*, tree_node**, tree_node* (*)(tree_node*))
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/gimple-match-head.c:189
0xdff2a7 gimple_simplify(gimple*, code_helper*, tree_node**, gimple**, tree_node* (*)(tree_node*), tree_node* (*)(tree_node*))
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/gimple-match-head.c:686
0x8294b1 fold_stmt_1
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/gimple-fold.c:3649
0xbbf562 execute
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/tree-ssa-forwprop.c:2309
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://gcc.gnu.org/bugs.html> for instructions.
```

Code for vector-compare-3.c
```
/* { dg-do compile } */
/* { dg-options "-O2" } */

typedef int v4i __attribute__((vector_size(4*sizeof(int))));

// fold should not turn (vec_other)(x<y) into (x<y)?vec_other(-1):vec_other(0).

void use (v4i const *z);

void
f (v4i *x, v4i *y)
{
  v4i const zz = *x < *y;
  use (&zz);
}

// Optimizations shouldn't introduce a boolean type in there

void
g (v4i *x, v4i const *y, v4i *z, v4i *t)
{
  *z = *x < *y | *x == *y;
  *t = *x < *y & *x > *y;
}
```

aswaterman, 2016-11-24:
I'm confused. Why should v4si be supported on RISC-V?

kito-cheng, 2016-11-24:
Those type can be lower by gcc automatically when target not native support such type.

kito-cheng, 2016-11-24:
However the crash point is really strange...

aswaterman, 2016-11-24:
I think I know the cause; I will look into it tonight.

aswaterman, 2016-11-24:
OK.  I'm reverting half my commit (but the x30 thing still matters).  Sorry for not noticing this regression earlier.

commit b530720a (2016-11-24): Support more 128-bit modes for RV32

...even if they don't make much sense.  Resolves #33.
closed 2016-11-24 by aswaterman