riscvarchive/riscv-gcc #41: soft divsi3 will hang when dividend is -2147483648(0x80000000)

kito-cheng, 2016-12-22:
Option: -march=rv32i or -march=rv64i (No M extension)

How to reproduce:
```
$ gcc div.c
$ ./a.out
-2147483648(0x80000000) 16777216(0x01000000) -128(0xffffff80)
$ riscv32-unknown-linux-gnu-gcc div.c -static
$ qemu-riscv32 a.out
// Hang forever
```

Code for div.c:
```
#include <stdio.h>

int div(int a, int b) __attribute__((noinline, noclone));
int div(int a, int b)
{
  return a / b;
}

int main()
{
  int a = 0x80000000;
  int b = 0x1000000;
  printf("%d(0x%08x) %d(0x%08x) %d(0x%08x)\n", a, a, b, b, div(a, b), div(a, b));
  return 0;
}
```

Key point I think is there:
```
  /* Handle negative arguments to __divdi3.  */
.L10:
  neg   a0, a0  // neg 0x80000000 = 0x80000000
```

aswaterman, 2016-12-22:
That's a little embarrassing.

aswaterman, 2016-12-22:
See 287c5cb3d8ba9af04160c502ec5ec90bf9d4c0af -- the algorithm was correct, but instead of calling __udiv, I accidentally recursed into __div...

Thanks for finding it.  Re-running tests now.

kito-cheng, 2016-12-23:
It's work for me, thanks :)

closed 2016-12-23 by kito-cheng

aswaterman, 2016-12-23:
Merry Christmas @kito-cheng !