riscvarchive/riscv-gcc #42: internal compiler error: Segmentation fault when compiling Contiki

epall, 2017-01-03:
I seem to have stumbled into a peculiar riscv-gcc compiler bug that's manifesting as a segfault. I'm working on porting Contiki to the SiFive FE310, and compiling the network stack triggers the bug.

Steps to reproduce:

    git clone https://github.com/osdomotics/osd-contiki.git
    cd osd-contiki/
    touch icosoc.h
    riscv32-unknown-elf-gcc -Icore -Iplatform/pico-rv32-icoboard -Icpu/pico-rv32 -I. -c core/net/ipv4/uip.c -Os -g

Error message:
```
core/net/ipv4/uip.c: In function 'uip_connect':
core/net/ipv4/uip.c:457:1: internal compiler error: Segmentation fault
}
^
0xa8b24f crash_signal
        /ssd/freedom-e-sdk/work/riscv32-gnu-toolchain/src/newlib-gcc/gcc/toplev.c:333
0xe984e0 try_combine
        /ssd/freedom-e-sdk/work/riscv32-gnu-toolchain/src/newlib-gcc/gcc/combine.c:4045
0xe9d751 combine_instructions
        /ssd/freedom-e-sdk/work/riscv32-gnu-toolchain/src/newlib-gcc/gcc/combine.c:1288
0xe9d751 rest_of_handle_combine
        /ssd/freedom-e-sdk/work/riscv32-gnu-toolchain/src/newlib-gcc/gcc/combine.c:14350
0xe9d751 execute
        /ssd/freedom-e-sdk/work/riscv32-gnu-toolchain/src/newlib-gcc/gcc/combine.c:14393
```

Here's where it gets fun. `-Os` without `-g` doesn't segfault. Neither does `-O2 -g`. Even more fun, adding the specific optimization flags to transform -O2 into -Os (`finline-functions -fno-optimize-strlen -freorder-blocks-algorithm=simple -fno-schedule-insns -g`) results in a successful compile.

I am way out of my depth debugging this. Is there any other information I can provide to help?

GCC was compiled from riscv-gcc@3e7b179f8b6.

```
$ riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=/ssd/freedom-e-sdk/toolchain/bin/riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/ssd/freedom-e-sdk/toolchain/libexec/gcc/riscv32-unknown-elf/6.1.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /ssd/freedom-e-sdk/work/riscv32-gnu-toolchain/src/newlib-gcc/configure --target=riscv32-unknown-elf --prefix=/ssd/freedom-e-sdk/toolchain --disable-shared --disable-threads --disable-tls --enable-languages=c,c++ --with-system-zlib --with-newlib --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --enable-checking=yes --disable-multilib --with-abi=ilp32 --with-arch=rv32ima
Thread model: single
gcc version 6.1.0 (GCC)
```

kito-cheng, 2017-01-03:
Can you provide .i file, it's can create by --save-temps since I can't compile it successfully due to lack of header files.

```
$ pwd
/home/kito/riscv-workspace/osd-contiki
$ ../rv32ima/bin/riscv32-unknown-linux-gnu-gcc  -Icore -Iplatform/pico-rv32-icoboard -Icpu/pico-rv32 -I. -c core/net/ipv4/uip.c -Os -g
In file included from core/net/ip/uip.h:86:0,
                 from core/net/ipv4/uip.c:73:
core/net/ip/uipopt.h:74:26: fatal error: contiki-conf.h: No such file or directory
 #include "contiki-conf.h"
                          ^
compilation terminated.
```

epall, 2017-01-03:
Whoops! I forgot the `git checkout ico` step to get to the right branch. That should give you a contiki-conf.h file.

kito-cheng, 2017-01-03:
OK, I can reproduce it now.

kito-cheng, 2017-01-03:
Arch: rv32ima (rv64ima is ok)
Option: -Os -g -march=rv32ima -mabi=ilp32

How to reproduce:
```
$ riscv32-unknown-linux-gnu-gcc uip.E.c -Os -g -march=rv32ima -mabi=ilp32
uip.E.c: In function 'uip_connect':
uip.E.c:146:1: internal compiler error: Segmentation fault
 }
 ^
0xdaf1e6 crash_signal
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/toplev.c:333
0x12ee360 try_combine
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/combine.c:4045
0x12e64fc combine_instructions
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/combine.c:1288
0x130a49c rest_of_handle_combine
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/combine.c:14350
0x130a544 execute
	/home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-gcc/gcc/combine.c:14393
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://gcc.gnu.org/bugs.html> for instructions.
```

Reduced case, code for uip.E.c:
```
typedef unsigned char uint8_t;
typedef unsigned short uint16_t;
typedef unsigned long uint32_t;

typedef union uip_ip4addr_t {
  uint8_t u8[4];
  uint16_t u16[2];
} uip_ip4addr_t;

typedef uip_ip4addr_t uip_ipaddr_t;

typedef union {
  uint32_t u32[((1280 + (0)) + 3) / 4];
  uint8_t u8[(1280 + (0))];
} uip_buf_t;

extern uip_buf_t uip_aligned_buf;
uint16_t uip_htons(uint16_t val);

struct uip_conn {
  uip_ipaddr_t ripaddr;
  uint16_t lport;
  uint16_t rport;
  uint8_t rcv_nxt[4];
  uint8_t snd_nxt[4];
  uint16_t len;
  uint16_t mss;
  uint16_t initialmss;
  uint8_t sa;
  uint8_t sv;
  uint8_t rto;
  uint8_t tcpstateflags;
  uint8_t timer;
  uint8_t nrtx;
};

extern struct uip_conn uip_conns[(4)];

struct uip_conn *uip_conn;
uint16_t uip_listenports[(4)];

static uint8_t iss[4];
static uint16_t lastport;
uint8_t uip_acc32[4];
static uint8_t c ;

static uint16_t
chksum(uint16_t sum, const uint8_t *data, uint16_t len)
{
  uint16_t t;
  const uint8_t *dataptr;
  const uint8_t *last_byte;

  dataptr = data;

  while(dataptr < last_byte) {
    t = (dataptr[0] << 8) + dataptr[1];
    if(sum < t) {
      sum++;
    }
    dataptr += 2;
  }
  return sum;
}

uint16_t
uip_ipchksum(void)
{
  uint16_t sum;

  sum = chksum(0, &(uip_aligned_buf.u8)[(0)], 20);
  return (sum == 0) ? 0xffff : uip_htons(sum);
}

void
uip_init(void)
{
  for(c = 0; c < (4); ++c) {
    uip_listenports[c] = 0;
  }
}

struct uip_conn *
uip_connect(const uip_ipaddr_t *ripaddr, uint16_t rport)
{
  register struct uip_conn *conn, *cconn;
 again:
  ++lastport;

  if(lastport >= 32000) {
    lastport = 4096;
  }

  for(c = 0; c < (4); ++c) {
    conn = &uip_conns[c];
    if(conn->tcpstateflags != 0 &&
       conn->lport == uip_htons(lastport)) {
      goto again;
    }
  }

  conn = 0;
  for(c = 0; c < (4); ++c) {
    cconn = &uip_conns[c];
    if(cconn->tcpstateflags == 0) {
      conn = cconn;
      break;
    }
    if(cconn->tcpstateflags == 7) {
      if(conn == 0 ||
  cconn->timer > conn->timer) {
 conn = cconn;
      }
    }
  }

  conn->tcpstateflags = 2;

  conn->snd_nxt[0] = iss[0];
  conn->snd_nxt[1] = iss[1];
  conn->snd_nxt[2] = iss[2];
  conn->snd_nxt[3] = iss[3];

  conn->rcv_nxt[0] = 0;
  conn->rcv_nxt[1] = 0;
  conn->rcv_nxt[2] = 0;
  conn->rcv_nxt[3] = 0;

  conn->initialmss = conn->mss = ((1280 + (0)) - (0) - (20 + 20));

  conn->len = 1;
  conn->nrtx = 0;
  conn->timer = 1;
  conn->rto = 3;
  conn->sa = 0;
  conn->sv = 16;
  conn->lport = uip_htons(lastport);
  conn->rport = rport;
  (*(&conn->ripaddr) = *(ripaddr));

  return conn;
}

void
uip_process(uint8_t flag)
{
  if(++iss[3] == 0) {
  }
}

uint16_t
uip_htons(uint16_t val)
{
  return (uint16_t)((((uint16_t) (val)) << 8) | (((uint16_t) (val)) >> 8));
}
```

aswaterman, 2017-01-05:
Reduced further, but no insight yet:

```
typedef unsigned char uint8_t;
typedef unsigned short uint16_t;
typedef unsigned long uint32_t;

typedef union uip_ip4addr_t {
  uint8_t u8[4];
  uint16_t u16[2];
} uip_ip4addr_t;

typedef uip_ip4addr_t uip_ipaddr_t;

struct uip_conn {
  uip_ipaddr_t ripaddr;
  uint16_t lport;
  uint8_t snd_nxt[4];
  uint16_t len;
  uint16_t mss;
  uint16_t initialmss;
  uint8_t sv;
  uint8_t rto;
  uint8_t tcpstateflags;
  uint8_t timer;
};

extern struct uip_conn uip_conns[(4)];

extern uint16_t uip_listenports[(4)];

extern uint8_t iss[4];
extern uint16_t lastport;
extern uint8_t c ;

uint16_t chksum()
{
  const uint8_t *dataptr;
  const uint8_t *last_byte;

  while(dataptr < last_byte) {
    if(*dataptr)
      return 1;
    dataptr++;
  }
  return 0;
}

void
uip_init(void)
{
  for(c = 0; c < (4); ++c) {
    uip_listenports[c] = 0;
  }
}

static uint16_t
uip_htons(uint16_t val)
{
  return (uint16_t)((((uint16_t) (val)) << 8) | (((uint16_t) (val)) >> 8));
}

struct uip_conn *
uip_connect(const uip_ipaddr_t *ripaddr, uint16_t rport)
{
  register struct uip_conn *conn, *cconn;
 again:
  ++lastport;

  if(lastport >= 32000) {
    lastport = 4096;
  }

  for(c = 0; c < (4); ++c) {
    conn = &uip_conns[c];
    if(conn->tcpstateflags != 0 &&
       conn->lport == uip_htons(lastport)) {
      goto again;
    }
  }

  for(c = 0; c < (4); ++c) {
    cconn = &uip_conns[c];
    if(cconn->tcpstateflags == 0) {
      break;
    }
    if(cconn->tcpstateflags == 7) {
      if(cconn->timer > conn->timer) {
				conn = cconn;
      }
    }
  }

  conn->tcpstateflags = 2;

  conn->snd_nxt[0] = iss[0];
  conn->snd_nxt[1] = iss[1];
  conn->snd_nxt[2] = iss[2];
  conn->snd_nxt[3] = iss[3];

  conn->initialmss = conn->mss = 1;

  conn->len = 1;
  conn->timer = 1;
  conn->rto = 1;
  conn->sv = 1;
  conn->lport = uip_htons(lastport);
  (*(&conn->ripaddr) = *(ripaddr));

  return conn;
}
```

aswaterman, 2017-01-05:
Seems like this problem is gone in the gcc7 port (i.e. the riscv-next branch).  It's a little disconcerting, but it is possible that this was a bug in the middle-end that has since been fixed.  I'm going to tentatively close the issue.

closed 2017-01-05 by aswaterman