riscvarchive/riscv-gcc #49: gcc emitting “add rd, zero, rs2” pattern for mv versus spec: “addi rd, rs1, 0”.

michaeljclark, 2017-03-01:
eyeballing objdump output from a gcc build from a fortnight ago, found that the generated register moves are using “add rd, zero, rs2” pattern instead of the mv pseudo from the spec: “addi rd, rs1, 0”

```
$ ls -l `which riscv64-unknown-elf-gcc`
-rwxr-xr-x 2 mclark mclark 5181232 Feb 15 19:44 /opt/riscv/toolchain/bin/riscv64-unknown-elf-gcc

$ rv-bin dump -a linux-4.6.2/vmlinux | grep "add " | grep zero | head
           800000d0:   892a                 add            s2, zero, a0
           800000d2:   8b2e                 add            s6, zero, a1
           800000f4:   854a                 add            a0, zero, s2
           80000104:   854a                 add            a0, zero, s2
           8000010e:   855a                 add            a0, zero, s6
           80000114:   85ca                 add            a1, zero, s2
           800001b2:   892a                 add            s2, zero, a0
           800001b6:   85aa                 add            a1, zero, a0
           800001c2:   85ce                 add            a1, zero, s3
           800001c4:   8556                 add            a0, zero, s5
```

michaeljclark, 2017-03-01:
It appears to be Binutils/gas that is emitting 'add' instead of 'addi' for 'mv'

```
$ grep mv build/riscv64-unknown-elf/sample-asm/test-loop-2-rv64.s
build/riscv64-unknown-elf/sample-asm/test-loop-2-rv64.s:	mv	a1,a5
```

```
#include <stdio.h>

int arr[] = {
	1, 2, 3,
	4, 5, 6,
	7, 8, 9,
	10, 11, 12,
	13, 14, 15,
	16, 17, 18,
	19, 20, 21,
	22, 23, 24,
	0
};

int main()
{
	int total = 0;
	int *ent = arr;
	while (*ent != 0) {
		total += *ent;
		ent++;
	}
	printf("total=%d", total);
}
```

aswaterman, 2017-03-01:
OK, now that I see this is RVC, this makes sense.  The RVC instruction C.MV does indeed expand to ADD (as opposed to the RVI pseudo-op MV, which expands to addi).

So this is not a bug in GCC/binutils, but actually corresponds to an unclarity/bug in the assembly programming guide.

closed 2017-03-01 by aswaterman

michaeljclark, 2017-03-01:
Oh. My mistake. I was not considering RVC. 2-byte opcodes.

GVTK7, 2024-02-03:
So how to change add

> It appears to be Binutils/gas that is emitting 'add' instead of 'addi' for 'mv'
>
> ```
> $ grep mv build/riscv64-unknown-elf/sample-asm/test-loop-2-rv64.s [...]

Can we rectify thsi issue ????

aswaterman, 2024-02-03:
This issue was closed seven years ago, and even then, it was deemed to be a non-issue.  If you believe you have a new issue, please feel free to create one--but from what you posted, there's nothing actionable.