riscvarchive/riscv-gcc #51: riscv_hard_regno_mode_ok_p for FPU register

kito-cheng, 2017-03-07:
Matthew point out some issue in the mov[qi|hi|si|di] pattern for FPR <-> GPR in bug 79912[1], because riscv_hard_regno_mode_ok_p is say FPR is not ok for MODE_INT and here is alternative for FPR <-> GPR in our integer move pattern.

There is two solution for this problem:
1. Let FPR ok for MODE_INT
2. Remove all FPR <-> GPR alternative and fix riscv_preferred_reload_class as Matthew suggest.

Ref:
[1] https://gcc.gnu.org/bugzilla/show_bug.cgi?id=79912

kito-cheng, 2017-03-07:
I prefer option 1, but I don't run testsuite yet :P

kito-cheng, 2017-03-07:
@aswaterman

kito-cheng, 2017-03-07:
Matthew provide another view for FPR <-> GPR, it's will make performance degradation for whole system if we have lazy context switch for FPR.

aswaterman, 2017-03-07:
Don't forget "cannot change mode class". If you FLW an int32 then FSD it, you get garbage. If you FLD an int64 then FSW it, you also get garbage. Same goes for FMV. So I'm pretty sure this could break HW implementations that use an internal recoding scheme for the FP regs.

Also, I agree with the context switch argument. Don't allow ints in FPRs is my vote.

kito-cheng, 2017-03-19:
https://github.com/riscv/riscv-gcc/commit/789d5eb018809ee1ba3d2dcdb4f075f334119e5c

closed 2017-03-19 by kito-cheng