riscvarchive/riscv-gcc #52: removing branches

jurobystricky, 2017-03-07:
Is there any particular reason you remove existing branches such as "riscv-gcc-6.1.0" and move everything from the branch into another branch __archive__? 
This causes all kinds of fetch errors from bitbake recipes.

palmer-dabbelt, 2017-03-07:
We're not supporting those old GCC versions any more, there's been ABI changes and such.  I wanted to keep the branches clean on repos we've upstreamed so we don't have a bunch of old versions floating around to confuse people.

We're keeping the tags around in the __archive__ branch so they don't get lost in case someone has silicon floating around that only works with the old stuff.  If you really want the old stuff you should be able to fetch from the hashes instead, which is safer because then your stuff won't be unexpectedly updated.

Since we're upstream you could also use the GCC prereleases.

jurobystricky, 2017-03-07:
Fair enough, but the purpose of version control is that you should be able to retrieve any historical version including unsupported versions.

closed 2017-03-08 by aswaterman