riscvarchive/riscv-gcc #53: atomic_exchange_explicit converts type through the stack

terpstra, 2017-03-08:
This method:
````
  static inline void l2ctl_barrier_0() { asm volatile("fence rw, io" : : : "memory" ); }
  static inline void l2ctl_barrier_1() { asm volatile("fence io, rw" : : : "memory" ); }
  static inline uint64_t l2ctl_flip_ways(uint64_t base_addr, int client, uint64_t ways) {
    volatile uint64_t *ways_v = (uint64_t *)(base_addr + L2CTL_WAYS);
    l2ctl_barrier_0();
    uint64_t old = atomic_exchange_explicit(&ways_v[client], ways, memory_order_relaxed);
    l2ctl_barrier_1();
    return old;
  }
````

Gets inlined into:
````
    8000028e:   03c0000f                fence   rw,io
    80000292:   e03e                    sd      a5,0(sp)
    80000294:   6782                    ld      a5,0(sp)
    80000296:   08f8b7af                amoswap.d       a5,a5,(a7)
    8000029a:   e43e                    sd      a5,8(sp)
    8000029c:   6e22                    ld      t3,8(sp)
    8000029e:   0c30000f                fence   io,rw
    800002a2:   4701                    li      a4,0
    800002a4:   4781                    li      a5,0
    800002a6:   00d05f63                blez    a3,800002c4 <llc_reserve+0x104>
    800002aa:   97b2                    add     a5,a5,a2
    800002ac:   03c0000f                fence   rw,io
    800002b0:   20f5b023                sd      a5,512(a1) # 2010200 <_prog_start-0x7dfefe00>
    800002b4:   0c30000f                fence   io,rw
    800002b8:   0007c783                lbu     a5,0(a5) # a000000 <_prog_start-0x76000000>
    800002bc:   9f29                    addw    a4,a4,a0
    800002be:   87ba                    mv      a5,a4
    800002c0:   fed7c5e3                blt     a5,a3,800002aa <llc_reserve+0xea>
    800002c4:   03c0000f                fence   rw,io
    800002c8:   e872                    sd      t3,16(sp)
    800002ca:   67c2                    ld      a5,16(sp)
    800002cc:   08f8b7af                amoswap.d       a5,a5,(a7)
    800002d0:   ec3e                    sd      a5,24(sp)
    800002d2:   67e2                    ld      a5,24(sp)
    800002d4:   0c30000f                fence   io,rw
````

Changing the relevant line to:
````
  uint64_t old = atomic_exchange_explicit((_Atomic(uint64_t*))&ways_v[client], ways, memory_order_relaxed);
````

Makes the assembler output:
````
    8000020a:	03c0000f          	fence	rw,io
    8000020e:	60a7362f          	amoand.d	a2,a0,(a4)
    80000212:	0c30000f          	fence	io,rw
    80000216:	0721                	addi	a4,a4,8
    80000218:	fa6d                	bnez	a2,8000020a <llc_reserve+0x60>
    8000021a:	8f46                	mv	t5,a7
    8000021c:	07c8d663          	ble	t3,a7,80000288 <llc_reserve+0xde>
    80000220:	02d88633          	mul	a2,a7,a3
    80000224:	02011737          	lui	a4,0x2011
    80000228:	00359313          	slli	t1,a1,0x3
    8000022c:	80070713          	addi	a4,a4,-2048 # 2010800 <_prog_start-0x7dfef800>
    80000230:	0007859b          	sext.w	a1,a5
    80000234:	0a0007b7          	lui	a5,0xa000
    80000238:	933a                	add	t1,t1,a4
    8000023a:	4e85                	li	t4,1
    8000023c:	02010537          	lui	a0,0x2010
    80000240:	963e                	add	a2,a2,a5
    80000242:	03c0000f          	fence	rw,io
    80000246:	011e9833          	sll	a6,t4,a7
    8000024a:	0903382f          	amoswap.d	a6,a6,(t1)
    8000024e:	0c30000f          	fence	io,rw
````

Either it should be an error to not use an _Atomic type or the conversion should not round-trip through memory.

aswaterman, 2017-03-09:
I see identical behavior on other backends, so while I am also confounded by it, I don't believe it's our problem to fix.

terpstra, 2017-03-13:
Pretty sure 'volatile uint64_t *ways_v'  means the uint64_t is volatile. Volatile acts like const AFAIK.
'uint64_t * volatile ways_v' would be a volatile pointer.