riscvarchive/riscv-gcc #58: Why GCC trunk try to convert unaligned memory operation to a series of aligned memory operation?

rgwan, 2017-04-01:
Code:
#define TEST (*(volatile int*)0x10000003)
void test()
{
TEST = 0x55aaaa55;
}

Generated Assemble code:
0000008e <test>:
  8e:	10000737          	lui	a4,0x10000
  92:	00374683          	lbu	a3,3(a4) # 10000003 <end+0xffffddf>
  96:	00370793          	addi	a5,a4,3
  9a:	05500693          	li	a3,85
  9e:	00d701a3          	sb	a3,3(a4)
  a2:	00474703          	lbu	a4,4(a4)
  a6:	faa00713          	li	a4,-86
  aa:	00e780a3          	sb	a4,1(a5)
  ae:	0027c603          	lbu	a2,2(a5)
  b2:	00e78123          	sb	a4,2(a5)
  b6:	0037c703          	lbu	a4,3(a5)
  ba:	00d781a3          	sb	a3,3(a5)
  be:	8082                	ret

Not only it doesn't generates a unaligned memory access, but also it converts it to a series of aligned memory operation, and it has some unnecessary code.

Command line:
riscv32-unknown-elf-gcc -Os -march=rv32ic -ffreestanding -nostdlib -o firmware.elf firmware.S firmware.c \
	 --std=gnu99 -Wl,-Bstatic,-T,firmware.lds,-Map,firmware.map,--strip-debug -lgcc

GCC version:
gcc version 7.0.1 20170208 (experimental) (GCC)

rgwan, 2017-04-01:
If I compile the following code:
#define TEST (*(int*)0x10000003)
void test()
{
	TEST = 0x55aaaa55;
}

It generates following assemble code which inclodes a unnecessary addi instruction:
0000008e <test>:
  8e:	100006b7          	lui	a3,0x10000
  92:	05500713          	li	a4,85
  96:	00368793          	addi	a5,a3,3 # 10000003 <end+0xffffdff>
  9a:	00e681a3          	sb	a4,3(a3)
  9e:	faa00693          	li	a3,-86
  a2:	00d780a3          	sb	a3,1(a5)
  a6:	00d78123          	sb	a3,2(a5)
  aa:	00e781a3          	sb	a4,3(a5)
  ae:	8082                	ret

kito-cheng, 2017-04-01:
Current RISC-V GCC prefer aligned memory access since we assume unaligned access may slow than aligned access, so GCC will try to convert to aligned access if possible, maybe we can add a option -munaliged-access to let GCC just generate unaligned access.

sorear, 2017-04-01:
The current gcc has an "incorrect" machine description which says that unaligned access is "impossible" rather than "slow".  This was done because current Berkely platforms require a M-mode trap handler to implement unaligned access, and if that trap handler were to access unaligned memory it would cause infinite recursion.

sorear, 2017-04-01:
https://groups.google.com/a/groups.riscv.org/d/msg/sw-dev/gRPLXGWNt2A/Q12i945JHwAJ where the decision was made to lie to gcc.

aswaterman, 2017-04-01:
The issues that @sorear mentioned still stand, but I think @kito-cheng's suggestion is the way to go.  I've added the -m[no-]strict-align option, for which the default value comes from -mtune.  https://github.com/riscv/riscv-gcc/pull/59