riscvarchive/riscv-gcc #61: Stack allocation alignment

coldney, 2017-04-04:
C Code:
```
void test_stack(void) 
{
    volatile char temp;
    temp = 0;
}
```

ASM Code
```
addi	sp,sp,-16
sb	zero,15(sp)
addi	sp,sp,16
ret
```

The compiler tries to keep items on the stack aligned to 16 byte. Can I control the stack to align to 2^n?

Reference:
http://stackoverflow.com/questions/1061818/stack-allocation-padding-and-alignment

palmer-dabbelt, 2017-04-04:
This is a common compiler problem: stacks tend to be allocated conservatively, which means they're aligned to the largest type that can be spilled to the stack.  On RISC-V, this is 16 bytes (due to the Q extension for 128-bit floating point).  This means the smallest stack allocation is 16 bytes.  While this might be worth optimizing on RISC-V (since nobody has Q), it would require a new sort of compiler hook so we haven't done it.

Here's some test code that attempts to demonstrate what I'm talking about: you can see that GCC will happily allocate up to 16 chars into a stack slot, but will then allocate another 16-byte stack slot for the rest of your code.

````
void test_stack(void) 
{
    volatile char temp0;
    volatile char temp1;
    volatile char temp2;
    volatile char temp3;
    volatile char temp4;
    volatile char temp5;
    volatile char temp6;
    volatile char temp7;
    volatile char temp8;
    volatile char temp9;
    volatile char tempa;
    volatile char tempb;
    volatile char tempc;
    volatile char tempd;
    volatile char tempe;
    volatile char tempf;
#ifdef EXTRA_STACK_SLOT
    volatile char next_slot;
    next_slot = 0;
#endif
    temp0 = temp1 = temp2 = temp3 = 0;
    temp4 = temp5 = temp6 = temp7 = 0;
    temp8 = temp9 = tempa = tempb = 0;
    tempc = tempd = tempe = tempf = 0;
}
````

````
$ riscv64-unknown-linux-gnu-gcc list.c -S -o- -O3 -march=rv32i -mabi=ilp32
        .file   "list.c"
        .option nopic
        .text
        .align  2
        .globl  test_stack
        .type   test_stack, @function
test_stack:
        add     sp,sp,-16
        sb      zero,3(sp)
        sb      zero,2(sp)
        sb      zero,1(sp)
        sb      zero,0(sp)
        sb      zero,7(sp)
        sb      zero,6(sp)
        sb      zero,5(sp)
        sb      zero,4(sp)
        sb      zero,11(sp)
        sb      zero,10(sp)
        sb      zero,9(sp)
        sb      zero,8(sp)
        sb      zero,15(sp)
        sb      zero,14(sp)
        sb      zero,13(sp)
        sb      zero,12(sp)
        add     sp,sp,16
        jr      ra
        .size   test_stack, .-test_stack
        .ident  "GCC: (GNU) 7.0.1 20170321 (experimental)"
````

````
$ riscv64-unknown-linux-gnu-gcc list.c -S -o- -O3 -march=rv32i -mabi=ilp32 -DEXTRA_STACK_SLOT
        .file   "list.c"
        .option nopic
        .text
        .align  2
        .globl  test_stack
        .type   test_stack, @function
test_stack:
        add     sp,sp,-32
        sb      zero,31(sp)
        sb      zero,18(sp)
        sb      zero,17(sp)
        sb      zero,16(sp)
        sb      zero,15(sp)
        sb      zero,22(sp)
        sb      zero,21(sp)
        sb      zero,20(sp)
        sb      zero,19(sp)
        sb      zero,26(sp)
        sb      zero,25(sp)
        sb      zero,24(sp)
        sb      zero,23(sp)
        sb      zero,30(sp)
        sb      zero,29(sp)
        sb      zero,28(sp)
        sb      zero,27(sp)
        add     sp,sp,32
        jr      ra
        .size   test_stack, .-test_stack
        .ident  "GCC: (GNU) 7.0.1 20170321 (experimental)"
````

What are you actually trying to accomplish with your stack alignment problem?

coldney, 2017-04-04:
```
addi	sp,sp,-16
sb	zero,15(sp)
addi	sp,sp,16
ret
```
From this example, the stack allocated 16 bytes of memory. Actually, only 1 byte of memory was be used.
This is a waste of memory space, I want to set the stack alignment to 4 bytes to save the memory space. 

Such as:
```
addi	sp,sp,-4
sb	zero,3(sp)
addi	sp,sp,4
ret
```
Can I have some extra options to control this?

sorear, 2017-04-04:
@palmer-dabbelt Is there common code we could easily reuse for `-mno-stack-align` (CRIS) or `-mpreferred-stack-boundary=2` (x86) ?

@coldney You're trading off memory size of code versus memory size of stacks, since RVC has a 2-byte format for stack pointer adjustments that preserve 16-byte alignment, i.o.w for stack frames between 40 and 492 bytes, you can save 4 bytes per function by rounding up the stack size.  Since code is much larger than total stacks for most programs, what you're proposing is likely only a win if code is in a ROM which is much larger than the RAM size.

coldney, 2017-04-04:
Sample 1:
```
1141                	addi	sp,sp,-16
000107a3          	sb	zero,15(sp)
0141                	addi	sp,sp,16
8082                	ret
```
Sample 2:
```
1171                	addi	sp,sp,-4
000101a3          	sb	zero,3(sp)
0111                	addi	sp,sp,4
8082                	ret
```

The two sample have the same code size. But , Sample 1 saves 12byte ram size.

And , in fact, we run the program with Maskrom or Flash, and use RAM to place variables. We take more care of the size of RAM size.

sorear, 2017-04-04:
c.addi supports arbitrary stack sizes up to 31 bytes.  c.addi16sp supports 16-byte aligned stack sizes up to 496 bytes.

coldney, 2017-04-04:
Oh, I see.

aswaterman, 2018-05-02:
The option -mpreferred-stack-boundary=3 now supports reducing the stack alignment to 8 bytes on RV32I.  Note this creates an ABI compatibility, so the entire application must be compiled the same way and it should not be used with shared libraries.

closed 2018-05-02 by aswaterman