riscvarchive/riscv-gcc #64: How to stop generating the RVC codes?

theotuck, 2017-05-02:
hello,

I want to generate a executable file without RVC codes. But even  compiling program with `-march=rv64g`, there still are RVC codes in standard library functions, for example, the `printf` and `memset`. I get this through  `riscv64-unknown-elf-objdump -d `. while if I use `-march=rv64gc`, both my own functions and the standard ones will include RVC codes.

Automatically genetating RVC codes as many as possible is the correct behavior of the cross compiler? If yes, what should I do to stop it? Or there are something I misunderstanding?

So many thanks for any advice!

kito-cheng, 2017-05-02:
RVC codes *NOT* generate by compiler, it's converte by assembler, you can use `riscv-gnu-toolchain` to build a pure RV64G toolchain to prevent any RVC codes:

```
<path-to-riscv-gnu-toolchain>/configure --with-arch=rv64g --disable-multilib
```

theotuck, 2017-05-02:
Oh yes, I recompiled the toolchain with the options and it stop generating the RVC codes.

Thanks for your advice!  Please forgive my noise!

closed 2017-05-02 by theotuck