riscvarchive/riscv-gcc #68: unrecognized command line option '-mmemcpy'

ilg-ul, 2017-05-10:
The error messages are:

```
riscv64-unknown-elf-g++: error: unrecognized command line option '-mmemcpy'
riscv64-unknown-elf-g++: error: unrecognized command line option '-mno-memcpy'; did you mean '-Wno-cpp'?
```

This happens with:

```
$ opt/riscv64-unknown-elf-gcc-20170503-x86_64-apple-darwin/bin/riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 7.1.0
```

palmer-dabbelt, 2017-05-10:
Sorry, it looks like this is a documentation bug.  I think when we removed the movmem patterns we removed the memcpy option as well.  We're ading the movmem patterns back in, see https://github.com/riscv/riscv-gcc/pull/67 .  That doesn't pass the test suite yet, but when it does I'll add the option back in.

ilg-ul, 2017-05-10:
should I temporarily disable it in the plug-in? or a new version of the compiler will be released shortly?

palmer-dabbelt, 2017-05-10:
I think it's best to disable it for now, the test failures might take a while to track down.

ilg-ul, 2017-05-10:
> disable it for now

ok, disabled.

jim-wilson, 2018-01-11:
This was fixed upstream.  The -mmemcpy option was removed from the RISC-V docs.

closed 2018-01-11 by jim-wilson