riscvarchive/riscv-gcc #69: GCC Testsuite fail: memmove-4.c and strlenopt-8.c

kito-cheng, 2017-05-12:
Seem cause by https://github.com/riscv/riscv-gcc/commit/2fe94d5a5a860a0226373397e05c20a873b77b4d
```
FAIL: gcc.dg/memmove-4.c scan-tree-dump-not optimized "memmove"
FAIL: gcc.dg/strlenopt-8.c scan-tree-dump-times strlen "memcpy \\(" 2
```

kito-cheng, 2017-05-12:
It's seem because we de-couple between SLOW_UNALIGNED_ACCESS and STRICT_ALIGNMENT, before https://github.com/riscv/riscv-gcc/commit/2fe94d5a5a860a0226373397e05c20a873b77b4d non_strict_align is `0` and now we got `1`.

gcc/testsuite/gcc.dg/memmove-4.c:
```
/* { dg-final { scan-tree-dump-not "memmove" "optimized" { xfail { ! non_strict_align } } } } */
```

gcc/testsuite/gcc.dg/strlenopt-8.c 
```
/* { dg-final { scan-tree-dump-times "strlen \\(" 0 "strlen" { xfail non_strict_align } } } */
/* { dg-final { scan-tree-dump-times "memcpy \\(" 2 "strlen" { target { non_strict_align } } } } */
/* { dg-final { scan-tree-dump-times "memcpy \\(" 4 "strlen" { target { ! non_strict_align } } } }  */
```

gcc/testsuite/lib/target-supports.exp
```
# Return 1 if the target does *not* require strict alignment.

proc check_effective_target_non_strict_align {} { 

    # On ARM, the default is to use STRICT_ALIGNMENT, but there
    # are interfaces defined for misaligned access and thus
    # depending on the architecture levels unaligned access is
    # available.
    if [istarget "arm*-*-*"] {
        return [check_effective_target_arm_unaligned]
    }    

    return [check_no_compiler_messages non_strict_align assembly {
        char *y;
        typedef char __attribute__ ((__aligned__(__BIGGEST_ALIGNMENT__))) c;
        c *z;
        void foo(void) { z = (c *) y; } 
    } "-Wcast-align"]
}
```

tree-ssa-strlen.c:
```
...
/* Handle a call to memcmp.  We try to handle small comparisons by
   converting them to load and compare, and replacing the call to memcmp
   with a __builtin_memcmp_eq call where possible.  */
 
static bool
handle_builtin_memcmp (gimple_stmt_iterator *gsi)
...
      if (mode != BLKmode
          && (align >= leni || !SLOW_UNALIGNED_ACCESS (mode, align)))
        {
...
```

aswaterman, 2017-05-12:
Are we the only target that decouples SLOW_UNALIGNED_ACCESS from STRICT_ALIGNMENT?

kito-cheng, 2017-05-12:
Yeah, seem like we are the only one, maybe we just can add dg-options for those two testcase.

aswaterman, 2017-05-12:
Yeah, that seems OK to me.  The ```check_effective_target_arm_unaligned``` approach seems like a hack, too.

kito-cheng, 2017-05-24:
This issue is gone since https://github.com/riscv/riscv-gcc/pull/73 merged

closed 2017-05-24 by kito-cheng