riscvarchive/riscv-gcc #70: input that causes an internal compiler error, minimized from bzip2.c in bzip2

dsw, 2017-05-17:
The input below (foo.c) causes an internal compiler error when compiled as follows using the v20170503 release of riscv-gnu-toolchain when configured with --with-arch=rv64imafd (note: no "c").

Command line:
$RISCV/bin/riscv64-unknown-elf-gcc -O2 -mcmodel=medany -o /dev/null foo.c

foo.c (minimized from bzip2.c in bzip2):
int x;
struct Z {
  char *n;
  struct Z *m;
};
void license();
int main() {
  struct Z *a;
  struct Z *b;
  for (b = a; b; b = b->m) {
    if (b->n[0] == '-') {
      int j;
      for (j = 1; b->n[j] != '\0'; j++) {
        switch (b->n[j]) {
        case '6':
          x = 6;
          break;
        case 'L':
          license();
          break;
        }
      }
    }
  }
}

palmer-dabbelt, 2017-05-17:
Can you show me the ICE message?  I've fixed a handful of -mcmodel=medany bugs since that release and I'm not getting any failures on my Linux toolchain right now.  Note that my compiler is optimizing away your entire test case, so YMMV

````
palmer gamma02 build-linux64 $ riscv64-unknown-linux-gnu-objdump -d test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <license>:
   0:	00008067          	ret

Disassembly of section .text.startup:

0000000000000000 <main>:
   0:	00000513          	li	a0,0
   4:	00008067          	ret
palmer gamma02 build-linux64 $ cat test.c 
int x;
struct Z {
  char *n;
  struct Z *m;
};
void license() {}
int main() {
  struct Z *a;
  struct Z *b;
  for (b = a; b; b = b->m) {
    if (b->n[0] == '-') {
      int j;
      for (j = 1; b->n[j] != '\0'; j++) {
        switch (b->n[j]) {
        case '6':
          x = 6;
          break;
        case 'L':
          license();
          break;
        }
      }
    }
  }
}
palmer gamma02 build-linux64 $ riscv64-unknown-linux-gnu-gcc test.c -c -O2 -mcmodel=medany -march=rv64imafd
palmer gamma02 build-linux64 $ riscv64-unknown-linux-gnu-objdump -d test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <license>:
   0:	00008067          	ret

Disassembly of section .text.startup:

0000000000000000 <main>:
   0:	00000513          	li	a0,0
   4:	00008067          	ret
````

dsw, 2017-05-17:
Here is the entirety of the output, except that I replace my home directory
with $HOME throughout.

$HOME/riscv/bin/riscv64-unknown-elf-gcc -O2 -mcmodel=medany -o /dev/null
foo.c
foo.c: In function 'main':
foo.c:25:1: error: unrecognizable insn:
 }
 ^
(insn 108 52 121 10 (set (mem/c:SI (symbol_ref:DI ("x") [flags 0x84]
<var_decl \
0x7f410f3a03f0 x>) [1 x+0 S4 A32])
        (reg:SI 120)) "foo.c":16 -1
     (nil))
foo.c:25:1: internal compiler error: in extract_insn, at recog.c:2311
0xa77985 _fatal_insn(char const*, rtx_def const*, char const*, int, char
const*\
)
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/rtl-error.c:108
0xa779b9 _fatal_insn_not_found(rtx_def const*, char const*, int, char
const*)
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/rtl-error.c:116
0xa4bf2a extract_insn(rtx_insn*)
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/recog.c:2311
0x1065933 get_implicit_reg_pending_clobbers(unsigned long (*) [2],
rtx_insn*)
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-deps.c:2871
0x10665db sched_analyze_insn
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-deps.c:2890
0x1067fef deps_analyze_insn(deps_desc*, rtx_insn*)
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-deps.c:3686
0x1068315 sched_analyze(deps_desc*, rtx_insn*, rtx_insn*)
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-deps.c:3833
0xa875e9 compute_block_dependences
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-rgn.c:2721
0xa875e9 sched_rgn_compute_dependencies(int)
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-rgn.c:3347
0xa89d80 schedule_region
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-rgn.c:3103
0xa89d80 schedule_insns
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-rgn.c:3506
0xa8a4b3 schedule_insns
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-rgn.c:3491
0xa8a4b3 rest_of_handle_sched
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-rgn.c:3710
0xa8a4b3 execute
        $HOME/riscv-gnu-toolchain-v20170503/riscv-gcc/gcc/sched-rgn.c:3818
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.

palmer-dabbelt, 2017-05-17:
Thanks.  I believe that bug has been fixed.  Do you mind trying the latest riscv-gnu-toolchain master?

dsw, 2017-05-17:
I cloned riscv-gnu-toolchain from scratch, did git submodule update --init
--recursive, configured, built, and ran the same command line using the
same input.

Using this commit:

commit
87fdf4305dc72e383d0334b66d28510179f1ad3c
Author: Andrew Waterman <andrew@sifive.com>

Date:   Tue May 16 19:40:56 2017
-0700

When configured as I initially reported (*NOTE*: no "c"):

$ ./configure --prefix=/home/dsw/riscv --with-arch=rv64imafd

(Again, note that I am not using the "c" in the configure argument.)

Running the same command line using the same input as I initially reported:

$HOME/riscv/bin/riscv64-unknown-elf-gcc -O2 -mcmodel=medany -o /dev/null
foo.c

I get the following output:

foo.c: In function 'main':
foo.c:25:1: error: unrecognizable insn:
 }
 ^
(insn 108 52 121 10 (set (mem/c:SI (symbol_ref:DI ("x") [flags 0x84]
<var_decl \
0x2b6e9dbea3f0 x>) [1 x+0 S4 A32])
        (reg:SI 120)) "foo.c":16 -1
     (nil))
foo.c:25:1: internal compiler error: in extract_insn, at recog.c:2311
0xa77725 _fatal_insn(char const*, rtx_def const*, char const*, int, char
const*\
)
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/rtl-error.c:108
0xa77759 _fatal_insn_not_found(rtx_def const*, char const*, int, char
const*)
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/rtl-error.c:116
0xa4bcca extract_insn(rtx_insn*)
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/recog.c:2311
0x1065dd3 get_implicit_reg_pending_clobbers(unsigned long (*) [2],
rtx_insn*)
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-deps.c:2871
0x1066a7b sched_analyze_insn
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-deps.c:2890
0x106848f deps_analyze_insn(deps_desc*, rtx_insn*)
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-deps.c:3686
0x10687b5 sched_analyze(deps_desc*, rtx_insn*, rtx_insn*)
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-deps.c:3833
0xa87399 compute_block_dependences
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-rgn.c:2721
0xa87399 sched_rgn_compute_dependencies(int)
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-rgn.c:3347
0xa89b30 schedule_region
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-rgn.c:3103
0xa89b30 schedule_insns
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-rgn.c:3506
0xa8a263 schedule_insns
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-rgn.c:3491
0xa8a263 rest_of_handle_sched
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-rgn.c:3710
0xa8a263 execute
        $HOME/riscv-gnu-toolchain-2017-05-16/riscv-gcc/gcc/sched-rgn.c:3818
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.

dsw, 2017-05-18:
Ok, if in riscv-gnu-toolchain/riscv-gcc I git pull and then git checkout
riscv-next, then rebuild in riscv-gnu-toolchain, I get a compiler that no
longer exhibits this bug.  However the current riscv-gnu-toolchain still
points at a version of riscv-gcc that exhibits this bug, as pointed out in
my previous email.

palmer-dabbelt, 2017-05-18:
Sorry about that, I forgot to bump riscv-gnu-toolchain.  This should be fixed by https://github.com/riscv/riscv-gnu-toolchain/commit/dcda255e1f3d114e8d12b23c80b80382b51109d6 .

closed 2017-05-18 by dsw