riscvarchive/riscv-gcc #72: compiling with -O2 produces an unaligned store double instruction

dsw, 2017-05-18:
Compiling bar.c below with -O2 produces an unaligned store double instruction.  Removing the -O2 fixes the problem.

Using this version of riscv-gcc:

```
commit a0d67b753fc2eeaf6aca047fe80e0c4a51b9417f                                 
Author: Kito Cheng <kito.cheng@gmail.com>                                       
Date:   Thu May 4 02:11:13 2017 +0800                                           
```

Configuring like this (note the lack of a "c" in the --with-arch argument):

```
$ ./configure --prefix=$RISCV --with-arch=rv64imafd
```

Compiling bar.c like this:

```
$RISCV/bin/riscv64-unknown-elf-gcc -o bar.o bar.c -c -mcmodel=medany -O2
```

Where bar.c (minimized from bzip2) is as follows:
```
struct DState {
  int    nInUse;
  int    save_i;
  int    save_j;
  int    save_t;
  int    save_alphaSize;
};

struct DState dstate;

struct DState *make_DState() {
  return &dstate;
}

void init_DState(struct DState * const s) {
  s->save_i           = 0;
  s->save_j           = 0;
  s->save_t           = 0;
  s->save_alphaSize   = 0;
}
```

We get a double-word aligned global dstate.

```
name: 'dstate'
        type: STT_OBJECT
        bind: STB_GLOBAL
        visibility: STV_DEFAULT
        value:  0x8 == 8
        size:   0x14 == 20
```

And an init_DState() that does an unaligned store double to its DState * parameter (note that 12 % 8 != 0):

```
init_DState:
        sw      x0, 4(a0)
        sw      x0, 8(a0)
        sd      x0, 12(a0)
        ret
```

Passing the address of the global dstate to init_DState() produces an unaligned store double.

aswaterman, 2017-05-18:
That's really bad code generation, but not technically incorrect, as RISC-V permits misaligned memory accesses.  You can use ```-mstrict-align``` to tell GCC it shouldn't do this.

aswaterman, 2017-05-18:
@kito-cheng it seems pretty weird that GCC would generate this code even though SLOW_UNALIGNED_ACCESS is true.  Even stranger: why does it use ```sd``` for words 3-4, but not for words 1-2?

aswaterman, 2017-05-18:
Seems like the new store-merging pass is doing this.  But it could be a cost model problem.

aswaterman, 2017-05-18:
OK... I think it is a bug in the store-merging pass: they are checking STRICT_ALIGNMENT instead of SLOW_UNALIGNED_ACCESS.  But the least controversial fix is to set PARAM_STORE_MERGING_ALLOW_UNALIGNED.  I'll make a PR.

aswaterman, 2017-05-18:
#73

closed 2017-05-18 by aswaterman