riscvarchive/riscv-gcc #78: RISC-V specific options not documented with enough details

ilg-ul, 2017-06-26:
The **3.18.37 RISC-V Options** section of the GCC manual lists the RISC-V specific options, but generally the explanations are very terse, and someone not very familiar with the RISC-V specifications can hardly make use of them.

For example `-mcmodel=code-model Specify the code model.`

What is a _code model_? What are the allowed values? What does these mean? When should a user select one over the other?

I suggest you try to put a user hat on you'll see how difficult is to get answers to these questions, not only for `-mcmodel`, but for the other options too.

jim-wilson, 2018-01-11:
Code model is used by multiple gcc targets, and means the same thing for all of them.  RISC-V should not be responsible for defining this term.

The old -mcmodel option docs were too terse.  Palmer fixed this a while ago, and the docs looks OK now.  So closing.

If you have concerns about other options, please be specific about which options need improvement, and what needs improving about them.

closed 2018-01-11 by jim-wilson

sorear, 2018-01-12:
I don't think is accurate.  Code model is a common concept, but the allowed code models differ between targets, as do their meanings in some cases.  `tiny` on aarch64 (all symbols within a relative distance of 1MiB) is very different from `tiny` on x86-16 (all symbols share a 64kiB segment).  aarch64 `small` corresponds to riscv64 `medany`.  These details need to be documented somewhere, and linking out to https://github.com/riscv/riscv-toolchain-conventions#specifying-the-target-code-model-with--mcmodel is not ideal.

palmer-dabbelt, 2018-01-12:
Here's all the GCC documentation for `-mcmodel`:

AArch64:
````
@item -mcmodel=tiny
@opindex mcmodel=tiny
Generate code for the tiny code model.  The program and its statically defined
symbols must be within 1MB of each other.  Programs can be statically or
dynamically linked.

@item -mcmodel=small
@opindex mcmodel=small
Generate code for the small code model.  The program and its statically defined
symbols must be within 4GB of each other.  Programs can be statically or
dynamically linked.  This is the default code model.

@item -mcmodel=large
@opindex mcmodel=large
Generate code for the large code model.  This makes no assumptions about
addresses and sizes of sections.  Programs can be statically linked only.
````

MSP430:
````
@item -mcmodel=@var{code-model}
@opindex mcmodel
Set the code model to one of
@table @asis
@item @samp{small}
All the data and read-only data segments must be within 512KB addressing space.
The text segment must be within 16MB addressing space.
@item @samp{medium}
The data segment must be within 512KB while the read-only data segment can be
within 4GB addressing space.  The text segment should be still within 16MB
addressing space.
@item @samp{large}
All the text and data segments can be within 4GB addressing space.
@end table
````

RISC-V:
````
@item -mcmodel=medlow
@opindex mcmodel=medlow
Generate code for the medium-low code model. The program and its statically
defined symbols must lie within a single 2 GiB address range and must lie
between absolute addresses @minus{}2 GiB and +2 GiB. Programs can be
statically or dynamically linked. This is the default code model.

@item -mcmodel=medany
@opindex mcmodel=medany
Generate code for the medium-any code model. The program and its statically
defined symbols must be within any single 2 GiB address range. Programs can be
statically or dynamically linked.
````

RS/6000:
````
@item -mcmodel=small
@opindex mcmodel=small
Generate PowerPC64 code for the small model: The TOC is limited to
64k.

@item -mcmodel=medium
@opindex mcmodel=medium
Generate PowerPC64 code for the medium model: The TOC and other static
data may be up to a total of 4G in size.  This is the default for 64-bit
Linux.

@item -mcmodel=large
@opindex mcmodel=large
Generate PowerPC64 code for the large model: The TOC may be up to 4G
in size.  Other data and code is only limited by the 64-bit address
space.
````

SPARC:
````
@item -mcmodel=@var{which}
@opindex mcmodel
Set the code model to one of

@table @samp
@item medlow
The Medium/Low code model: 64-bit addresses, programs
must be linked in the low 32 bits of memory.  Programs can be statically
or dynamically linked.

@item medmid
The Medium/Middle code model: 64-bit addresses, programs
must be linked in the low 44 bits of memory, the text and data segments must
be less than 2GB in size and the data segment must be located within 2GB of
the text segment.

@item medany
The Medium/Anywhere code model: 64-bit addresses, programs
may be linked anywhere in memory, the text and data segments must be less
than 2GB in size and the data segment must be located within 2GB of the
text segment.

@item embmedany
The Medium/Anywhere code model for embedded systems:
64-bit addresses, the text and data segments must be less than 2GB in
size, both starting anywhere in memory (determined at link time).  The
global register %g4 points to the base of the data segment.  Programs
are statically linked and PIC is not supported.
@end table
````

Tile-Gx:
````
@item -mcmodel=small
@opindex mcmodel=small
Generate code for the small model.  The distance for direct calls is
limited to 500M in either direction.  PC-relative addresses are 32
bits.  Absolute addresses support the full address range.

@item -mcmodel=large
@opindex mcmodel=large
Generate code for the large model.  There is no limitation on call
distance, pc-relative addresses, or absolute addresses.
````

x86:
````
@item -mcmodel=small
@opindex mcmodel=small
Generate code for the small code model: the program and its symbols must
be linked in the lower 2 GB of the address space.  Pointers are 64 bits.
Programs can be statically or dynamically linked.  This is the default
code model.

@item -mcmodel=kernel
@opindex mcmodel=kernel
Generate code for the kernel code model.  The kernel runs in the
negative 2 GB of the address space.
This model has to be used for Linux kernel code.

@item -mcmodel=medium
@opindex mcmodel=medium
Generate code for the medium model: the program is linked in the lower 2
GB of the address space.  Small symbols are also placed there.  Symbols
with sizes larger than @option{-mlarge-data-threshold} are put into
large data or BSS sections and can be located above 2GB.  Programs can
be statically or dynamically linked.

@item -mcmodel=large
@opindex mcmodel=large
Generate code for the large model.  This model makes no assumptions
about addresses and sizes of sections.
````

As far as I can tell, our documentation looks exactly like every other port's documentation.  If there are other places this should be documented I'd be happy to add them, but I don't really see any reason to mess with this.

sorear, 2018-01-12:
The above looks fine to me.  I missed the part where you fixed this already, sorry:

> The old -mcmodel option docs were too terse. Palmer fixed this a while ago, and the docs looks OK now. So closing.

palmer-dabbelt, 2018-01-12:
Ah, OK, that makes sense -- the old docs were a mess.  Looks like these patches didn't make 7.2.0, but as far as I can tell they're on `gcc-7-branch` so they should make it to the GCC website as part of the 7.3.0 release.

sorear, 2018-01-12:
Point of no consequence: Trunk documentation is visible on the website, https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Options.html#RISC-V-Options

palmer-dabbelt, 2018-01-12:
Ah, that's super useful :)