riscvarchive/riscv-gcc #81: FDE Emitted Incorrectly

palmer-dabbelt, 2017-07-17:
A full example of this brokenness can be seen in the "smp" example from <https://github.com/sifive/freedom-e-sdk>.  The problem appears to be that when emitting "ra" into the FDE we end up with the wrong offset.  I've reduced the test case a bit.  When compiling this file

````
$ cat smp.c 
#define ATOMIC_INIT(x)          \
        {                       \
                .counter = (x), \
        }

typedef struct {
  int counter;
} atomic_t;

static inline int atomic_xchg(atomic_t *v, int n)
{
  register int c;

  __asm__ __volatile__ (
    "amoswap.w.aqrl %0, %2, %1"
    : "=r" (c), "+A" (v->counter)
    : "r" (n));
  return c;
}

static inline void mb(void)
{
  __asm__ __volatile__ ("fence");
}

atomic_t tty_lock = ATOMIC_INIT(0);

void get_lock(atomic_t *lock)
{
  while (atomic_xchg(lock, 1) == 1);
  mb();
}

void put_lock(atomic_t *lock)
{
  mb();
  atomic_xchg(lock, 0);
}

int secondary_main(int hartid)
{
  volatile int counter;

  while (1) {
    get_lock(&tty_lock);
    write(1, "hello world from hart ", 22);
    char s[] = {'0', '\n', '\0'};
    s[0] += hartid;
    write(1, s, 2);
    put_lock(&tty_lock);

    for (counter = 0; counter < 10000 + 100*hartid; ++counter)
        mb();
  }
}

int main()
{
  return secondary_main(0);
}
````

I appear to get an incorrect FDE entry spit out into the assembly file.  The "-O2" code looks like this

````
secondary_main:
.LFB4:
        .loc 1 41 0
.LVL6:
        add     sp,sp,-96
.LCFI0:
        sd      s2,64(sp)
.LCFI1:
.LBB35:
        .loc 1 52 0
        li      s2,100
        mulw    s2,s2,a0
        li      a5,8192
        addw    a0,a0,48
.LVL7:
        addw    a5,a5,1807
.LBE35:
        .loc 1 41 0
        sd      s0,80(sp)
        sd      s1,72(sp)
        sd      s3,56(sp)
        sd      s4,48(sp)
        sd      s5,40(sp)
        sd      s6,32(sp)
        sd      s7,24(sp)
        sd      ra,88(sp)
.LCFI2:
        and     s4,a0,0xff
        lui     s0,%hi(tty_lock)
        lui     s5,%hi(.LC0)
````

but the FDE entry looks like this

````
.LASFDE4:
        .4byte  .Lframe0
        .8byte  .LFB4
        .8byte  .LFE4-.LFB4
        .byte   0x4
        .4byte  .LCFI0-.LFB4
        .byte   0xe
        .byte   0x60
        .byte   0x4
        .4byte  .LCFI1-.LCFI0
        .byte   0x92
        .byte   0x8
        .byte   0x4
        .4byte  .LCFI2-.LCFI1
        .byte   0x88
        .byte   0x4
        .byte   0x89
        .byte   0x6
        .byte   0x93
        .byte   0xa
        .byte   0x94
        .byte   0xc
        .byte   0x95
        .byte   0xe
        .byte   0x96
        .byte   0x10
        .byte   0x97
        .byte   0x12
        .byte   0x81
        .byte   0x2
        .align  3
````

This decodes to something like this (I've jumped to a slightly different toolchain here, so it might be a touch off)

````
00000040 000000000000003c 00000000 FDE cie=00000000 pc=0000000040400206..00000000404002a4
  DW_CFA_advance_loc4: 2 to 0000000040400208
  DW_CFA_def_cfa_offset: 96
  DW_CFA_advance_loc4: 2 to 000000004040020a
  DW_CFA_offset: r18 at cfa-32
  DW_CFA_advance_loc4: 34 to 000000004040022c
  DW_CFA_offset: r8 at cfa-16
  DW_CFA_offset: r9 at cfa-24
  DW_CFA_offset: r19 at cfa-40
  DW_CFA_offset: r20 at cfa-48
  DW_CFA_offset: r21 at cfa-56
  DW_CFA_offset: r22 at cfa-64
  DW_CFA_offset: r23 at cfa-72
  DW_CFA_offset: r1 at cfa-8
  DW_CFA_nop
  DW_CFA_nop
  DW_CFA_nop
  DW_CFA_nop
  DW_CFA_nop
````

The problem is that last CFA entry:

````
.byte 0x81
.byte 0x2
````

which decodes to

````
  DW_CFA_offset: r1 at cfa-8
````

because our CFA offset is "-4".  This doesn't match the generated assembly

````
  44:   ec86                    sd      ra,88(sp)
````

I'm not sure what's going on, so I'm dumping a bug in here for now...

palmer-dabbelt, 2017-07-17:
Hey @kito-cheng: I'm pretty slammed right now, do you understand how the CFA stuff works?

palmer-dabbelt, 2017-07-17:
@kuanlinchentw Do you happen to know how to both emit non-zero bits and generate a fixup in GAS?

kito-cheng, 2017-07-18:
@palmer-dabbelt Ooops, I don't know why riscv-gcc generate raw CFI info here, our riscv-gcc (internal build) can generate human readable CFI directive, let't me investigate how to make it human readable first, raw CFI info is too terrible...

asm output generate from our gcc:
```
$ riscv32-elf-gcc pr81.c -S -O2 -g
$ cat pr81.s
...
        .align  2
        .globl  secondary_main
        .type   secondary_main, @function
secondary_main:
.LFB4:
        .loc 1 41 0
        .cfi_startproc
.LVL6:
        add     sp,sp,-64
        .cfi_def_cfa_offset 64
        sw      s2,48(sp)
        .cfi_offset 18, -16
...
```

kuanlinchentw, 2017-07-18:
@palmer-dabbelt 
```
DW_CFA_def_cfa_offset: 96
...
DW_CFA_offset: r1 at cfa-8
```

```
add     sp,sp,-96
...
sd      ra,88(sp)
```
The CFA information seems correct.
I don't understand what you mean "because our CFA offset is "-4". This doesn't match the generated assembly"

palmer-dabbelt, 2017-07-18:
Ah, sorry -- I was being an idiot and missed the offset.  That means the CFA appears to be correct...

kito-cheng, 2017-07-25:
Hi @palmer-dabbelt, just curious what the problem behind this issue and it's resolved or not?

palmer-dabbelt, 2017-07-26:
The problem is that debugging with GDB seems to go off the rails right after stepping over that store of RA (it think the previous frame is 0).  The SMP example in freedom-e-sdk triggers the bug, if you're interested in looking.

kito-cheng, 2017-07-26:
Humm, ok, so it's sound like is bug in GDB if FDE is correct? Do you have try to run GDB testsuite before?

palmer-dabbelt, 2017-07-26:
It's probably a GDB bug, I'm closing this.

closed 2017-07-26 by palmer-dabbelt