riscvarchive/riscv-gcc #82: Debug info generated by default; -gtoggle or -g0 won't disable

michaeljclark, 2017-07-23:
I'm currently investigating binary sizes (-Os vs -O2), but in the process noticed that the default binaries were quite large. I then discovered that all binaries had debug info whether or not I had requested it with -g, and even trying to turn debug into off with -gtoggle or -g0 has no effect.

Reference: https://gcc.gnu.org/onlinedocs/gcc-4.9.2/gcc/Debugging-Options.html

I've tried with gcc 7.0.1 20170321 (experimental) and the gcc 7.1 release, with various configurations (newlib, glibc, musl) and they all appear to add debug info by default.

It seems `riscv64-unknown-elf-strip --strip-debug` works okay in the mean time, however shouldn't the RISC-V GCC port honour the -g flags?

```
$ riscv64-unknown-elf-gcc -static -Os -g0 src/aes.c -o bin/riscv64/aes 
$ riscv64-unknown-elf-objdump -h bin/riscv64/aes 

bin/riscv64/aes:     file format elf64-littleriscv

Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 .text         0000d5fa  00000000000100b0  00000000000100b0  000000b0  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .rodata       00003438  000000000001d6b0  000000000001d6b0  0000d6b0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .eh_frame     00000004  0000000000020ae8  0000000000020ae8  00010ae8  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  3 .init_array   00000008  0000000000021af0  0000000000021af0  00010af0  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  4 .fini_array   00000008  0000000000021af8  0000000000021af8  00010af8  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  5 .data         00000ff8  0000000000021b00  0000000000021b00  00010b00  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  6 .sdata        00000070  0000000000022af8  0000000000022af8  00011af8  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  7 .sbss         00000030  0000000000022b68  0000000000022b68  00011b68  2**3
                  ALLOC
  8 .bss          00000078  0000000000022b98  0000000000022b98  00011b68  2**3
                  ALLOC
  9 .comment      00000029  0000000000000000  0000000000000000  00011b68  2**0
                  CONTENTS, READONLY
 10 .debug_aranges 00000c40  0000000000000000  0000000000000000  00011ba0  2**4
                  CONTENTS, READONLY, DEBUGGING
 11 .debug_info   0002ed21  0000000000000000  0000000000000000  000127e0  2**0
                  CONTENTS, READONLY, DEBUGGING
 12 .debug_abbrev 00008bac  0000000000000000  0000000000000000  00041501  2**0
                  CONTENTS, READONLY, DEBUGGING
 13 .debug_line   00013a71  0000000000000000  0000000000000000  0004a0ad  2**0
                  CONTENTS, READONLY, DEBUGGING
 14 .debug_frame  00003520  0000000000000000  0000000000000000  0005db20  2**3
                  CONTENTS, READONLY, DEBUGGING
 15 .debug_str    00003476  0000000000000000  0000000000000000  00061040  2**0
                  CONTENTS, READONLY, DEBUGGING
 16 .debug_loc    0002cbd1  0000000000000000  0000000000000000  000644b6  2**0
                  CONTENTS, READONLY, DEBUGGING
 17 .debug_ranges 00003260  0000000000000000  0000000000000000  00091087  2**0
                  CONTENTS, READONLY, DEBUGGING
```

aswaterman, 2017-07-23:
Looking at the assembly output, -g is honored. Are you statically linking against a a C library with debug symbols?  -g won't remove them.

michaeljclark, 2017-07-23:
That makes sense,

... however now I'm building 4 toolchains: riscv32, riscv64, i386, x86_64 using the exact same configure switches, sources and options and I'm not passing -g to the musl-riscv build. This is an analysis toolchain primarily so that I have a baseline to measure across platforms:

- https://github.com/rv8-io/musl-riscv-toolchain/releases/tag/v7.1.0-2

And when I objdump the x86_64 binaries I don't see any debug info. So somewhere a -g option is getting added for parts of the riscv build? have to sift through the build log to see where the -g gets added...

michaeljclark, 2017-07-24:
The C library had no debug info, just symbols, as I'm not passing in -g.

After looking a little more closely it appears that gcc compiles libgcc with debug info by default for all targets. I was surprised by the default size of riscv executables compared to x86. Debug info appears to be a lot bigger on riscv which is interesting. Will close this issue anyhow...

```
425672 bin/riscv32/miniz
425328 bin/riscv64/miniz
350896 bin/x86_64/miniz
286920 bin/i386/miniz
194856 bin/riscv32/norx
191796 bin/riscv32/aes
179344 bin/riscv32/sha512
175704 bin/riscv32/dhrystone
175068 bin/riscv32/qsort
165660 bin/riscv32/primes
129440 bin/riscv64/norx
126264 bin/riscv64/aes
110536 bin/riscv64/sha512
110268 bin/i386/miniz.stripped
107424 bin/riscv64/qsort
107008 bin/riscv64/dhrystone
99956 bin/riscv32/miniz.stripped
99624 bin/riscv64/primes
99168 bin/x86_64/miniz.stripped
91392 bin/riscv64/miniz.stripped
73104 bin/i386/aes
70144 bin/i386/norx
61552 bin/x86_64/norx
56808 bin/x86_64/aes
55980 bin/i386/dhrystone
55672 bin/i386/sha512
55340 bin/riscv32/aes.stripped
51540 bin/i386/qsort
49144 bin/riscv64/aes.stripped
47824 bin/i386/aes.stripped
47540 bin/riscv32/norx.stripped
46764 bin/i386/primes
44624 bin/x86_64/aes.stripped
42584 bin/riscv32/dhrystone.stripped
42380 bin/riscv32/sha512.stripped
41980 bin/riscv32/qsort.stripped
38752 bin/x86_64/sha512
38104 bin/x86_64/qsort
37792 bin/riscv32/primes.stripped
37416 bin/riscv64/norx.stripped
37000 bin/x86_64/norx.stripped
36640 bin/x86_64/dhrystone
35932 bin/i386/norx.stripped
32216 bin/riscv64/dhrystone.stripped
31944 bin/riscv64/sha512.stripped
31384 bin/riscv64/qsort.stripped
31264 bin/riscv64/primes.stripped
31132 bin/i386/dhrystone.stripped
30764 bin/i386/sha512.stripped
27768 bin/x86_64/dhrystone.stripped
27432 bin/x86_64/sha512.stripped
26904 bin/x86_64/qsort.stripped
26368 bin/x86_64/primes
26292 bin/i386/qsort.stripped
26200 bin/i386/primes.stripped
22688 bin/x86_64/primes.stripped
```

closed 2017-07-24 by michaeljclark