riscvarchive/riscv-gcc #83: redundant sign-extension in RV64

aswaterman, 2017-07-26:
@michaeljclark to produce minimal test cases.

michaeljclark, 2017-07-26:
baseline gcc 7.1.0 seems to have redundant sign extension at subroutine return but is not as bad as master snapshot as of a few weeks ago (see below) which has pathological sign extension. The test program needs to be assembled on current master (which I don't have a local build of currently). In any case gcc 7.1.0 codegen is not too bad but is not ideal. We need to check current master to rule out any breakage that was there at the time when I checked out my last release (gcc 7.1.1 `20170630`). My intention was to focus my measurements on a new baseline of official `gcc 7.1.0`. `gcc 6.1.0` appears to not have the issue. The compiler explorer link shows 7.1.0 codegen.

- https://cx.rv8.io/g/ssM5UJ

test program

```
#include <assert.h>

unsigned bswap1(unsigned p)
{
       return __builtin_bswap32(p);
}

unsigned bswap2(unsigned p)
{
       return ((p >> 24) & 0x000000ff) | ((p << 8 ) & 0x00ff0000) | 
        ((p >> 8 ) & 0x0000ff00) | ((p << 24) & 0xff000000);
}

int main()
{
       assert(bswap1(0x01020304) == bswap2(0x01020304));
}
```

gcc baseline from ~ Dec 2016 pre-gcc 7.1 merge

```
$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 6.1.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

bswap1:
       add     sp,sp,-32
       sd      ra,24(sp)
       sd      s0,16(sp)
       add     s0,sp,32
       sw      a0,-20(s0)
       lw      a0,-20(s0)
       call    __bswapsi2
       mv      a5,a0
       mv      a0,a5
       ld      ra,24(sp)
       ld      s0,16(sp)
       add     sp,sp,32
       jr      ra
       .size   bswap1, .-bswap1
       .align  2
       .globl  bswap2
       .type   bswap2, @function
bswap2:
       add     sp,sp,-32
       sd      s0,24(sp)
       add     s0,sp,32
       sw      a0,-20(s0)
       lw      a5,-20(s0)
       srlw    a4,a5,24
       lw      a5,-20(s0)
       sllw    a3,a5,8
       li      a5,16711680
       and     a5,a3,a5
       or      a4,a4,a5
       lw      a5,-20(s0)
       srlw    a3,a5,8
       li      a5,65536
       addw    a5,a5,-256
       and     a5,a3,a5
       or      a4,a4,a5
       lw      a5,-20(s0)
       sllw    a5,a5,24
       or      a5,a4,a5
       mv      a0,a5
       ld      s0,24(sp)
       add     sp,sp,32
       jr      ra
```

current master

```
$ /opt/riscv/toolchain-master-20170630/bin/riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 7.1.1 20170509
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

bswap1:
	add	sp,sp,-32
	sd	ra,24(sp)
	sd	s0,16(sp)
	add	s0,sp,32
	mv	a5,a0
	sw	a5,-20(s0)
	lw	a0,-20(s0)
	call	__bswapsi2
	mv	a5,a0
	sext.w	a5,a5
	mv	a0,a5
	ld	ra,24(sp)
	ld	s0,16(sp)
	add	sp,sp,32
	jr	ra

bswap2:
	add	sp,sp,-32
	sd	s0,24(sp)
	add	s0,sp,32
	mv	a5,a0
	sw	a5,-20(s0)
	lw	a5,-20(s0)
	srlw	a5,a5,24
	sext.w	a4,a5
	lw	a5,-20(s0)
	sllw	a5,a5,8
	sext.w	a5,a5
	mv	a3,a5
	li	a5,16711680
	and	a5,a3,a5
	sext.w	a5,a5
	or	a5,a4,a5
	sext.w	a4,a5
	lw	a5,-20(s0)
	srlw	a5,a5,8
	sext.w	a5,a5
	mv	a3,a5
	li	a5,65536
	add	a5,a5,-256
	and	a5,a3,a5
	sext.w	a5,a5
	or	a5,a4,a5
	sext.w	a4,a5
	lw	a5,-20(s0)
	sllw	a5,a5,24
	sext.w	a5,a5
	or	a5,a4,a5
	sext.w	a5,a5
	mv	a0,a5
	ld	s0,24(sp)
	add	sp,sp,32
	jr	ra
	.size	bswap2, .-bswap2
	.section	.rodata
	.align	3
```

michaeljclark, 2017-08-17:
@aswaterman @palmer-dabbelt do you have any insight into where to look for this redundant sign extension? would it likely be a core compiler change or a riscv metadata change? BTW I still have a vintage riscv gcc 6.1 on my mac. The other notable difference is that the preceding add is not an addw thus the word is no longer canonical word form and might be why the sext.w is needed, so that points at instruction selection for add perhaps?

```
$ cat sw.c 
unsigned sw(unsigned rs1)
{
  return ((rs1 >> 24) & 0x000000ff) |
          ((rs1 << 8 ) & 0x00ff0000) |
          ((rs1 >> 8 ) & 0x0000ff00) |
          ((rs1 << 24) & 0xff000000);
}
```

```
$ diff -U100 sw-6.1.0.s sw-7.1.0.s
--- sw-6.1.0.s	2017-08-18 09:55:32.000000000 +1200
+++ sw-7.1.0.s	2017-08-18 09:55:45.000000000 +1200
@@ -1,22 +1,23 @@
 	.file	"sw.c"
 	.option nopic
 	.text
-	.align	2
+	.align	1
 	.globl	sw
 	.type	sw, @function
 sw:
 	sllw	a3,a0,24
 	srlw	a5,a0,24
 	sllw	a4,a0,8
 	or	a5,a5,a3
 	li	a3,16711680
 	and	a4,a4,a3
 	or	a5,a5,a4
 	li	a4,65536
-	addw	a4,a4,-256
+	add	a4,a4,-256
 	srlw	a0,a0,8
 	and	a0,a0,a4
 	or	a0,a5,a0
+	sext.w	a0,a0
 	ret
 	.size	sw, .-sw
-	.ident	"GCC: (GNU) 6.1.0"
+	.ident	"GCC: (GNU) 7.1.0"
```

aswaterman, 2017-08-17:
I believe the addw->add change is a red herring, but I have an idea... I'll report back later.

michaeljclark, 2017-08-17:
No worries. Thanks.

aswaterman, 2017-08-17:
The problem is that lshiftrt isn't combining with sign_extend.  A much simpler example (rs1 >> 24) manifests it.  I can't figure out why, though.

michaeljclark, 2017-08-17:
It's interesting that right shift 24 manifests it given the other issue with that bit field sign extension pattern splits a shift 27 into shift 24 + shift 3. It could be in target independent code, if it is related, as that issue affects arm and x86 also. e.g. some target independent quirk/optimization that is causing the issue.

michaeljclark, 2017-08-17:
one way is to bisect between riscv-gcc-6.3 and riscv-gcc-7.1. I could do some grunt work like that... as we know that we have a working version, and that way we could isolate the changeset that regressed the compiler.

michaeljclark, 2017-08-17:
might very well be in target independent code.

michaeljclark, 2017-08-17:
sorry. riscv-gcc-6.1 and riscv-gcc-7.1. low precision day today.

michaeljclark, 2017-08-17:
I can dig out the commit id that has the good codegen. I guess then the problem is if it crops up after a big upstream merge, then it will be harder to bisect.

aswaterman, 2017-08-17:
I don't think that will help.  During upstreaming, we totally revamped the compiler to use TRULY_NOOP_TRUNCATION.  Bisecting will just say that's where the regression occurred.

The right way to fix this is different for TRULY_NOOP_TRUNCATION than without.

michaeljclark, 2017-08-18:
OK. It only happens on right shift.

https://cx.rv8.io/g/8QVG9h

x86 and aarch64 have register aliases for different register widths so potentially different code paths. It would be interesting to know whether it was target independent code or metadata. It's hard to know, given x86 and aarch64 use register aliases for truncation.

michaeljclark, 2017-08-18:
We should look at MIPS64 on compiler explorer.

aswaterman, 2017-08-18:
MIPS64 doesn't use TRULY_NOOP_TRUNCATION so it isn't helpful here.

aswaterman, 2017-08-18:
Actually, at least part of this regression is recent: on May 24 (16210e6270e200cd4892a90ecef608906be3a130) the simple ```rs1 >> 24``` case doesn't have a sext.w.  Can you bisect from there to current?

michaeljclark, 2017-08-18:
OK. Give me some time...

aswaterman, 2017-08-18:
thanks for the help

michaeljclark, 2017-08-18:
No worries. It's kind of interesting. The torture tests don't check for regressions in instruction counts. It's interesting to think how we could test for these types of regressions. If we had tests running all of the time, they would be easier to spot.

michaeljclark, 2017-08-18:
I think we might have to go back further... I'm seeing `sext.w` with that commit. 

```
$ git branch -a
* (HEAD detached at 16210e6270e)
$ git show
commit 16210e6270e200cd4892a90ecef608906be3a130
Author: Kito Cheng <kito.cheng@gmail.com>
Date:   Thu May 4 02:11:13 2017 +0800

    RISC-V: Handle non-legitimate address in riscv_legitimize_move
    
    GCC may generate non-legitimate address due to we allow some
    load/store with non-legitimate address in pic.md.
    
      2017-05-12  Kito Cheng  <kito.cheng@gmail.com>
    
          * config/riscv/riscv.c (riscv_legitimize_move): Handle
          non-legitimate address.
$ cat test.c 
unsigned rs24(unsigned rs1) { return rs1 >> 24; }
$ /opt/riscv/toolchain-16210e62/bin/riscv64-unknown-elf-gcc -O3 -c --save-temps test.c 
$ cat test.s 
	.file	"test.c"
	.option nopic
	.text
	.align	1
	.globl	rs24
	.type	rs24, @function
rs24:
	srlw	a0,a0,24
	sext.w	a0,a0
	ret
	.size	rs24, .-rs24
	.ident	"GCC: (GNU) 7.1.1 20170509"
```

michaeljclark, 2017-08-18:
@aswaterman  What date did the TRULY_NOOP_TRUNCATION change occur?

```
commit 6b7b37e4fe6ca9182a0ca51538a679b2c1617944 (HEAD)
Author: palmer <palmer@138bc75d-0d04-0410-961f-82ee72b054a4>
Date:   Mon Feb 6 21:39:18 2017 +0000

    Alphebetize RISC-V Options section
    
    2017-02-06  Palmer Dabbelt  <palmer@dabbelt.com>
    
            * docs/invoke.texi (RISC-V Options): Alphabetize.
    
    git-svn-id: svn+ssh://gcc.gnu.org/svn/gcc/trunk@245231 138bc75d-0d04-0410-961f-82ee72b054a4
```

```
$ cat test.s 
	.file	"test.c"
	.option nopic
	.text
	.align	1
	.globl	rs24
	.type	rs24, @function
rs24:
	srlw	a0,a0,24
	sext.w	a0,a0
	ret
	.size	rs24, .-rs24
	.ident	"GCC: (GNU) 7.0.1 20170206 (experimental)"
```

michaeljclark, 2017-08-18:
I have a feeling this is where this branch starts, with patch commits again gcc 7.0.1 mainline. The prior commits are the port commits.

michaeljclark, 2017-08-18:
So should we perhaps be looking at `simplify_truncation` in `gcc/simplify-rtx.c`, and figure out how the logic works to prove whether zero extend or sign extend should be no-ops,.

... however wouldn't then the add vs addw be significant?

The other question is how does TRULY_NOOP_TRUNCATION work with targets that don't have subregs? The use of a different width subreg (as on x86-64 and aarch64) makes the truncation a no-op. It seems one relies on use of subregs whereas RISC-V relies on use of instructions to make the NOOP truncation guarantee.

It makes one pause regarding the 'add', as the proof requires the use of "instructions" (versus "registers") that can't overflow beyond the width.

how big was the change? and what benefit did it bring? i.e. are there some test cases where it eliminates redundant sign extends versus adding them?

Thinking aloud...

michaeljclark, 2017-08-18:
Now I'm curious about whether one can see the current width mode in the RTL dumps, or run gdb and put some breakpoints or prints in `simplify_truncation` in `gcc/simplify-rtx.c` to see which clause is emitting the `sext.w`

michaeljclark, 2017-08-18:
```
$ riscv64-linux-musl-gcc -c -O3 -S -fdump-rtl-all shift.c
$ ls shift.*
```
...quite a lot of info...

what is interesting is this: _*(lshiftrt:SI (subreg/s/u:SI (reg/v:DI 73 [ rs1 ]) 0)*_ i.e. SI subreg DI reg if I am reading it correctly. going to see what the widths are for left shift

```
$ cat shift.c.229r.expand
;;
;; Full RTL generated for this function:
;;
(note 1 0 4 NOTE_INSN_DELETED)
(note 4 1 2 2 [bb 2] NOTE_INSN_BASIC_BLOCK)
(insn 2 4 3 2 (set (reg/v:DI 73 [ rs1 ])
        (reg:DI 10 a0 [ rs1 ])) "shift.c":1 -1
     (nil))
(note 3 2 6 2 NOTE_INSN_FUNCTION_BEG)
(insn 6 3 7 2 (set (reg:SI 74)
        (lshiftrt:SI (subreg/s/u:SI (reg/v:DI 73 [ rs1 ]) 0)
            (const_int 24 [0x18]))) "shift.c":1 -1
     (nil))
(insn 7 6 8 2 (set (reg:DI 75)
        (sign_extend:DI (reg:SI 74))) "shift.c":1 -1
     (nil))
(insn 8 7 12 2 (set (reg:DI 72 [ <retval> ])
        (reg:DI 75)) "shift.c":1 -1
     (nil))
(insn 12 8 13 2 (set (reg/i:DI 10 a0)
        (reg:DI 72 [ <retval> ])) "shift.c":1 -1
     (nil))
(insn 13 12 0 2 (use (reg/i:DI 10 a0)) "shift.c":1 -1
     (nil))
```

michaeljclark, 2017-08-18:
so the RTL in early expand for left and right shift appears identical, but left shift has the sign extend elided. curious... will have to see what stage it's eliminated from left shift...

```
$ cat rshift.c
unsigned rs24(unsigned rs1) { return rs1 >> 24; }
$ cat lshift.c
unsigned rs24(unsigned rs1) { return rs1 << 24; }
$ riscv64-linux-musl-gcc -O3 -S -fdump-rtl-all -c lshift.c -o lshift.s
$ riscv64-linux-musl-gcc -O3 -S -fdump-rtl-all -c rshift.c -o rshift.s
$ diff -U1000 lshift.c.229r.expand rshift.c.229r.expand
 ;;
 ;; Full RTL generated for this function:
 ;;
 (note 1 0 4 NOTE_INSN_DELETED)
 (note 4 1 2 2 [bb 2] NOTE_INSN_BASIC_BLOCK)
 (insn 2 4 3 2 (set (reg/v:DI 73 [ rs1 ])
-        (reg:DI 10 a0 [ rs1 ])) "lshift.c":1 -1
+        (reg:DI 10 a0 [ rs1 ])) "rshift.c":1 -1
      (nil))
 (note 3 2 6 2 NOTE_INSN_FUNCTION_BEG)
 (insn 6 3 7 2 (set (reg:SI 74)
-        (ashift:SI (subreg/s/u:SI (reg/v:DI 73 [ rs1 ]) 0)
-            (const_int 24 [0x18]))) "lshift.c":1 -1
+        (lshiftrt:SI (subreg/s/u:SI (reg/v:DI 73 [ rs1 ]) 0)
+            (const_int 24 [0x18]))) "rshift.c":1 -1
      (nil))
 (insn 7 6 8 2 (set (reg:DI 75)
-        (sign_extend:DI (reg:SI 74))) "lshift.c":1 -1
+        (sign_extend:DI (reg:SI 74))) "rshift.c":1 -1
      (nil))
 (insn 8 7 12 2 (set (reg:DI 72 [ <retval> ])
-        (reg:DI 75)) "lshift.c":1 -1
+        (reg:DI 75)) "rshift.c":1 -1
      (nil))
 (insn 12 8 13 2 (set (reg/i:DI 10 a0)
-        (reg:DI 72 [ <retval> ])) "lshift.c":1 -1
+        (reg:DI 72 [ <retval> ])) "rshift.c":1 -1
      (nil))
-(insn 13 12 0 2 (use (reg/i:DI 10 a0)) "lshift.c":1 -1
+(insn 13 12 0 2 (use (reg/i:DI 10 a0)) "rshift.c":1 -1
      (nil))
```

and

```
$ diff -U1000 lshift.s rshift.s 
--- lshift.s	2017-08-18 19:19:29.984319210 +1200
+++ rshift.s	2017-08-18 19:19:29.996319035 +1200
@@ -1,11 +1,12 @@
-	.file	"lshift.c"
+	.file	"rshift.c"
 	.option nopic
 	.text
 	.align	1
 	.globl	rs24
 	.type	rs24, @function
 rs24:
-	slliw	a0,a0,24
+	srliw	a0,a0,24
+	sext.w	a0,a0
 	ret
 	.size	rs24, .-rs24
 	.ident	"GCC: (GNU) 7.1.0"
```

aswaterman, 2017-08-18:
Sorry for sending you down the wrong path.  The commit I suggested definitely has the same code generation deficiency.  Not sure what I screwed up earlier today.

I'm not going to have much more time to look into this for a while, but if you find a promising lead, let me know.

michaeljclark, 2017-08-18:
No worries. At least we know for sure. I have other things to do, but had finished up some things so thought i would take a look at gcc codegen. A change is as good as a break. I'll keep it on my mind, look at it when I have time and will let you know if I find anything further...

michaeljclark, 2017-08-18:
It's actually a good bug to use as an opportunity to get to know gcc internals a little better...

kito-cheng, 2017-08-18:
`combine` pass is the most powerful magical black box in GCC back-end, maybe you can find something there :P

michaeljclark, 2017-08-18:
Combine looks very different between left and right shift. left shift instruction is deleted. assume that means it has already been lowered.

```
$ diff -U100  lshift.c.259r.combine  rshift.c.259r.combine
 (note 4 0 2 2 [bb 2] NOTE_INSN_BASIC_BLOCK)
 (note 2 4 3 2 NOTE_INSN_DELETED)
 (note 3 2 6 2 NOTE_INSN_FUNCTION_BEG)
-(note 6 3 7 2 NOTE_INSN_DELETED)
+(insn 6 3 7 2 (set (reg:SI 74)
+        (lshiftrt:SI (reg:SI 10 a0 [ rs1 ])
+            (const_int 24 [0x18]))) "rshift.c":1 148 {lshrsi3}
+     (expr_list:REG_DEAD (reg:DI 10 a0 [ rs1 ])
+        (nil)))
 (note 7 6 12 2 NOTE_INSN_DELETED)
 (insn 12 7 13 2 (set (reg/i:DI 10 a0)
-        (sign_extend:DI (ashift:SI (reg:SI 10 a0 [ rs1 ])
-                (const_int 24 [0x18])))) "lshift.c":1 152 {*ashlsi3_extend}
-     (nil))
-(insn 13 12 0 2 (use (reg/i:DI 10 a0)) "lshift.c":1 -1
+        (sign_extend:DI (reg:SI 74))) "rshift.c":1 86 {extendsidi2}
+     (expr_list:REG_DEAD (reg:SI 74)
+        (nil)))
+(insn 13 12 0 2 (use (reg/i:DI 10 a0)) "rshift.c":1 -1
      (nil))
```

When I add -dp I can see the ashlsi3_extend pattern selected for left shift but no associated extend pattern for right shift. Assume there should be a lshrsi3_extend in the metadata and its not being selected. Will take a look. 

Here is the asm with pattern match annotations:

```
rs24:
	srliw	a0,a0,24	# 6	lshrsi3	[length = 4]
	sext.w	a0,a0	# 12	extendsidi2/1	[length = 4]
	ret	# 19	simple_return	[length = 4]
	.size	rs24, .-rs24
	.ident	"GCC: (GNU) 7.2.0"
```

```
ls24:
	slliw	a0,a0,24	# 12	*ashlsi3_extend	[length = 4]
	ret	# 19	simple_return	[length = 4]
	.size	ls24, .-ls24
	.ident	"GCC: (GNU) 7.2.0"
```

Guess you've already been down this path and are already aware of this, however I am learning...

michaeljclark, 2017-08-18:
This is curious. There is a right shift only pattern followed by a pattern for all shifts. Why can't right shift use any_shift? It seems the any_shiftrt is going to take precedence? 

```
;; This code iterator allows the two right shift instructions to be
;; generated from the same template.
(define_code_iterator any_shiftrt [ashiftrt lshiftrt])

;; This code iterator allows the three shift instructions to be generated
;; from the same template.
(define_code_iterator any_shift [ashift ashiftrt lshiftrt])
```

aswaterman, 2017-08-18:
The iterators have no effect on code generation; they just allow multiple
patterns to be written as one pattern generator.

I don't remember what any_shiftrt is used for, but it's harmless.

michaeljclark, 2017-08-19:
OK. So it seems as @kito-cheng  mentioned, its 'combine' where the RTL diverges between left shift and right shift. The left shift instruction is deleted and the sign extension removed whereas it remains for the right shift. I'll have to learn more about combine.c. Glad that you reduced the test code further.

```
$ cat lshift.c
unsigned rs24(unsigned rs1) { return rs1 << 24; }
$ cat rshift.c
unsigned rs24(unsigned rs1) { return rs1 >> 24; }
```

michaeljclark, 2017-08-28:
I've made progress on isolating this. I made try_combine return 0 and the problem showed up later on in ree (Redundant Extension Elimination pass). Interestingly this pass has no shift logic at all which points at simplify_rtx as both combine.c and ree.c call simplify_rtx and simplify_rtx has shift logic and TRULY_NOOP_TRUNCATION logic. This points to simplify_rtx.c not eliminating the sign extension.

I still wonder why it is trying to change the mode from SI to DI in the set expressions. Do you think this is an artefact of TRULY_NOOP_TRUNCATION?

```
$ diff -u lshift.c.275r.split2 rshift.c.275r.split2 
--- lshift.c.275r.split2	2017-08-29 09:06:36.000000000 +1200
+++ rshift.c.275r.split2	2017-08-29 09:06:36.000000000 +1200
@@ -19,13 +19,13 @@
 (note 4 1 3 2 [bb 2] NOTE_INSN_BASIC_BLOCK)
 (note 3 4 6 2 NOTE_INSN_FUNCTION_BEG)
 (insn 6 3 7 2 (set (reg:SI 10 a0 [74])
-        (ashift:SI (reg:SI 10 a0 [orig:73 rs1 ] [73])
-            (const_int 24 [0x18]))) "lshift.c":1 146 {ashlsi3}
+        (lshiftrt:SI (reg:SI 10 a0 [orig:73 rs1 ] [73])
+            (const_int 24 [0x18]))) "rshift.c":1 148 {lshrsi3}
      (nil))
 (insn 7 6 13 2 (set (reg:DI 10 a0 [75])
-        (sign_extend:DI (reg:SI 10 a0 [74]))) "lshift.c":1 86 {extendsidi2}
+        (sign_extend:DI (reg:SI 10 a0 [74]))) "rshift.c":1 86 {extendsidi2}
      (nil))
-(insn 13 7 15 2 (use (reg/i:DI 10 a0)) "lshift.c":1 -1
+(insn 13 7 15 2 (use (reg/i:DI 10 a0)) "rshift.c":1 -1
      (nil))
 (note 15 13 16 NOTE_INSN_DELETED)
 (note 16 15 0 NOTE_INSN_DELETED)
```

```
$ diff -u lshift.c.276r.ree rshift.c.276r.ree 
--- lshift.c.276r.ree	2017-08-29 09:06:36.000000000 +1200
+++ rshift.c.276r.ree	2017-08-29 09:06:36.000000000 +1200
@@ -10,20 +10,20 @@
 df_worklist_dataflow_doublequeue: n_basic_blocks 3 n_edges 2 count 3 (    1)
 Trying to eliminate extension:
 (insn 7 6 13 2 (set (reg:DI 10 a0 [75])
-        (sign_extend:DI (reg:SI 10 a0 [74]))) "lshift.c":1 86 {extendsidi2}
+        (sign_extend:DI (reg:SI 10 a0 [74]))) "rshift.c":1 86 {extendsidi2}
      (nil))
 Tentatively merged extension with definition :
 (insn 6 3 7 2 (set (reg:DI 10 a0)
-        (sign_extend:DI (ashift:SI (reg:SI 10 a0 [orig:73 rs1 ] [73])
-                (const_int 24 [0x18])))) "lshift.c":1 -1
+        (zero_extend:DI (lshiftrt:SI (reg:SI 10 a0 [orig:73 rs1 ] [73])
+                (const_int 24 [0x18])))) "rshift.c":1 -1
      (nil))
-deferring rescan insn with uid = 6.
-All merges were successful.
-Eliminated the extension.
-deferring deletion of insn with uid = 7.
-Elimination opportunities = 1 realized = 1
+Merge cancelled, non-mergeable definitions:
+(insn 6 3 7 2 (set (reg:SI 10 a0 [74])
+        (lshiftrt:SI (reg:SI 10 a0 [orig:73 rs1 ] [73])
+            (const_int 24 [0x18]))) "rshift.c":1 148 {lshrsi3}
+     (nil))
+Elimination opportunities = 1 realized = 0
 starting the processing of deferred insns
-rescanning insn with uid = 6.
 ending the processing of deferred insns
 
 
@@ -37,16 +37,19 @@
 ;;  entry block defs 	 1 [ra] 2 [sp] 7 [t2] 10 [a0] 11 [a1] 12 [a2] 13 [a3] 14 [a4] 15 [a5] 16 [a6] 17 [a7] 42 [fa0] 43 [fa1] 44 [fa2] 45 [fa3] 46 [fa4] 47 [fa5] 48 [fa6] 49 [fa7]
 ;;  exit block uses 	 1 [ra] 2 [sp] 10 [a0]
 ;;  regs ever live 	 10 [a0]
-;;  ref usage 	r1={1d,1u} r2={1d,2u} r7={1d} r10={2d,3u} r11={1d} r12={1d} r13={1d} r14={1d} r15={1d} r16={1d} r17={1d} r42={1d} r43={1d} r44={1d} r45={1d} r46={1d} r47={1d} r48={1d} r49={1d} 
-;;    total ref usage 26{20d,6u,0e} in 2{2 regular + 0 call} insns.
+;;  ref usage 	r1={1d,1u} r2={1d,2u} r7={1d} r10={3d,4u} r11={1d} r12={1d} r13={1d} r14={1d} r15={1d} r16={1d} r17={1d} r42={1d} r43={1d} r44={1d} r45={1d} r46={1d} r47={1d} r48={1d} r49={1d} 
+;;    total ref usage 28{21d,7u,0e} in 3{3 regular + 0 call} insns.
 (note 1 0 4 NOTE_INSN_DELETED)
 (note 4 1 3 2 [bb 2] NOTE_INSN_BASIC_BLOCK)
 (note 3 4 6 2 NOTE_INSN_FUNCTION_BEG)
-(insn 6 3 13 2 (set (reg:DI 10 a0)
-        (sign_extend:DI (ashift:SI (reg:SI 10 a0 [orig:73 rs1 ] [73])
-                (const_int 24 [0x18])))) "lshift.c":1 152 {*ashlsi3_extend}
+(insn 6 3 7 2 (set (reg:SI 10 a0 [74])
+        (lshiftrt:SI (reg:SI 10 a0 [orig:73 rs1 ] [73])
+            (const_int 24 [0x18]))) "rshift.c":1 148 {lshrsi3}
+     (nil))
+(insn 7 6 13 2 (set (reg:DI 10 a0 [75])
+        (sign_extend:DI (reg:SI 10 a0 [74]))) "rshift.c":1 86 {extendsidi2}
      (nil))
-(insn 13 6 15 2 (use (reg/i:DI 10 a0)) "lshift.c":1 -1
+(insn 13 7 15 2 (use (reg/i:DI 10 a0)) "rshift.c":1 -1
      (nil))
 (note 15 13 16 NOTE_INSN_DELETED)
 (note 16 15 0 NOTE_INSN_DELETED)
```

michaeljclark, 2017-08-28:
I looked at the RTL from riscv-gcc 6.1

```
$ /opt/riscv/toolchain-master/bin/riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 6.1.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

and there is no promotion from SI mode to DI mode.

```
$ diff -u lshift.c.213r.expand rshift.c.213r.expand 
--- lshift.c.213r.expand	2017-08-29 09:35:41.000000000 +1200
+++ rshift.c.213r.expand	2017-08-29 09:35:41.000000000 +1200
@@ -26,18 +26,18 @@
 (note 1 0 4 NOTE_INSN_DELETED)
 (note 4 1 2 2 [bb 2] NOTE_INSN_BASIC_BLOCK)
 (insn 2 4 3 2 (set (reg/v:SI 73 [ rs1 ])
-        (reg:SI 10 a0 [ rs1 ])) lshift.c:1 -1
+        (reg:SI 10 a0 [ rs1 ])) rshift.c:1 -1
      (nil))
 (note 3 2 6 2 NOTE_INSN_FUNCTION_BEG)
 (insn 6 3 7 2 (set (reg:SI 74)
-        (ashift:SI (reg/v:SI 73 [ rs1 ])
-            (const_int 24 [0x18]))) lshift.c:1 -1
+        (lshiftrt:SI (reg/v:SI 73 [ rs1 ])
+            (const_int 24 [0x18]))) rshift.c:1 -1
      (nil))
 (insn 7 6 11 2 (set (reg:SI 72 [ <retval> ])
-        (reg:SI 74)) lshift.c:1 -1
+        (reg:SI 74)) rshift.c:1 -1
      (nil))
 (insn 11 7 12 2 (set (reg/i:SI 10 a0)
-        (reg:SI 72 [ <retval> ])) lshift.c:1 -1
+        (reg:SI 72 [ <retval> ])) rshift.c:1 -1
      (nil))
-(insn 12 11 0 2 (use (reg/i:SI 10 a0)) lshift.c:1 -1
+(insn 12 11 0 2 (use (reg/i:SI 10 a0)) rshift.c:1 -1
      (nil))
```

aswaterman, 2017-08-28:
Yes, it's an artifact of TRULY_NOOP_TRUNCATION.  In GCC 6.1 we used a different TRULY_NOOP_TRUNCATION discipline, which is why there was no promotion to DImode there.  However, it would be incorrect to revert to the old definition of TRULY_NOOP_TRUNCATION.

aswaterman, 2017-08-28:
Sorry, I meant to say it's an artifact of PROMOTE_MODE.

michaeljclark, 2017-08-28:
OK. I see now. It's problematic that it's trying to sign extend during the promotion from SI to DI when the sign extension is essentially free. i.e. is the canonical form. So then isn't it the code that is inserting the sign_extend into the set expression at fault?

```
/* Define this macro if it is advisable to hold scalars in registers
   in a wider mode than that declared by the program.  In such cases,
   the value is constrained to be within the bounds of the declared
   type, but kept valid in the wider mode.  The signedness of the
   extension may differ from that of the type.  */

#define PROMOTE_MODE(MODE, UNSIGNEDP, TYPE)     \
  if (GET_MODE_CLASS (MODE) == MODE_INT         \
      && GET_MODE_SIZE (MODE) < UNITS_PER_WORD) \
    {                                           \
      if ((MODE) == SImode)                     \
        (UNSIGNEDP) = 0;                        \
      (MODE) = word_mode;                       \
    }
```

michaeljclark, 2017-08-28:
It seems like eliminating redundant sign extension is not at fault and instead papers over the issue, rather the insertion of the sign extension is at fault, given it is redundant.

Thanks for explaining. I've got a clearer picture now.

aswaterman, 2017-08-28:
I'd say "cause" but not "fault"... that code is necessary for correctness.  If you change it to not promote SImode to DImode, it'll fail to meet the ABI constraint when variables aren't in canonical form.  For example, if you pass a long to a function that takes an int, that is the code that is responsible for inserting the sext.w.

michaeljclark, 2017-08-28:
OK. long to int is a demotion which is distinct from a value being held "in a wider mode than that declared by the program". It could be sign extended on demotion. I guess riscv-gcc-6.1 must have had some failing tests, however codegen with riscv-gcc-7.1 seems to have regressed.

The rhetorical question is what ops on narrower types can allow the register to be in non-canonical form, but that implies an expense on those operations. Nevermind...

So it seems we need to match the SI-mode shift (or any 32-bit instruction) followed by set with sign-extend to DI-mode. Both of the shifts are 32-bit so we know the value is in canonical form in this case...

michaeljclark, 2017-08-28:
Explains why we see `add` instead of `addw` in assembly diffs between gcc-6.1 and gcc-7.1

aswaterman, 2017-08-28:
One thing I'm still not clear on is whether this is REE's job to know that the value has been canonicalized so the sext.w can be deleted.  Whose fault is this, really?

michaeljclark, 2017-08-28:
I think it is simplify-rtx.c where ASHIFT plus sign-extend is simplified...? I'm trying to find the logic... When I switched off combine, the same elimination happened later in REE and both call simply_rtx. simply_rtx seems to be the common element, and it has shift and sign extension patterns. Reading the code...

if it sees any SI mode shift followed by sign_extend promotion to DI mode, the sign_extend can be eliminated.

michaeljclark, 2017-08-28:
Found it. This code prevents the sign extension from being eliminated.

```
diff --git a/gcc/simplify-rtx.c b/gcc/simplify-rtx.c
index ce632ae..3261ad4 100644
--- a/gcc/simplify-rtx.c
+++ b/gcc/simplify-rtx.c
@@ -1500,12 +1500,14 @@ simplify_unary_operation_1 (enum rtx_code code, machine_mode mode, rtx op)
            }
        }
 
+#if 0
       /* (sign_extend:M (lshiftrt:N <X> (const_int I))) is better as
          (zero_extend:M (lshiftrt:N <X> (const_int I))) if I is not 0.  */
       if (GET_CODE (op) == LSHIFTRT
          && CONST_INT_P (XEXP (op, 1))
          && XEXP (op, 1) != const0_rtx)
        return simplify_gen_unary (ZERO_EXTEND, mode, op, GET_MODE (op));
+#endif
 
 #if defined(POINTERS_EXTEND_UNSIGNED)
       /* As we do not know which address space the pointer is referring to,
```

Look at this:

```
$ more rshift.s 
        .file   "rshift.c"
        .option nopic
        .text
        .align  1
        .globl  rs24
        .type   rs24, @function
rs24:
        srlw    a0,a0,24
        ret
        .size   rs24, .-rs24
        .ident  "GCC: (GNU) 7.1.1 20170509"
```

palmer-dabbelt, 2017-08-28:
Thanks for tracking this down.

michaeljclark, 2017-08-28:
That code is probably preventing a more general simplification that happens later. I wonder what target context we have at this point.

aswaterman, 2017-08-28:
Interesting.  I have no idea what the right fix is.

michaeljclark, 2017-08-28:
It looks like an X86 hack in target independent code because zero extension is cheaper on X86.

#define SIGN_EXTENSION_IS_CHEAPER_THAN_ZERO_EXTENSION 1

:-D

michaeljclark, 2017-08-28:
because logical right shift is always going to shift a zero bit into the sign.

aswaterman, 2017-08-28:
Well that sucks.

aswaterman, 2017-08-28:
Maybe use PROMOTE_MODE to see if unsigned-SImode gets promoted to signed-SImode and disable this "optimization" if so?

aswaterman, 2017-08-28:
That would do the right thing for RISC-V, and it would preserve current behavior on x86 and AArch64.

michaeljclark, 2017-08-28:
Yes. That sounds like it would work.

It sounds safe and correct given promotion to signed SImode

michaeljclark, 2017-08-28:
This looks like the test to elide the sign extension, so it's clause would be the ones to use?

```
      /* Check for a sign extension of a subreg of a promoted
	 variable, where the promotion is sign-extended, and the
	 target mode is the same as the variable's promotion.  */
      if (GET_CODE (op) == SUBREG
	  && SUBREG_PROMOTED_VAR_P (op)
	  && SUBREG_PROMOTED_SIGNED_P (op)
	  && GET_MODE_SIZE (mode) <= GET_MODE_SIZE (GET_MODE (XEXP (op, 0))))
	{
	  temp = rtl_hooks.gen_lowpart_no_emit (mode, op);
	  if (temp)
	    return temp;
	}
```

aswaterman, 2017-08-28:
This is well beyond my understanding of the middle-end, sorry :)

michaeljclark, 2017-08-28:
Oh well. Before and after torture tests and some upstream patch review. I'll see what I can do...

michaeljclark, 2017-08-29:
That code is before the optimisation that causes the issue so it's clauses don't trigger, but it gives some indication of how to check the promoted sign... I will experiment...

michaeljclark, 2017-08-29:
One of us should send to gcc-patches and double-check if the approach is okay, but I think it is low risk given they are avoiding sign-extension on architectures that naturally zero-extend given they know a right shift above 0 with a zero in bit 31 for SI mode.

aswaterman, 2017-08-29:
Palmer and I are both working on a pressing deadline, so we'd appreciate it if you can talk to the GCC folks about this patch.  If you want, we can give you feedback on a draft of the email before you send it to the list.

[33 more replies]