riscvarchive/riscv-gcc #85: Movmem patterns in riscv.md

iz-sc, 2017-08-03:
Hi all!

Sorry if this already was discussed but I haven't managed to find any related discussions.
We've noticed that currently the latest compiler doesn't have momvem patterns which means eg any memcpy is expanded in library call. 
Is this intented for some reasons or it will be added at some moment?

Thanks in advance,
Igor

palmer-dabbelt, 2017-08-03:
https://github.com/riscv/riscv-gcc/pull/67

Some of the tests fail

iz-sc, 2017-08-09:
Ah, sorry, haven't seen this in the issue list with default filters.

Any plans about handling it?

palmer-dabbelt, 2017-08-10:
Sure: the plan is to fix it -- there's just no timeframe on that :).  This manifests as a performance regression in Dhrystone on rv32, so I'll probably get around to it at some point.  Since it's not a correctness issue it's lower priority than getting Linux and glibc upstream, so I don't think it'll happen all that soon.

iz-sc, 2017-08-15:
>  there's just no timeframe on that :). it's lower priority than getting Linux and glibc upstream
Thanks, I see :)

>  This manifests as a performance regression in Dhrystone on rv32
Isn't it a regression on 64 bits also?

Please, just one more question about failing tests - are they performance regression or correctness tests?

palmer-dabbelt, 2017-09-08:
Dhrystone doesn't manifest the performance regression on rv64 in as noticeable of a manner, as with the wider registers GCC will inline longer memcpy instances.

IIRC the failures are correctness related.

jim-wilson, 2018-01-11:
Apparently fixed in November when pull request #67 was merged.

closed 2018-01-11 by jim-wilson