riscvarchive/riscv-gcc #86: some   error for  riscv-gcc

Pengnian, 2017-08-06:
Hi,all 
when  I  want to compile  my  program by riscv-toolchain  some errors happened ,please experts explanation !  thanks ~
[image]
[image]

kito-cheng, 2017-08-06:
-march= only accept lowercase letter now, so you could try -march=rv64ima instead of -march=RV64IMA.

-msoft-float is an legacy option which is already removed now, you need specify by -mabi=, I suggest you can use -mabi=lp64 in this case; Here is more detail if you have interesting: argument could begin with ilp32 or lp64 for rv32 or rv64, and it could be suffix with f, d or no suffix, its represent for hard float or soft float abi which mean pass floating point value.

Sorry for such non-backward compatible changes, we promise there is no further non-backward compatible changes.

Pengnian, 2017-08-07:
@kito-cheng 
Oh,Thanks kito

Your explanation very useful,thanks you again !

kito-cheng, 2017-08-07:
:)

closed 2017-08-07 by kito-cheng