riscvarchive/riscv-gcc #97: Bug: Invalid RV32C Instruction

coldney, 2017-09-23:
When I compile the code like this:

.c
```
extern u32 __text_size;
__attribute__((noinline))
void test(int n1, int n3)
{
    asm("nop");
}

AT(.init_text)
int main(void)
{
    test(0, (u32)&__text_size); 
    return 0;
}
```
.ld:
`__text_size= SIZEOF(.comm);`

the disassembly listing looks like:

```
10000012 <main>:
10000012:	6581                	lui	a1,0x0
10000014:	1171                	addi	sp,sp,-4
10000016:	4501                	li	a0,0
10000018:	60000593          	li	a1,1536
1000001c:	c006                	sw	ra,0(sp)
1000001e:	f0020317          	auipc	t1,0xf0020
10000022:	1a0300e7          	jalr	416(t1) # 201be <test>
10000026:	4082                	lw	ra,0(sp)
10000028:	4501                	li	a0,0
1000002a:	0111                	addi	sp,sp,4
1000002c:	8082                	ret
```

Notice the first instruction:
`lui	a1,0x0`

in the RISCV-SPEC:

> C.LUI is only valid when rd6={x0 , x2}, and when the immediate is not equal to zero. 

So, it's an invalid instruction.

aswaterman, 2017-09-24:
This is a linker bug, not a gcc bug, it turns out.  I've proposed a fix here: https://github.com/riscv/riscv-binutils-gdb/pull/105

coldney, 2017-09-24:
@aswaterman 
Thank you! 

I've used this fix, and now there's a new link problem:
```
relocation truncated to fit: R_RISCV_RVC_LUI against symbol `__text_size' defined in *ABS* section in 
```
What could be the cause of this?

aswaterman, 2017-09-24:
Can you provide the files we need to to replicate the problem (github gist or something)? @palmer-dabbelt can take a look next week.

coldney, 2017-09-24:
https://gist.github.com/coldney/1860ec279fde1791dc4518dde05ddc59

iz-sc, 2017-10-05:
Seems the check VALID_RVC_LUI_IMM works incorrectly.
At least for the following pair (from main.o)
00000000 <main>:
   0:   00000537                lui     a0,0x0
                        0: R_RISCV_ALIGN        *ABS*
                        0: R_RISCV_HI20 __comm_size
                        0: R_RISCV_RELAX        *ABS*

00000004 <.L0 >:
   4:   1141                    addi    sp,sp,-16

00000006 <.L0 >:
   6:   00050513                mv      a0,a0
                        6: R_RISCV_LO12_I       __comm_size
                        6: R_RISCV_RELAX        *ABS*

aswaterman, 2017-10-05:
I think that is the bug I already fixed.

iz-sc, 2017-10-05:
Can you please elaborate which one?

iz-sc, 2017-10-11:
This patch seems to fix the problem

diff --git a/bfd/elfnn-riscv.c b/bfd/elfnn-riscv.c
index 87fd46f..39bdbc5 100644
--- a/bfd/elfnn-riscv.c
+++ b/bfd/elfnn-riscv.c
@@ -2929,6 +2929,9 @@ _bfd_riscv_relax_lui (bfd *abfd,
       if (((lui >> OP_SH_RD) & OP_MASK_RD) == X_SP)
        return TRUE;
 
+      if (EXTRACT_RVC_LUI_IMM (ENCODE_RVC_LUI_IMM(symval)) == 0)
+       return TRUE;
+
       lui = (lui & (OP_MASK_RD << OP_SH_RD)) | MATCH_C_LUI;
       bfd_put_32 (abfd, lui, contents + rel->r_offset);
 
and it's not quite clear why  RISCV_CONST_HIGH_PART is used in "if" before...

palmer-dabbelt, 2017-11-16:
I think this was fixed by 3342be5dabeeaf2218dfbf4d38f92214612436f4

palmer-dabbelt, 2017-11-16:
On the release branch it's https://github.com/riscv/riscv-binutils-gdb/commit/d51d92c4d06b6f1b8fee12041cf882e025b84400#diff-192022aebaf71b5fb118e5d01e234659

closed 2017-11-16 by palmer-dabbelt