riscvarchive/riscv-gcc #100: Compiler bug when running riscv64-gcc on target

martinmaas, 2017-09-26:
Running riscv64 gcc on linux-riscv results in an internal compiler error, even for a minimal program:

```
#include <stdio.h>
int main() { printf("Hello World\n"); return 0; }
```

The error that is shown is:

```
In file included from /usr/include/bits/types.h:130:0,
                 from /usr/include/stdio.h:35,
                 from test.c:1:
/usr/include/bits/types.h:143:12: internal compiler error: in grokdeclarator, at c/c-decl.c:6769
 __STD_TYPE __FSID_T_TYPE __fsid_t; /* Type of file system IDs.  */
            ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
```

This was with the versions of the tools from abe63db64e56f42b53a38cb72f8b8eb0c8b3550f in [riscv-poky](https://github.com/riscv/riscv-poky/commit/abe63db64e56f42b53a38cb72f8b8eb0c8b3550f). The GCC version is 7.2.0 from upstream.

rwmjones, 2017-12-27:
I was about to file a very similar bug.  In my case I can reproduce it with a very simple test case (adapted from an autoconf test which is why it looks peculiar):
```
#include <sys/types.h>
int
main (void)
{
  if (sizeof (size_t)) return 0;
  return 0;
}
```
When compiled with riscv-gcc @ b731149757b93ddc80e6e4b5483a6931d5f9ad60 I see the following internal error:
```
# gcc -c test.c
In file included from /usr/include/bits/pthreadtypes.h:23:0,
                 from /usr/include/sys/types.h:254,
                 from test.c:1:
/usr/include/bits/thread-shared-types.h:139:3: internal compiler error: in grokdeclarator, at c/c-decl.c:6769
   };
   ^
0x11d6d5 grokdeclarator
	../../gcc/c/c-decl.c:6768
0x11e3f5 grokfield(unsigned int, c_declarator*, c_declspecs*, tree_node*, tree_node**)
	../../gcc/c/c-decl.c:7606
0x150ae1 c_parser_struct_declaration
	../../gcc/c/c-parser.c:3107
0x150d29 c_parser_struct_declaration
	../../gcc/c/c-parser.c:3063
0x13ef0f c_parser_struct_or_union_specifier
	../../gcc/c/c-parser.c:2970
0x13ef0f c_parser_declspecs(c_parser*, c_declspecs*, bool, bool, bool, bool, bool, c_lookahead_kind)
	../../gcc/c/c-parser.c:2553
0x13a797 c_parser_declaration_or_fndef
	../../gcc/c/c-parser.c:1647
0x1519f3 c_parser_external_declaration
	../../gcc/c/c-parser.c:1468
0x15230d c_parser_translation_unit
	../../gcc/c/c-parser.c:1349
0x15230d c_parse_file()
	../../gcc/c/c-parser.c:18176
0x185ecf c_common_parse_file()
	../../gcc/c-family/c-opts.c:1107
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
```
The file where it is failing contains:
```
    struct
    {
      unsigned int __low;
      unsigned int __high;
    } __wseq32;
  };                       <<< line 139
  __extension__ union
  {
    __extension__ unsigned long long int __g1_start;
    struct
    {
      unsigned int __low;
      unsigned int __high;
    } __g1_start32;
  };
  unsigned int __g_refs[2] __LOCK_ALIGNMENT;
  unsigned int __g_size[2];
  unsigned int __g1_orig_size;
  unsigned int __wrefs;
  unsigned int __g_signals[2];
};

#endif /* _THREAD_SHARED_TYPES_H  */
```

rwmjones, 2017-12-27:
Adding -save-temps changes the apparent location of the bug, go figure.  Anyway I placed the preprocessed source here:
http://oirase.annexia.org/tmp/test.i
```
In file included from /usr/include/sys/types.h:197:0,
                 from test.c:1:
/usr/include/sys/select.h:67:5: internal compiler error: in grokdeclarator, at c/c-decl.c:6639
     __fd_mask __fds_bits[__FD_SETSIZE / __NFDBITS];
     ^~~~~~~~~
0x11cdd1 grokdeclarator
	../../gcc/c/c-decl.c:6638
```

jim-wilson, 2017-12-28:
On Wed, Dec 27, 2017 at 2:11 PM, rwmjones <notifications@github.com> wrote:

> Adding -save-temps changes the apparent location of the bug, go figure.
> Anyway I placed the preprocessed source here:
> http://oirase.annexia.org/tmp/test.i
>
> In file included from [...]
It is the same assert in two places, one check for typenames and one check
for structure/union fields.  If this is some kind of memory corruption bug,
it could be triggering differently depending on exactly how the compiler is
run.

I think the problem here is that your native gcc was miscompiled by the
cross gcc.  Or maybe it is a linux kernel problem, but my first guess would
be a compiler bug.  Either way, I don't have any way to reproduce the
failure currently, so I don't think I can do anything about it.  This kind
of problem is very difficult to debug remotely.

Jim

jim-wilson, 2017-12-28:
On Wed, Dec 27, 2017 at 7:18 PM, Jim Wilson <jimw@sifive.com> wrote:
> On Wed, Dec 27, 2017 at 2:11 PM, rwmjones <notifications@github.com> wrote:
>>
>> Adding -save-temps changes the apparent location of the bug, go figure.
>> Anyway I placed the preprocessed source here:
>> http://oirase.annexia.org/tmp/test.i

Actually, you could try playing with gcc options.  You could try
building a native compiler with -O1 instead of -O2 for instance, to
see if that fixes the problem, and if not, you can try with -O0.  If
you can identify an option that builds a working compiler, then you
can try bisecting across gcc source files, compiling half with the
good option and half with the bad option, repeating until you have it
down to one file.  Then ideally down to one function in the file being
miscompiled, maybe using pragmas to control optimization within a
file.  But that is a lot of work for a non-compiler person to do, and
it isn't clear if it is even worthwhile for you to try.

Jim

rwmjones, 2017-12-28:
OK that's a good idea.  FWIW I have another "weird" bug which might be caused by a similar miscompilation issue (https://github.com/riscv/riscv-gcc/issues/114).

palmer-dabbelt, 2017-12-28:
I'm trying to bring up enough of a userspace locally that I can run the glibc test suite, which will also probably be enough to run GCC locally.  IIRC you're not the only person who's run into the issue, one of the Debian guys did as well.

If I had to guess, I'd say it's either a GCC or a QEMU bug.  Do you have hardware?  If not, you could try -singlestep in QEMU (though that will be slow).

rwmjones, 2017-12-28:
I don't have hardware.

I cross-compiled GCC with `-O0` and ... it fixed it!  And also issue #114 too.  That was quite unexpected.

This is for the (bootstrap) stage 3 environment which in a sense is not something we care about beyond the fact that it can compile the first round of stage 4 RPMs.  In stage 4 we will be compiling GCC natively and then I can see if either of these bugs manifest ‘for real’.

palmer-dabbelt, 2017-12-28:
OK, well, I anticipate that this will manifest again soon so let's leave this bug open :).  If you get to a point where it manifests again and you can get me the stuff necessary to run on hardware then I can give it a shot.

rwmjones, 2017-12-29:
Here's an interesting thing: When bootstrapping Perl (which requires Perl), we compile a thing called `miniperl`.  This was segfaulting the first time it was run, but I reduced the optimization level from `-O2` -> `-O0` (and no other change) it starts working.  I don't have much further information because debugging anything inside the stage 3 environment is an exercise in frustration.

jim-wilson, 2018-01-13:
Reproduced using rwmjones's fedora disk image and testcase, and a canadian cross built native riscv-linux compiler.  I noticed that on the FSF gcc-7-branch, the top of the branch worked, but the gcc-7_2_0-release tag failed.  This is a common ancestor with our riscv-gcc tree.  So I did a git bisect over the FSF gcc-7-branch, narrowed down the problem to one patch, and confirmed that this patch backported to our riscv-gcc tree solved the problem for the testcase.  The patch we need is from this bug report
    https://gcc.gnu.org/bugzilla/show_bug.cgi?id=81803

I haven't done any rigorous testing of the patch in our riscv-gcc yet.  I only confirmed that it works for the testcase.  Proper testing for the patch will come next.

palmer-dabbelt, 2018-01-13:
Thanks, Jim!  I checked my Gentoo build and it doesn't appear to include this patch, so it's still a mystery as to why it WFM.  I'm assigning this bug to you.

jim-wilson, 2018-01-16:
I looked at code generated for grokdeclarator using unpatched and patched compilers, and verified that the lra bug fix is fixing the right problem.  The lra bug fix has been cherry picked onto the riscv-gcc-7.2.0 branch and pushed.

closed 2018-01-16 by jim-wilson