riscvarchive/riscv-gcc #109: objcopy does not correctly break data into bytes

pentin-as, 2017-11-17:
I am trying to decompose the elf file data into different 8-bit RAM chips, which together form a 64-bit data bus. For this, I use "riscv64-unknown-elf-objcopy -I elf64-littleriscv -O verilog --interleave = 8 --interleave-width = 1 --byte = 0". But when the elf file starts with addresses that are not 8 bytes long, an error occurs.

In the attachment, an example of a test.

dump:
0000000000000000 <.a0>:
   0: 00000013 nop
   4: 00000033 add zero, zero, zero
Disassembly of section .a1:
000000000000000c <_start + 0xc>:
   c: 00000013 nop
  10: 00000033 add zero, zero, zero

I expect that the file test0.hex will:
@ 00000000
13
@ 00000002
33
... in the file test4.hex will be:
@ 00000000
33
@ 00000001
13
... in fact in the file test0.hex:
@ 00000000
13
@ 00000001
13
... in the file test4.hex:
@ 00000000
33
@ 00000001
33

[test.zip](https://github.com/riscv/riscv-gcc/files/1481300/test.zip)

jim-wilson, 2017-11-17:
Objcopy fails when a section address is not an exact multiple of the interleave.  I have a quick hack that seems to fix it, but it needs a little more testing (e.g. with address with high bit set), and needs a proper binutils regression testcase added to it.
[objcopy-interleave.txt](https://github.com/riscv/riscv-gcc/files/1483655/objcopy-interleave.txt)

pentin-as, 2017-11-19:
Thank you very much. It's work.
P.S. Sorry that I created Issue not in that project. It was necessary in riscv-binutils-gdb.

jim-wilson, 2017-12-06:
The patch is in the FSF binutils tree.
https://sourceware.org/ml/binutils-cvs/2017-12/msg00050.html
I can backport to the riscv-binutils-gdb tree if necessary.

closed 2017-12-06 by jim-wilson