riscvarchive/riscv-gcc #111: Why is the last register ignored?

leinlawun, 2017-12-02:
https://github.com/riscv/riscv-gcc/blob/b731149757b93ddc80e6e4b5483a6931d5f9ad60/gcc/config/riscv/riscv.c#L3363
https://github.com/riscv/riscv-gcc/blob/b731149757b93ddc80e6e4b5483a6931d5f9ad60/gcc/config/riscv/riscv.c#L3451
https://github.com/riscv/riscv-gcc/blob/b731149757b93ddc80e6e4b5483a6931d5f9ad60/gcc/config/riscv/riscv.c#L3579

jim-wilson, 2017-12-02:
Looks like a typo.  I don't have any old versions of the risc-v gcc port, so I can only speculate how we got here.  For instance, maybe this was FP_REG_FIRST-1 originally, and someone forgot to remove the -1 when GP_REG_LAST was added.  Anyways, this appears to be a harmless bug, as x31 is a call clobbered reg, so frame.mask will never be set for it, and hence it will never be saved/restored.

While looking at this, I noticed another typo.  In riscv.h, the CALL_USED_REGS comment says
/* a0-a7, t0-a6, fa0-fa7, and ft0-ft11 are volatile across calls.               
   The call RTLs themselves clobber ra.  */
That should be t0-t6 on the first line.

I can fix both of these on Monday.

leinlawun, 2017-12-02:
I modified GCC for academic purposes, in particular, added an attribute that allows you to dispense with assembler inserts when switching stack / save and context loading in interrupts, and in my case, this error can lead to context corruption due to ignoring t6.

palmer-dabbelt, 2017-12-02:
Thanks Jim.  I'm going to assign this to you, just so it doesn't get lost.

jim-wilson, 2017-12-04:
The typos have been fixed in the FSF GCC tree.  Do you need this patch on one of the riscv-gcc branches?

closed 2017-12-11 by jim-wilson