riscvarchive/riscv-gcc #114: error: using result of function returning 'void'

rwmjones, 2017-12-28:
There's something very strange going on with this compiler build, but I can often get it to give an error which makes no sense:
```
libtool: compile:  gcc -g -O2 -DDEBUG -funsigned-char -fno-strict-aliasing -Wall -DVERSION=\"2.4.47\" -DLOCALEDIR=\"/usr/share/locale\" -DPACKAGE=\"attr\" -I../include -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -include libattr.h -c libattr.c  -fPIC -DPIC -o .libs/libattr.o
libattr.c: In function 'attr_list':
libattr.c:280:2: error: using result of function returning 'void'
  bzero(buffer, sizeof(attrlist_t));
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libattr.c: In function 'attr_listf':
libattr.c:326:2: error: using result of function returning 'void'
  bzero(buffer, sizeof(attrlist_t));
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
In [the code of libattr.c](https://github.com/philips/attr/blob/1cc88bd4c17ef99ace22c8be362d513f155b1387/libattr/libattr.c#L280) you can see that the result of bzero is clearly not being used.

I'm using riscv-gcc @ b731149757b93ddc80e6e4b5483a6931d5f9ad60, cross-compiled into the RISC-V environment, and then running it inside this environment.

rwmjones, 2017-12-28:
This error happens in lots of other builds too, also in places where it makes no obvious sense.

rwmjones, 2017-12-28:
This is likely fixed by cross-compiling riscv-gcc with -O0.  See comments https://github.com/riscv/riscv-gcc/issues/100#issuecomment-354226236 and following.

palmer-dabbelt, 2017-12-28:
I'm going to go ahead and close this one, as it's almost certainly the same bug as #100.  Since there's a lot more discussion there, let's just keep it to one thread.

Thanks for looking into this!

closed 2017-12-28 by palmer-dabbelt