riscvarchive/riscv-gcc #121: Missed CSE opportunity

aswaterman, 2018-01-31:
This example (extracted from coremark) results in the constant ```0x4002``` being loaded into a register twice, once for each inlined copy of ```crcu8```.  It also happens on AArch64, and a different variant makes it manifest on AArch32, so it seems to be a missed opportunity in the middle end.

    static uint16_t crcu8(uint8_t data, uint16_t crc )
    {
      uint8_t i=0,x16=0,carry=0;

      x16 = (uint8_t)((data & 1) ^ ((uint8_t)crc & 1));

      if (x16 == 1)
        crc ^= 0x4002;
      return crc;
    }

    uint16_t crcu16(uint16_t newval, uint16_t crc) {
      crc=crcu8(newval     ,crc);
      crc=crcu8(newval>>8  ,crc);
      return crc;
    }

Not at all pressing, but possibly worth looking into, or worth reporting to the GCC mailing list?

kito-cheng, 2018-01-31:
It seems more like Partial-PRE[1-2] issue instead of CSE?

[1] https://en.wikipedia.org/wiki/Partial_redundancy_elimination
[2] Enabled by -ftree-partial-pre in GCC

kito-cheng, 2018-01-31:
Some thought, I guess because PPRE implemented on Gimple layer, and RTL's one only implement PRE without PPRE, 0x4002 seems like a cheep constant during Gimple layer, so PPRE do nothing for that.