riscvarchive/riscv-gcc #125: ANDI seems to take only 8 bits

rishikhan, 2018-03-13:
The RISCV spec 2.2. says:

> ANDI, ORI, XORI are logical operations that perform bitwise AND, OR, and XOR on register rs1 and the sign-extended 12-bit immediate and place the result in rd.

This code fails
```
int main(int argc, char** argv)
{
  __asm__ volatile (
     "andi %0, %0, 0xfff"
    : "=r" (argc)
  );

  return (int ) argc;
}
```
How do we get 12 bits?

jim-wilson, 2018-03-13:
__asm__ volatile (
     "andi %0, %0, -1"
    : "=r" (argc)
   );
It is a sign extended value so the range is from -2048 to 2047.  0xffffffff will also work for a rv32 compiler, but not a rv64 compiler, but that isn't a 64-bit -1.

rishikhan, 2018-03-13:
Thank you! That was the issue. I guess negative numbers don't make much sense in hexadecimal. I was trying to find a way to set bits and didn't realize it had to be converted into small integer to work.

closed 2018-03-13 by rishikhan