riscvarchive/riscv-gcc #126: Possible operand modes missing

rishikhan, 2018-03-14:
When compiling riscv-gcc, I saw the following warnings:
../.././riscv-gcc/gcc/config/riscv/riscv.md:1954:1: warning: operand 0 missing mode?
../.././riscv-gcc/gcc/config/riscv/riscv.md:1976:1: warning: operand 1 missing mode?
../.././riscv-gcc/gcc/config/riscv/riscv.md:1999:1: warning: operand 0 missing mode?
../.././riscv-gcc/gcc/config/riscv/riscv.md:2022:1: warning: operand 1 missing mode?
../.././riscv-gcc/gcc/config/riscv/riscv.md:2088:1: warning: source missing a mode?

I don't know if they are problems, but wanted to bring them up to you. Are these OK?

palmer-dabbelt, 2018-03-14:
One of the first things I did when working on the RISC-V port was attempt to fix these warnings, but I couldn't figure it out so I gave up.  Maybe you'll have better luck :)

jim-wilson, 2018-03-14:
They are unimportant.  A lot of ports get warnings like this.  As long as the missing modes are in patterns where the mode will be specified elsewhere, it isn't a problem, and this is true for the RISC-V port.  This is fixable, but will require some work, and may require adding additional patterns.  We have enough other things to worry about that I haven't had a chance to look at this.

We already have a bug report for this in
    https://github.com/riscv/riscv-gnu-toolchain/issues/180

closed 2018-03-20 by rishikhan