riscvarchive/riscv-gcc #127: LR/SC gcc builtins

rishikhan, 2018-03-20:
Is there a way to have GCC emit a LD and SC assembly based on some builtins or macros? Here is my scenario. I'm trying to implement a semaphore and skip the extra loads that __sync_XXX_compare_and_swap code will generate. Here is my assembly:
```
int sem_wait (sem_t *__sem) __THROW
{
  __uint64_t value, success;
  __asm__ __volatile__("\
L%=:\n\t\
     lr.d %[value],(%[__sem])            # load reserved\n\t\
     beqz %[value],L%=                   # if zero, try again\n\t\
     addi %[value],%[value],-1           # value --\n\t\
     sc.d %[success],%[value],(%[__sem]) # store conditionally\n\t\
     bnez %[success], L%=                # if the store failed, try again\n\t\
"
    : [value] "=r"(value), [success]"=r"(success)
   : [__sem] "r"(__sem)
    : "memory");
  return 0;
}
```

To write this more effectively, I'd like to do something like this:
```
int sem_wait(register sem_t *__const__ sem)
{
   register uint64_t value;
   do value = riscv_LRD(sem);
   while(unlikely(!value) || !riscv_SCD(sem, value - 1, "memory"));
   return 0;
}
```

Where the definitions of riscv_LRD and riscv_SRD are:
```
#define likely(x)((__extension__(\
   (__typeof__((__extension__((_Bool)1))))\
   !__builtin_expect(!(x), 0)\
)))
#define unlikely(x)((__extension__(\
   (__typeof__((__extension__((_Bool)1))))\
   !!__builtin_expect(!!(x), 0)\
)))
#define static_fence()(__extension__({\
	__asm__ __volatile__("" ::: "memory");\
}))

// Emit a dummy “read” of *ptr, then lr.d from it and emit the loaded value.
#define riscv_LRD(ptr)(__extension__({\
   register __const__ __volatile__ uint64_t *__const__ riscv_LRD__0 = \
       (__const__ __volatile__ uint64_t *)(void *)(ptr); \
   register uint64_t riscv_LRD__1; \
   __asm__ __volatile__("" :: "m"(*riscv_LRD__0)); \
   __asm__ __volatile__(\
       "lr.d %0, (%1)" \
       : "=r"(riscv_LRD__1) : "r"(riscv_LRD__0)); \
   riscv_LRD__1;\
}))

// sc.d value to *ptr, then emit a dummy write of *ptr.
#define riscv_SCD(ptr, value, clobbers...)(__extension__({\
   register __volatile__ uint64_t *const riscv_SCD__0 = \
       (__volatile__ uint64_t *)(void *)(ptr); \
   register uint64_t riscv_SCD__1; \
   __asm__ __volatile__(\
       "sc.d %0, %1, (%2)" \
       : "=r"(riscv_SCD__1) \
       : "r"((uint64_t)(value)), "r"(riscv_SCD__0)); \
   __asm__ __volatile__("" : "+m"(*riscv_SCD__0) :: clobbers); \
   likely(!riscv_SCD__1);\
}))
```
I could just add this type of stuff to my own headers, but I thought it might benefit the community if there were some standard definitions and best practices.

Thoughts?

aswaterman, 2018-03-20:
This is unsafe in general, unfortunately. LR/SC has strict constraints in order to guarantee forward progress, so these sequences should always be hand-coded—even including the branch to retry the sequence.

rishikhan, 2018-03-20:
Thanks for the response. Can you point me to where these constraints are outlined? There is a quick blurb on page 41 of the riscv-spec2.2. Any other places I should read more about this?

aswaterman, 2018-03-20:
Yep, that's what I was referring to.

aswaterman, 2018-05-02:
Closing since I don't think we want to support this feature.

closed 2018-05-02 by aswaterman