riscvarchive/riscv-gcc #128: Optimization of pointer dereference

emelcher, 2018-03-23:
An old riscv-gcc dating from September 2016, compiled this code:
`int main() {`
`   volatile int * const io = (int *)252;`
`   int x,y;`
`   x = *io;`
`   y = *io;`
`   return x+y`
`}`

using option -O1 to this:
`	lw	a5,252(zero)`
`	lw	a5,252(zero)`
`	add	a0,a0,a5`
`	ret`

The up to date riscv-gcc produces this:
`	li	a5,252`
`	lw	a0,0(a5)`
`	lw	a5,0(a5)`
`	add	a0,a0,a5`
`	ret`

This is a 25% code increase for this example.
Is there an option that makes the new riscv-gcc produce code as compact as the old riscv-gcc?

jim-wilson, 2018-03-23:
Use of volatile tends to disable optimizations, which is a little unfortunate, but this is a hard issue to get right, as some simple optimizations can break use of volatile.  Because of the complexity of getting volatile support correct, sometimes different gcc versions will emit different code with volatile, due to bug fixes for volatile support.  There is no way to avoid this problem, other than to avoid use of volatile when it isn't strictly necessary.

You mentioned the old compiler is from Sept 2016, but that does not clearly identify what the old compiler is, so I am not able to say what might have changed.

It would be nice if gcc got this right.  I'll keep this open until I have a chance to look at it.

palmer-dabbelt, 2018-03-23:
I think in this case it's not a volatile issue: he wants the value loaded twice, he just doesn't want the `li`.  I'm not sure how we ever avoided emitting the `li` there, as I don't ever remember being able to fold `li` into `lw`.  I'd let the linker hand this, as that's where we usually optimize these sorts of things:

````
$ cat test.c
volatile int io;

int main() {
int x,y;
x = io;
y = io;
return x+y;
}
$ riscv64-unknown-elf-gcc test.c -o test -O3 -nostdlib -Wl,--defsym=io=252
$ riscv64-unknown-elf-objdump -d test
/scratch/palmer/work/20180320-toolchain_update/riscv-gnu-toolchain/regression/install/newlib-rv64imafdc-lp64d-medany/lib/gcc/riscv64-unknown-elf/7.3.0/../../../../riscv64-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 0000000000010078

test:     file format elf64-littleriscv

Disassembly of section .text:

0000000000010078 <main>:
   10078:       0fc02703                lw      a4,252(zero) # fc <io>
   1007c:       0fc02503                lw      a0,252(zero) # fc <io>
   10080:       9d39                    addw    a0,a0,a4
   10082:       8082                    ret
````

emelcher, 2018-03-23:
Sorry, I should have been specific about the 2016 gcc:
`$ riscv64-unknown-elf-gcc -v`
`Using built-in specs.`
`COLLECT_GCC=riscv64-unknown-elf-gcc`
`COLLECT_LTO_WRAPPER=/usr/local/riscv/libexec/gcc/riscv64-unknown-elf/5.3.0/lto-wrapper`
`Target: riscv64-unknown-elf`
`Configured with: /usr/local/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/configure --target=riscv64-unknown-elf --prefix=/usr/local/riscv --disable-shared --disable-threads --enable-tls --enable-languages=c,c++ --with-newlib --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-multilib`
`Thread model: single`
`gcc version 5.3.0 (GCC) `

The linker solution works with gcc version 7.2.0, although it is more difficult to understand:

Why io must not be a pointer?

I tried using the linker solution with io as pointer and the result was identical to what i had before, i.e. with `li a5,252`.

I also tried this test.c with version 7.2.0:
`volatile int io=252;`
`int main() {`
`int x,y;`
`x = io;`
`y = io;`
`return x+y;`
`}`
`$ riscv32-unknown-elf-gcc -o test test.c -O3 -nostdlib`
`$ riscv32-unknown-elf-objdump -d test`
`   10074:	000117b7          	lui	a5,0x11`
`   10078:	0887a503          	lw	a0,136(a5) # 11088 <io>`
`   1007c:	0887a783          	lw	a5,136(a5)`
`   10080:	00f50533          	add	a0,a0,a5`
`   10084:	00008067          	ret`

Now I am really confused because 136+0x11=153, not equal 252.

emelcher, 2018-03-23:
I got it!
io must be a variable so the linker will be responsible to assign an address in memory to it. With -defsum the linker is forced to locate io at address 252.
In my last example I let the linker chose an address where to store the content of io (in this case 153). Assigning the value 252 only sets the content of variable io, not its address.

closed 2018-03-23 by emelcher