riscvarchive/riscv-gcc #130: Building with GCC 7.2 fails with 'riscv_stack_boundary' was not declared in this scope

ilg-ul, 2018-05-06:
Attempts to build the 7.3.0 branch with GCC 7.2 fail:

```console
In file included from ./tm.h:20:0,
                 from /Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/target.h:52,
                 from /Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/c-family/c-attribs.c:23:
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/c-family/c-attribs.c: In function 'bool check_cxx_fundamental_alignment_constraints(tree, unsigned int, int)':
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/config/riscv/riscv.h:522:24: error: 'riscv_stack_boundary' was not declared in this scope
 #define STACK_BOUNDARY riscv_stack_boundary
                        ^
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/defaults.h:1135:29: note: in expansion of macro 'STACK_BOUNDARY'
 #define MAX_STACK_ALIGNMENT STACK_BOUNDARY
                             ^~~~~~~~~~~~~~
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/c-family/c-attribs.c:1546:14: note: in expansion of macro 'MAX_STACK_ALIGNMENT'
  max_align = MAX_STACK_ALIGNMENT;
              ^~~~~~~~~~~~~~~~~~~
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/config/riscv/riscv.h:522:24: note: suggested alternative: 'stv_secondarg'
 #define STACK_BOUNDARY riscv_stack_boundary
                        ^
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/defaults.h:1135:29: note: in expansion of macro 'STACK_BOUNDARY'
 #define MAX_STACK_ALIGNMENT STACK_BOUNDARY
                             ^~~~~~~~~~~~~~
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/c-family/c-attribs.c:1546:14: note: in expansion of macro 'MAX_STACK_ALIGNMENT'
  max_align = MAX_STACK_ALIGNMENT;
              ^~~~~~~~~~~~~~~~~~~
In file included from /Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/c-family/c-attribs.c:25:0:
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/c-family/c-attribs.c: In function 'tree_node* handle_aligned_attribute(tree_node**, tree, tree, int, bool*)':
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/config/riscv/riscv.h:522:24: error: 'riscv_stack_boundary' was not declared in this scope
 #define STACK_BOUNDARY riscv_stack_boundary
                        ^
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/tree.h:4323:36: note: in definition of macro 'size_int'
 #define size_int(L) size_int_kind (L, stk_sizetype)
                                    ^
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/config/riscv/riscv.h:133:27: note: in expansion of macro 'STACK_BOUNDARY'
 #define BIGGEST_ALIGNMENT STACK_BOUNDARY
                           ^~~~~~~~~~~~~~
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/defaults.h:1161:33: note: in expansion of macro 'BIGGEST_ALIGNMENT'
 #define ATTRIBUTE_ALIGNED_VALUE BIGGEST_ALIGNMENT
                                 ^~~~~~~~~~~~~~~~~
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/c-family/c-attribs.c:1582:28: note: in expansion of macro 'ATTRIBUTE_ALIGNED_VALUE'
     align_expr = size_int (ATTRIBUTE_ALIGNED_VALUE / BITS_PER_UNIT);
                            ^~~~~~~~~~~~~~~~~~~~~~~
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/config/riscv/riscv.h:522:24: note: suggested alternative: 'stv_secondarg'
 #define STACK_BOUNDARY riscv_stack_boundary
                        ^
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/tree.h:4323:36: note: in definition of macro 'size_int'
 #define size_int(L) size_int_kind (L, stk_sizetype)
                                    ^
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/config/riscv/riscv.h:133:27: note: in expansion of macro 'STACK_BOUNDARY'
 #define BIGGEST_ALIGNMENT STACK_BOUNDARY
                           ^~~~~~~~~~~~~~
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/defaults.h:1161:33: note: in expansion of macro 'BIGGEST_ALIGNMENT'
 #define ATTRIBUTE_ALIGNED_VALUE BIGGEST_ALIGNMENT
                                 ^~~~~~~~~~~~~~~~~
/Users/ilg/Work/riscv-none-gcc-7.3.0-1/riscv-none-gcc-7.3.0-1-20180506/gcc/c-family/c-attribs.c:1582:28: note: in expansion of macro 'ATTRIBUTE_ALIGNED_VALUE'
     align_expr = size_int (ATTRIBUTE_ALIGNED_VALUE / BITS_PER_UNIT);
                            ^~~~~~~~~~~~~~~~~~~~~~~
```

The fix is simple, you need to define `riscv_stack_boundary`:

```c
extern unsigned riscv_stack_boundary;
```

in `riscv.h`.

For a functional example, you can take a look at:

https://github.com/gnu-mcu-eclipse/riscv-none-gcc/commit/6d6363ebaf0190dc5af3ff09bc5416d4228fdfa2

jim-wilson, 2018-05-06:
Are you trying to use the github riscv-gcc-7.3.0 branch?  I would strongly advise against that.  This is unfinished experimental work to try to bring FSF GCC 7.3 into our github tree, and forward port local patches.  This branch is in an unknown state, and may require significant work to be usable.

The default branches used by riscv-gnu-toolchain are known to work.  The upstream FSF Binutils and GCC branches are known to work.  Random branches in riscv-gcc should be avoided unless you have seen any info that says it is OK to use them.

If you want FSF GCC 7.3, take it from gcc.gnu.org.

ilg-ul, 2018-05-06:
> Are you trying to use the github riscv-gcc-7.3.0 branch?

Yes, I thought the first line of the bug report is self-explanatory.

> I would strongly advise against that.

ok.

---

but the bug report still stands, building 7.3.0 with 7.2 fails.