riscvarchive/riscv-gcc #131: Allow -march to have a default

alonamid, 2018-05-10:
RISC-V requires an explicit march string. However, many build systems for standard software use -march=native.
It would be nice it -march=native would default to something generic like rv64 
(maybe with a warning, but not with an error?)

jim-wilson, 2018-05-10:
-march=native is a target dependent option.  It should not be used if the target does not support it.  

Proper support for -march=native requires parsing /proc/cpuinfo or modifying the kernel to pass info down via AT_HWCAP.   GCC then needs to use the info to determine how to set arch/tune options.  I see that /proc/cpuinfo does have the info I need on a Fedora system, so it should be possible to implement this.  It would be a nice feature to have.  It may take some time to implement, as there are many items on my to do list for the RISC-V gcc port.

I'd rather not have a fake -march=native option unless there is some overriding need for it.  None of the people building linux distros have asked about this for instance, so it doesn't seem to be a major issue.  We should add support for it though, to make porting easier from ARM and x86.

palmer-dabbelt, 2018-05-18:
I agree with Jim: no fake `-march=native` option, but it'd be nice to have a native one on real systems.  IIRC the x86 GCC port doesn't accept `-march=native` for cross compilers, so this isn't even really a RISC-V specific issue.  We probably want `-mtune=native` as well.

Unfortunately, this won't be a high priority: the only RISC-V systems that can run GCC right now are `-march=rv64gc` and `-mtune` doesn't really do anything, so there's no benefit.  I vote we leave the bug open, though, as maybe someone will submit a patch :)

vineetgarc, 2023-12-12:
@kito-cheng I'd like to tackle this for gcc-14.
So looking at aarch64 they seem to be parsing /proc/cpuinfo. For RV, if we want to use riscv_hwprobe stuff then it needs to be beefed up big time. ATM the kernel only supports a handful of extensions.
```
arch/riscv/include/uapi/asm/hwprobe.h
#define RISCV_HWPROBE_KEY_IMA_EXT_0     4
#define         RISCV_HWPROBE_IMA_FD            (1 << 0)
#define         RISCV_HWPROBE_IMA_C             (1 << 1)
#define         RISCV_HWPROBE_IMA_V             (1 << 2)
#define         RISCV_HWPROBE_EXT_ZBA           (1 << 3)
#define         RISCV_HWPROBE_EXT_ZBB           (1 << 4)
#define         RISCV_HWPROBE_EXT_ZBS           (1 << 5)
#define         RISCV_HWPROBE_EXT_ZICBOZ        (1 << 6)
```

kito-cheng, 2023-12-12:
@vineetgarc Yeah, I guess start from parsing `/proc/cpuinfo` is much easier, the arch string parser already there, appropriate you take this :)

luhenry, 2023-12-12:
Isn't it that /proc/cpuinfo only reports extensions that the kernel is aware of? And if the kernel is aware of an extension, shouldn't it always be reported as part of hwprobe? (Sounds like a bug otherwise IMO)

What information do you see in /proc/cpuinfo that's not in hwprobe?

From contributing to OpenJDK, hwprobe is the most reliable source of information we have, especially given the current reliance on QEMU and the lack of good, reliable hardware using fairly-close-to-mainline kernels.

vineetgarc, 2023-12-12:
As I pasted above from the current kernel source tree, the uapi part of hwprobe seems pretty minimal ATM - only exports a handful of extensions. I don't know if that is by choice or just that there aren't enough users of facility to start complaining. @palmer-dabbelt ?

luhenry, 2023-12-12:
We are relying on hwprobe to provide as much information as the kernel has. So if some information isn't available in hwprobe, that's IMO a bug. (I'm not a kernel maintainer, only an interested user, so take that as an opinion)

kito-cheng, 2023-12-12:
Even hwprob and /proc/cpuinfo provide same info, I would still preferred /proc/cpuinfo at this moment, because it not depends on the glibc and kernel version.

vineetgarc, 2023-12-12:
Discussed this with @palmer-dabbelt and he seems to prefer the hwpobe way of doing stuff. 
There's an upcoming kernel patch which seems to beef up hwprobe significantly.
https://git.kernel.org/pub/scm/linux/kernel/git/palmer/linux.git/commit/?h=for-next&id=6c312912da5cefafd501f47afb59c1a3cf54e6cc

kito-cheng, 2023-12-13:
@vineetgarc I am OK with hwprobe, moving forward better than stop here :)