riscvarchive/riscv-gcc #133: Assembler and mtval/mbadaddr CSR names

AntonKrug, 2018-06-21:
I'm using compiler build from the 7.2.0 branch and I have to still use mbadaddr name for the CSR instead of the new one mtval. Somewhere in the mailing list or here on the GitHub, I was asking about this issue and was told that the assembler is lagging behind. 

That was around half year ago, so now I wonder, got this fixed eventually in other branches (7.3.0 / 8.1.0) or the situation is the same everywhere? It's a bit confusing to try and follow the newest spec when the assembler itself clearly doesn't adhere fully to the spec.

jim-wilson, 2018-06-21:
This is fixed in FSF binutils sources.

Everything riscv-gnu-toolchain is hopelessly dated, and there is no target date to fix it, though everyone is aware that it desperately needs fixing.  The situation is complicated because half the stuff is upstream and half isn't, which makes any upgrade a major project.  There are also far too many other projects that depend on riscv-gnu-toolchain and its components, some of which will likely break once we start upgrading riscv-gnu-toolchain and its parts.  There are also multiple gcc releases to choose from now, and it isn't obvious which one we should be upgrading too, as some people want gcc-7.3 and some people want gcc-8.1.  Etc.  I don't think there is any good solution here other than to start telling people to stop relying on riscv-gnu-toolchain.

I would suggest using FSF binutils and gcc releases instead of riscv-gnu-toolchain.  You could even try using FSF top-of-tree, which is pretty safe for binutils, but a little risky for gcc as gcc is sometimes accidentally broken by new optimization work.  Depending on what you are doing, you might still need the riscv-gnu-toolchain gdb though.

AntonKrug, 2018-06-21:
Thanks @jim-wilson 
I sounds as tricky situation, with the upstreaming stuff, probably the best is to wait it out and not to try mess with things until the upstreaming got finished.

It's hard (if not impossible) to make perfect solution to make everybody happy :/

jim-wilson, 2018-07-31:
riscv-gnu-toolchain was upgraded to gcc-8.1.0 and binutils-2.30, and has the mtval support now.

closed 2018-07-31 by jim-wilson