riscvarchive/riscv-gcc #137: SW and LW instead of SD and LD

pentin-as, 2018-08-07:
RV64I

I have a code:
```
typedef unsigned int uint32_t;
int main(int argc, char** argv)
{
  uint32_t volatile a0 = 0;
  uint32_t volatile a1 = 1;
  uint32_t volatile a2 = 2;
  uint32_t volatile a3 = 3;
  uint32_t* p = (uint32_t*) 0x81000000;
  *(p + 0) = a0;
  *(p + 1) = a1;
  *(p + 2) = a2;
  *(p + 3) = a3;
}
```
It compiles to:
```
    80001bd0:	00100793          	li	a5,1
    80001bd4:	00f12223          	sw	a5,4(sp)
    80001bd8:	00200793          	li	a5,2
    80001bdc:	00f12423          	sw	a5,8(sp)
    80001be0:	00300793          	li	a5,3
    80001be4:	00f12623          	sw	a5,12(sp)
    80001be8:	00012703          	lw	a4,0(sp)
    80001bec:	08100793          	li	a5,129
    80001bf0:	01879793          	slli	a5,a5,0x18
    80001bf4:	00e7a023          	sw	a4,0(a5)
    80001bf8:	00412703          	lw	a4,4(sp)
    80001bfc:	00000513          	li	a0,0
    80001c00:	00e7a223          	sw	a4,4(a5)
    80001c04:	00812703          	lw	a4,8(sp)
    80001c08:	00e7a423          	sw	a4,8(a5)
    80001c0c:	00c12703          	lw	a4,12(sp)
    80001c10:	00e7a623          	sw	a4,12(a5)
```
The increment of the pointer is calculated correctly: 0(x), 4(x), 8(x), 12(x). Memory in the stack stands out correctly.
But instead of the instructions SD and LD, instructions SW and LW are used to write 64-bit words.

[SW_LW_test.zip](https://github.com/riscv/riscv-gcc/files/2266449/SW_LW_test.zip)

jim-wilson, 2018-08-07:
uint32_t is a 32-bit type, so of course 32-bit loads and stores are being used.  If you want 64-bit loads and stores, you have to use a 64-bit type like uint64_t.  If you want a type that is 32-bit for a 32-bit compiler and 64-bit for a 64-bit compiler you can use "unsigned long" instead of one of the uintXX_t types.

pentin-as, 2018-08-07:
Excuse me.
I'm confused. I thought LW and SW are 64-bit, and LD and SD are 32-bit.

closed 2018-08-07 by pentin-as

jim-wilson, 2018-08-07:
FYI W is for word.  D is for double-word.  This follows the pattern of a lot of 32b/64b processors, which originally had 32-bit word sizes, and so had to use double-word to refer to 64-bit values.