riscvarchive/riscv-gcc #138: Cannot build shared library with load-time relocation

cejpmart, 2018-09-05:
library.c:

```c
static int my_int;

int getvalue() {
	return my_int;
}
```

Attempt to compile without -fPIC:

```
➤ riscv-none-embed-gcc -march=rv32i -mabi=ilp32 -ffreestanding -shared -o library.so library.c
/opt/gnu-mcu-eclipse/riscv-none-gcc/7.2.0-4-20180606-1631/bin/../lib/gcc/riscv-none-embed/7.2.0/../../../../riscv-none-embed/bin/ld: /tmp/ccqXLEjF.o: relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/tmp/ccqXLEjF.o: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
```

This seems wrong, because the same works fine with arm-none-eabi-gcc and x86-linux-gcc

ilg-ul, 2018-09-05:
I don't have much experience with shared libraries, but, from what I recall, the common use case requires two steps, first to compile the sources with `-fpic`, and the second to link all objects together with `-shared`.

cejpmart, 2018-09-05:
@ilg-ul , that is exactly what I need to avoid. Using `gcc -fPIC` generates a metric ton of GOT/PLT bloat which might make sense for virtual-memory operating systems, but is completely useless for my purposes :)

My ultimate goal is what PE/COFF does (or used to do?), that is load-time in-place relocation + import resolution, without all the runtime indirection of GOT.

aswaterman, 2018-09-05:
The compiler's doing the right thing here, since the default code model uses absolute addressing but the RISC-V ABIs do not support load-time in-place relocations.

In this special case, you can use -mcmodel=medany to force PC-relative relocations.  That won't work in general, though: without a GOT, any symbols unresolved at static link time will resolve incorrectly.

(FWIW, in your example, -fPIC would not generate any GOT references, because the variable is known to resolve locally.)

cejpmart, 2018-09-06:
Andrew, thanks for your reply.

In technical terms, what exactly do you mean by "RISC-V ABIs do not support load-time in-place relocations"? Are the needed relocation types missing in RISC-V ELF?

I now understand that trying to avoid the GOT for imported symbols doesn't make much sense, but is there at least a way to skip the PLT for imported functions? Neither `-fno-plt` nor `-z now` have had any effect in my experiments.

aswaterman, 2018-09-06:
I just meant that support for this kind of reloc was a conscious omission from the existing ABIs.  They were designed for Linux ELF (where the GOT and PLT are appropriate) and embedded static links (where this discussion is immaterial).  Alternate ABIs are of course conceivable.

There is a way to bypass the PLT for locally binding functions.  I'm not an expert on ELF linkage, so I can't give you the right incantation, but I do know that gnu libc uses this mechanism internally to skip the PLT for calls known to be internal to libc.

closed 2018-09-06 by cejpmart