riscvarchive/riscv-gcc #141: compressed instructions in rv32imc and rv32imac

brabect1, 2018-12-03:
Hi,

I am getting differences in compressed instructions between the two architectures.

I tried to compile the `hello` example from sifive/freedom-e-sdk using a pre-compiled toolchain (used latest from SiFive and GNU MCU Eclipse). There were differences in code size between `-march=rv32imac` and `-march=rv32imc` (both with `-mabi=ilp32` and other flags the same). I would expect the two be the same on a such a simple example as there is little opportunity for the A extension.

The code generated with `rv32imc` does not use compressed instructions where the other, `rv32imac` does. It seems to occur only for certain parts of the code (`libwrap`???), as some other parts use the same compressed instruction in for both architectures. For example:

```
// rv32imc
...
20400458 <atexit>:
20400458:   00050593            mv  a1,a0
2040045c:   00000693            li  a3,0 
20400460:   00000613            li  a2,0 
20400464:   00000513            li  a0,0 
20400468:   2300006f            j   20400698 <__register_exitproc>
...

// rv32imac
...
20400450 <atexit>:
20400450:   85aa                    mv  a1,a0
20400452:   4681                    li  a3,0
20400454:   4601                    li  a2,0
20400456:   4501                    li  a0,0
20400458:   a299                    j   2040059e <__register_exitproc>
...
```

Could this be a compiler issue?

jim-wilson, 2018-12-03:
It is a multilib issue.  We have a potentially infinite number of possible architecture variants, and can't build a separate copy of the libraries for each one, so we just build a subset.  You can see the set of libraries that were built by using, e.g. "riscv64-unknown-elf-gcc --print-multi-lib".  You can see the library that will be used for your compilation by using, e.g. "riscv64-unknown-elf-gcc --print-libgcc -march=rv32imc -mabi=rv32im", which shows that we don't have a rv32imc copy of the libraries, so it uses a rv32im one instead, which won't have compressed instructions.

You can get better results if you configure a build a compiler that specifically targets your architecture variant of choice, instead of using a general compiler and relying on the multilib support.

brabect1, 2018-12-04:
Thanks, Jim! I believe this explains and closes the issue.

closed 2018-12-04 by brabect1