riscvarchive/riscv-gcc #144: gcc -O2 duplicates single asm statement inside infinite loop.

oskarguan, 2019-01-02:
I found "gcc -O2" duplicates single "asm volatile" statement inside a infinite loop in the following context.
```
// ...
printf("Entering infinite loop...\n");
for(;;) {
    asm volatile ("wfi"); // generates two consecutive wfi instructions with "gcc -O2"
}
```

So that I make the following test program (test_loop.c):
```
void test_loop1()
{
    asm volatile ("nop");
    while (1) {
	asm volatile ("wfi");
    }
    return;
}

void test_loop2()
{
    while (1) {
	asm volatile ("wfi");
    }
    return;
}
```
And I compile with:
```
riscv32-elf-gcc -O2 -S test_loop.c
```

The result is 
```
	.file	"test_loop.c"
	.option nopic
	.text
	.align	2
	.globl	test_loop1
	.type	test_loop1, @function
test_loop1:
 #APP
# 4 "test_loop.c" 1
	nop
# 0 "" 2
 #NO_APP
.L2:
 #APP
# 6 "test_loop.c" 1
	wfi
# 0 "" 2
# 6 "test_loop.c" 1
	wfi
# 0 "" 2
 #NO_APP
	j	.L2
	.size	test_loop1, .-test_loop1
	.align	2
	.globl	test_loop2
	.type	test_loop2, @function
test_loop2:
.L5:
 #APP
# 14 "test_loop.c" 1
	wfi
# 0 "" 2
 #NO_APP
	j	.L5
	.size	test_loop2, .-test_loop2
	.ident	"GCC: (GNU) 8.2.0"
```

test_loop1 has two "wfi" instruction between ".L2" and "j .L2"; while test_loop2 has only one "wfi" between ".L5" and "j .L5".

aswaterman, 2019-01-02:
GCC is doing this as part of the basic-block reordering pass.  I'm not sure why it chose to do this, or why it differs between the two examples, but it's correct either way.  You'll see that the second WFI disappears if you use the `-fno-reorder-blocks` flag.

oskarguan, 2019-01-02:
I also tested the same program with sparc and powerpc gcc. They both show the same reordering behavior.  Should I report this "bug" or "feature" to GCC rather than this backend-repo?

jim-wilson, 2019-01-02:
GCC is an optimizing compiler.  In this case, it is trying to increase the size of basic blocks and reduce the number of branches by duplicating a block of code.  It is a bit silly in this case, but gcc doesn't know what the asm statement is for, and the code is still valid after the optimization.

Besides disabling the basic-block reordering pass, you can also fix this by changing the bb reorder algorithm.  The "simple" algorithm is used with -O1 and below, and with -Os.  The "stc" (software trace cache) algorithm is used with -O2 and up.  The stc algorithm will duplicate blocks.  The simple algorithm won't.  So you can avoid the duplication by using -freorder-blocks-algorithm=simple along with -O2.

jim-wilson, 2019-01-02:
You are welcome to file upstream.  We won't do anything about this here.  I wouldn't call this a bug, just an unexpected and unwanted optimization, and it might be useful to have a way to disable it in the source code.  There was a recent thread about an asm with a label, suggesting that we add some kind of don't copy attribute that could be added to asms that should never be copied.  If that feature was added, you could use it to prevent the unwanted optimization here.

closed 2019-01-21 by oskarguan