riscvarchive/riscv-gcc #145: Why is there no nop in the assembly code when the compiler uses -O2?

l919898756, 2019-01-09:
Hello, take the liberty to ask you a question about gcc.
When I compile a program with -O0, there will be nop instructions in the generated assembly code. There is no nop command when using -O2.
I found that -O2 will use -fschedule-insns and -fschedule-insns2 for instruction scheduling.
I used -fsched-verbose to print the scheduling information of this program and found that some places are like this.

[image]

2 and 4 need to wait for one cycle.
Here I think it should be necessary to add a nop in the assembly code, because if my processor is a single launch, there should be no error without this nop.

When using -O0 to compile the program, the instruction will not be dispatched, so there will be nop. After the instruction is dispatched, is the generated nop deleted?
Because there is also a waiting cycle here, I think there will be a nop here.

I can't generate scheduling information when compiling with -O0, so I don't know how O0's nop is generated.

May i know what is this all about? Thank you very much for your answer.
@jim-wilson

l919898756, 2019-01-10:
I don't think my machine has hardware interlocks. So I need to add some nop to him. Excuse me, where should I plug in nop? When I execute instruction scheduling in gcc, if I find a free period, do I add the corresponding nop in RTL? Then RTL will generate the corresponding nop when it converts the assembly code? I found that in the GCC RTL to assembly code there is a final function that will traverse all the RTL, and then the production will change the code, where final will call get_insns () to get each RTL, but I can not find the instruction to save and save Where will the final RTL code be saved? May I ask what I think? My English is a bit poor, I don't know if you can understand it. @jim-wilson

jim-wilson, 2019-01-10:
Machines without interlocks or pipeline forwarding or other tricks to avoid hazards are not very common anymore, unless maybe this is your own personal design, and is your first cpu design.

If only a few instructions lack interlocks, like multiply/divide, then the easy solution is to add nops to those instruction patterns in the gcc/config/riscv/riscv.md file.  You will also need to change the instruction length to be correct, counting the nops you added.

If a lot of instructions lack interlocks, or if you want better optimization, then you probably need to add a machine dependent optimization pass to go through the entire instruction stream, look at each instruction, and figure out where to emit nops, and how many nops in eash location.  This can get pretty complicated, and requires a fair amount of gcc internals knowledge to implement.  You can find an example in the mips port, as some old MIPS parts lack hardware interlocks for some instructions.  In gcc/config/mips/mips.c, the mips_avoid_hazard function computes how many nops to emit after an instruction.    This is ultimately called from the mips_machine_reorg2 machine dependent optimization pass.  This uses hazard define_attr in mips.md to specify which instructions have hazards that require nops, and what kind of hazard they generate.    See also mips_adjust_insn_length which is used to compute the instruction length plus the hazard avoiding nops.  There may also be other things scattered around the MIPS port required to make this work.

There are also a few other ports that do this.  You could try looking for other ports with "reorg" functions in the port .c file that can add nops, if you want other ideas on how to do this.

yxj1, 2019-01-11:
I can write an insert nop according to this nop. But I don't know how to locate the source code generated by this instruction.
[image]

jim-wilson, 2019-01-11:
​If you want to do anything in this area, you will have to spend some time learning gcc internals.  Or hire someone that already know how to do this kind of work.  I gave pointers to code in the MIPS port that implements what you need.​

This will emit a nop
emit_insn (gen_nop ());
but of course the hard part is writing the optimization pass to find where to emit the nops that you need.

Jim

yxj1, 2019-01-22:
Thank you very much for your help, I think I should find a solution.@jim-wilson