riscvarchive/riscv-gcc #146: C.LUI instead of ADDI if imm is >= 32 or imm <=-33

MorgothCreator, 2019-01-13:
Not tested how bigger need to be the imm to generate the correct instruction.

If imm is >= 32 or imm <=-33

For example if imm is 32 the disasempler will show correctly uncompressed instruction but illegal compressed instruction because imm is bigger than 31 "addi sp,sp,32" and the generated code is 0x6105 that is a C.LUI instruction.

```
    8220:	6105                	addi	sp,sp,32
    8612:	7179                	addi	sp,sp,-48
```

riscv-none-embed-g++ -march=rv32iac -mabi=ilp32 -msmall-data-limit=8 -Og -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections  -g3

aswaterman, 2019-01-13:
This is an objdump thing, not a GCC thing. The disassembler is expanding the instruction from C.ADDI16SP to the base-ISA instruction ADDI, because it’s more readable. If you want to see the actual compressed opcode, pass the -Mno-aliases flag to objdump.

MorgothCreator, 2019-01-13:
0x6105 = 16b'0110000100000101 and 0x7189 = 16'b0111000101111001 these codes are for 16'b011xxxxxxxxxxx01 that is C.LUI instruction.
For the C.ADDI16SP is 16'b000x00000xxxxx01 the same as C.ADDI but with rs/rd = 0 and rearanged nzimm.
The C.ADDI16SP bits [11:7] need to be all 0, and in this case is the SP register x2.

Take a look on the attached lss files, line 351, 444, 532, 856 and 894.

[TestRiscVXilinx.zip](https://github.com/riscv/riscv-gcc/files/2753631/TestRiscVXilinx.zip)

[TestRiscVXilinx_-Mno-aliases.zip](https://github.com/riscv/riscv-gcc/files/2753642/TestRiscVXilinx_-Mno-aliases.zip)

I can share the entire test application, to replicate the issue.

aswaterman, 2019-01-13:
C.ADDDI16SP has the same opcode as LUI, but with bits[11:7] = 2.  Both of those instructions are C.ADDI16SP.

MorgothCreator, 2019-01-14:
Oh, this clarify everything.
I read this doc https://people.eecs.berkeley.edu/~krste/papers/riscv-compressed-spec-v1.7.pdf for the compressed implementation and there figure bits 11-7 as zero.
Thank you very much for clarifiyng this for me.
Best regards.

closed 2019-01-14 by MorgothCreator

aswaterman, 2019-01-14:
Yep, this is a newer version. You’re welcome.