riscvarchive/riscv-gcc #152: Unnecessary sext.w after amoor.w in __atomic_fetch_or on RV64

bennoleslie, 2019-02-28:
Apologies in advance if this is the wrong forum for this question / issue report. If there is a better place please let me know.

I'm compiling some simple code to put a character on the SiFive Unleashed board. o

```
#define UART_TXDATA (uint32_t *)0x10010000

void
uart_putc(char ch)
{
    uint32_t r;
    do {
        r = __atomic_fetch_or(UART_TXDATA, ch, __ATOMIC_RELAXED);
    } while (r);
}
```

The generated code is mostly what I expect, however there appears to be a redundant `sext.w` instruction in the generated code:

```
0000000000000000 <uart_putc>:
uart_putc():
/home/ubuntu/riscv/start.c:10
   0:   10010737                lui     a4,0x10010

0000000000000004 <.L0 >:
/home/ubuntu/riscv/start.c:10 (discriminator 1)
   4:   40a727af                amoor.w a5,a0,(a4)
   8:   2781                    sext.w  a5,a5

000000000000000a <.L0 >:
/home/ubuntu/riscv/start.c:11 (discriminator 1)
   a:   ffed                    bnez    a5,4 <.L0 >

000000000000000c <.L0 >:
/home/ubuntu/riscv/start.c:12
   c:   8082                    ret
```

Of course, I could be wrong here, and it is necessary for some reason.

I'm not sure if this older issue is related https://github.com/riscv/riscv-gcc/issues/83

I'm running version 7.3 (if you need me to compile the latest version from source I can do that, this is just whats available on Ubuntu 18.04):

```
$ riscv64-linux-gnu-gcc --version
riscv64-linux-gnu-gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
```

Compiling with `-O2 -g3` flags. (Also tried with -O1 and -O3 but not significant difference with respect to the sext.w)

jim-wilson, 2019-02-28:
The compiler doesn't know that amoor.w sign extends its result to 64-bits.  There are a lot of issues like this in the RISC-V port that can result in redundant sign extension instructions.  Some of them can be fixed by adding more patterns to the riscv md files.  Some of them are harder to fix and might require a redesign of how we handle .w instructions for the 64-bit target.  Some of them might require improving optimization passes to better optimize for the RISC-V instruction set which is an even harder problem to solve.

bennoleslie, 2019-02-28:
Thanks Jim. In this case I'm a bit confused as all my types are are unsigned values, so I'm a little confused as to why any sign extension code is generated (but I'm probably misunderstanding something).

palmer-dabbelt, 2019-02-28:
The RISC-V ISA is biased towards sign extension (pretty much every instruction sign extends the result when it writes a register), so as a result our GCC port maintains the invariant that all types are stored in registers sign extended to `XLEN`.  Specifically in this case the GCC port doesn't understand that `amoor.w` writes back a sign extended value, so it goes and sign extends the value before doing the comparison in the while (which expects a sign extended value).  In this specific case it turns out it's doubly unnecessary (the AMO sign extends, and the compare with 0 doesn't care), but our GCC port isn't structured to understand this.

The canonical way to fix this is to inform GCC that the relevant pattern sign extends the result to XLEN.  I think the pattern in question is in sync.md: https://github.com/gcc-mirror/gcc/blob/1ca0c398dc308c08cba3dad47fe09204306ddddc/gcc/config/riscv/sync.md#L82 .  That's a fairly complex pattern, but I think just adding a sign extension around the output register should do it.

This sort of problem is endemic to the RISC-V port because it's hard to find these extra sign extensions -- they don't manifest as incorrect code, so they only get found when someone is looking at the assembly.  Since we usually only look at working assembly when there's a performance problem, these sort of issues tend to get found in a fairly ad-hoc manner.

jim-wilson, 2019-02-28:
On Thu, Feb 28, 2019 at 12:17 AM Ben Leslie <notifications@github.com> wrote:
> Thanks Jim. In this case I'm a bit confused as all my types are are unsigned values, so I'm a little confused as to why any sign extension code is generated (but I'm probably misunderstanding something).

GCC thinks that amoor.w produces a 32-bit result, where the high order
32-bits are undefined.  We have only 64-bit compares.  So we need to
extend it.  The type of extension doesn't matter as this is a compare
against zero, and as Palmer mentioned, we sign-extend by default.

jim-wilson, 2019-02-28:
Adding a sign-extend to the output is the obvious solution, but is problematic.  Some gcc optimization passes will be confused by sign extensions in unexpected places, and then fail to perform the optimizations they are designed to do.  The loop optimizers in particular get confused by this.  So the end result is that if you aren't careful, you end up disabling loop optimization to eliminate an unnecessary sign extend, which is a cure worse than the disease.  I know because I've tried this.  This needs some very careful testing and time intensive work to fix.  It may be the case that if we only add combiner patterns with sign extends and don't change the named patterns it might work.  But again, this needs careful testing, and a lot of time to investigate multiple solutions to find the best one.

palmer-dabbelt, 2019-02-28:
Ya, I agree: these things are always super fragile and frequently the obvious solution results in worse overall performance (or just blows up).  It's definitely not an easy thing to fix, but it is at least a concrete issue to play with.

bennoleslie, 2019-02-28:
Thanks Palmer and Jim for the detailed explanation. That's helped a lot in my understanding of what's happening, and helped solidify my understanding of how the ISA works. Unfortunately my knowledge of GCC internals is non-existent so I don't think I can contribute much more. Before I leave it to the experts one more question: is there any value in reporting any other unnecessary sext.w? I don't want to contribute unnecessary noise if it is unhelpful.

jim-wilson, 2019-03-01:
It is useful to know about them.  We can't fix them if we don't about them.  But we are unlikely to be able to fix most of them in the short term.  And some of them are likely to be the same issue.  Any unnecessary sign extension after a .w instruction is probably the same issue as this one.

palmer-dabbelt, 2019-03-01:
If you have a short C snippit that generates bad code then you should report it as a new bug.  Like Jim said, we might not have time to fix it, but we do try to at least collect them here so we know about them.  We have a bunch of this sort of thing floating around this repo, so I think it's the best place to keep them.